FrameworkComputer/SoftwareFirmwareIssueTracker #2: Cros-usbpd-charger error -71 in journal

andreaippo, 2024-11-06:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
03.05

### DIY Edition information
If you are experiencing an issue on a DIY system, Please also fill out the memory and storage devices you are using. 

Memory: Manufacture and SKU
Crucial DDR5-5600 CT16G56C46S5.M8G1 (https://www.amazon.it/Crucial-5600MHz-5200MHz-Portatile-CT16G56C46S5/dp/B0BLTDRRLF?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dib=eyJ2IjoiMSJ9.UCVENdfzRFgchPDy2Umg6UgjMdbodZ0myHJ2sYl3iW_GjHj071QN20LucGBJIEps.L3stUHcSAdXgi6Am_DrgDlfB7wl1qKTtllpubnJqHXU&dib_tag=se&keywords=CT16G56C46S5&qid=1730880495&sr=8-1&th=1)

Storage: Manufacture and SKU
Lexar NM790 1TB

### Port/Peripheral information

I'm using a USB-C 7 (or 8)-in-1 dongle by ANKER. It's plugged into one of the USB-C adapters at the back of my FW 13 and acts as a charging bypass as well as expansion hub allowing me to plug an HDMI external monitor and two USB-A radio receivers (for wireless mouse and keyboard).
```
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 32ac:0002 Framework HDMI Expansion Card
Bus 001 Device 003: ID 27c6:609c Shenzhen Goodix Technology Co.,Ltd. Goodix Fingerprint USB Device
Bus 001 Device 004: ID 0e8d:e616 MediaTek Inc. Wireless_Device
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 0bda:5634 Realtek Semiconductor Corp. Laptop Camera
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 002: ID 2109:2822 VIA Labs, Inc. USB2.0 Hub             
Bus 007 Device 003: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub             
Bus 007 Device 004: ID 248a:8327 Maxxter Wireless Receiver
Bus 007 Device 005: ID 062a:4102 MosArt Semiconductor Corp. Wireless Mouse
Bus 007 Device 006: ID 291a:8383 Anker                   Anker USB-C HUB Device 
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 002: ID 2109:0822 VIA Labs, Inc. USB3.1 Hub             
Bus 008 Device 003: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub             
Bus 008 Device 004: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
Bus 008 Device 005: ID 2537:1081 Norelsys NS1081
```

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] No

## Describe the bug
A clear and concise description of what the bug is.

## Steps To Reproduce
- Boot device on Linux (opensuse Tumbleweed 20241104 or any other recent snapshot)
- check `journalctl -b -k` logs and look for `cros-`
- notice errors related to cros modules:
```
Oct 31 08:16:53 andromeda kernel: cros-charge-control cros-charge-control.4.auto: Framework charge control detected, preventing load
Oct 31 08:16:53 andromeda kernel: cros-usbpd-charger cros-usbpd-charger.5.auto: No USB PD charging ports found
Oct 31 08:16:53 andromeda kernel: cros-usbpd-charger cros-usbpd-charger.5.auto: Unexpected number of charge port count
Oct 31 08:16:53 andromeda kernel: cros-usbpd-charger cros-usbpd-charger.5.auto: Failing probe (err:0xffffffb9)
Oct 31 08:16:53 andromeda kernel: cros-usbpd-charger cros-usbpd-charger.5.auto: probe with driver cros-usbpd-charger failed with error -71
```

## Expected behavior
No errors in the logs

## Screenshots
N/A

## Operating System (please complete the following information):
 - OS/Distribution: opensuse tumbleweed snapshot 20241104
- Linux Kernel Version: 6.11.5-2-default (opensuse tumbleweed's)

## Additional context

andreaippo, 2024-11-06:
Logs:
[cros-usbpd-charger-error--71.txt](https://github.com/user-attachments/files/17643911/cros-usbpd-charger-error--71.txt)

DHowett, 2024-11-06:
[Relevant discussion on LKML](https://lore.kernel.org/lkml/20240406191734.137797-1-superm1@gmail.com/t/)

Additional information snipped from e-mails:

---

`EC_FEATURE_USB_PD` should not be enabled by `CHARGE_MANAGER`, rather,
it should be enabled by `USB_POWER_DELIVERY`. That seems an oversight on
the part of the upstream ec maintainers.

```diff
diff --git a/common/ec_features.c b/common/ec_features.c
index b7e16ec6ba..781094a884 100644
--- a/common/ec_features.c
+++ b/common/ec_features.c
@@ -82,7 +82,7 @@ uint32_t get_feature_flags0(void)
 #ifdef CONFIG_HOSTCMD_PD
                          | EC_FEATURE_MASK_0(EC_FEATURE_SUB_MCU)
 #endif
-#ifdef CONFIG_CHARGE_MANAGER
+#ifdef CONFIG_USB_POWER_DELIVERY
                          | EC_FEATURE_MASK_0(EC_FEATURE_USB_PD)
 #endif
 #ifdef CONFIG_ACCEL_FIFO
```

sdmcclain, 2024-12-02:
I am seeing the same error in the journalctl logs of my Framework 16. What is the impact of this issue on the operation of the laptop?

DHowett, 2024-12-02:
This does not impact the functioning of the laptop.

quinchou77, 2025-03-24:
The code has been merged. Will be released in the future release.

JohnAZoidberg, 2025-04-22:
Public commits:
- Framework 13 and Framework 16 AMD Ryzen 7040: https://github.com/FrameworkComputer/EmbeddedController/commit/1d7bcf1d50137c8c01969eb65880bc83e424597e
- Framework 13 Intel Core Ultra: https://github.com/FrameworkComputer/EmbeddedController/commit/420ad126ce8b0a4f28de04b6899907d5c2f602dd

All new platforms in 2025 will ship with this change already included.

rubin55, 2025-12-27:
@JohnAZoidberg This happens on Framework 13 13th gen (raptor lake) too; when will this end up in a firmware update that we can test?