FrameworkComputer/SoftwareFirmwareIssueTracker #3: ucsi_acpi USBC000:00: unknown error 0

tlvince, 2024-11-06:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [x] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION

03.05

### DIY Edition information

- Memory: [Crucial CT2K16G56C46S5](https://uk.crucial.com/memory/DDR5/CT2K16G56C46S5) 32GB DDR5-5600 SODIMM
- Storage: [SK hynix Platinum P41](https://ssd.skhynix.com/platinum_p41/) 1TB SSD

### Port/Peripheral information

1. USB-C expansion card: Framework 60W charger
2. USB-C expansion card (no peripheral)
3. USB-C expansion card (no peripheral)
4. USB-C expansion card (no peripheral)

### Standalone Operation

- [ ] Yes
- [x] No

## Describe the bug

acpi kernel logs on boot:

```
kernel: ucsi_acpi USBC000:00: unknown error 0
kernel: ucsi_acpi USBC000:00: GET_CABLE_PROPERTY failed (-5)
kernel: ucsi_acpi USBC000:00: unknown error 256
kernel: ucsi_acpi USBC000:00: GET_CABLE_PROPERTY failed (-5)
```

## Operating System (please complete the following information):

- OS/Distribution: NixOS
- Version: 24
- Linux Kernel Version:  6.11.6

## Additional context

Various other reports of the same across kernel versions/distros: [USB C Error on boot](https://community.frame.work/t/usb-c-error-on-boot/52012)

Abdull, 2025-05-02:
I also get this error message (albeit a little different) during bootup on my Linux 13 AMD 7040, BIOS 03.05, Debian 12, (all kernels so far, including current 6.12.22).

```
May 02 15:33:26 foobar kernel: ucsi_acpi USBC000:00: unknown error 0
May 02 15:33:26 foobar kernel: ucsi_acpi USBC000:00: GET_CABLE_PROPERTY failed (-5)
May 02 15:33:26 foobar kernel: ucsi_acpi USBC000:00: unknown error 0
May 02 15:33:26 foobar kernel: ucsi_acpi USBC000:00: GET_CABLE_PROPERTY failed (-5)
```

Laptop connected to Samsung M7 UHD monitor via  USB-C vor USB-PD power delivery and video. Happens on both Framework 13 AMD USB-PD-capable ports 1 and 3.

Error messages also appear when unplugging and replugging this USB cable.

Sometimes then, charging/power supply via USB port completely stops (no more orange-/white-glowing USB port LED; also Debian 12 no longer reporting that charging is in progress, i.e. laptop is being powered from battery).

Repeateadly unplugging and replugging the USB-C cable while the laptop/Debian 12 desktop is running may eventually bring laptop/monitor to the state where monitor is providing laptop with power (i.e., LED on USB port will turn on with white or orange color).

```
May 02 16:28:46 foobar acpid[2001]: input device has been disconnected, fd 14
May 02 16:28:55 foobar kernel: amdgpu 0000:c1:00.0: [drm] *ERROR* No EDID read.
May 02 16:28:56 foobar kernel: ucsi_acpi USBC000:00: unknown error 0
May 02 16:28:56 foobar kernel: ucsi_acpi USBC000:00: UCSI_GET_PDOS failed (-5)
May 02 16:29:13 foobar kernel: amdgpu 0000:c1:00.0: [drm] REG_WAIT timeout 1us * 100 tries - dcn31_program_compbuf_size line:141
May 02 16:29:14 foobar acpid[2001]: input device has been disconnected, fd 14
May 02 16:29:16 foobar kernel: amdgpu 0000:c1:00.0: [drm] *ERROR* No EDID read.
May 02 16:29:17 foobar kernel: ucsi_acpi USBC000:00: unknown error 0
May 02 16:29:17 foobar kernel: ucsi_acpi USBC000:00: UCSI_GET_PDOS failed (-5)

May 02 16:29:47 foobar kernel: amdgpu 0000:c1:00.0: [drm] REG_WAIT timeout 1us * 100 tries - dcn31_program_compbuf_size line:141
May 02 16:29:48 foobar acpid[2001]: input device has been disconnected, fd 14

May 02 16:30:02 foobar kernel: hub 5-1:1.0: activate --> -19
May 02 16:30:03 foobar kernel: ucsi_acpi USBC000:00: unknown error 0
May 02 16:30:03 foobar kernel: ucsi_acpi USBC000:00: GET_CABLE_PROPERTY failed (-5)
May 02 16:30:04 foobar kernel: ucsi_acpi USBC000:00: unknown error 0
May 02 16:30:04 foobar kernel: ucsi_acpi USBC000:00: UCSI_GET_PDOS failed (-5)
May 02 16:30:04 foobar kernel: amdgpu 0000:c1:00.0: [drm] *ERROR* No EDID read.
```

Framework forum discussions maybe related to this issue:

* [*USB C Error on boot*](https://community.frame.work/t/usb-c-error-on-boot/52012)
* [*[RESPONDED] Linux kernel errors about cros_usbpd_charger*](https://community.frame.work/t/responded-linux-kernel-errors-about-cros-usbpd-charger/26264)
* [*Cros-usbpd-charger error -71*](https://community.frame.work/t/cros-usbpd-charger-error-71/59785)

pcoreyster, 2025-09-20:
During failure: xrandr --query shows only eDP-*; xrandr --output DP-1 --auto → “output DP-1 not found”.

Cross-checks
	•	Same monitor/cables/docks enumerate instantly on macOS (Apple Silicon) and on Framework Intel 11th/12th-gen.

Can attach on request
	•	journalctl -k / dmesg around monitor power-on
	•	xrandr --listproviders (good vs bad)
	•	/sys/class/typec snapshots (good vs bad)
	•	EDID blobs from macOS (good path) and Linux
	•	Exact cable models/lengths and Expansion Card P/Ns

pjbuterbaugh, 2026-01-21:
Having the same issue on Framework Laptop 16 (AMD Ryzen™ 7040 Series). NixOS 25.11, kernel 6.18.5, BIOS 4.03.

JohnAZoidberg, 2026-03-04:
The UCSI error messages mean that the kernel is unable to figure out the cable properties. Our firmware currently does not pass this information correctly to the kernel.
We have fixed this for some cases but some more cases remain.

 This does not impact the functionality - so any functional problems we need to debug separately.

pheiduck, 2026-05-30:
Is this upstreamed?: https://github.com/FrameworkComputer/linux/commit/00125504f43fef9a671b0d6608bb0f458896103d

daniel-g-carrasco, 2026-08-25:
Another data point, with a workaround that may help narrow this down.

**Hardware**: Framework Laptop 13 AMD 7040 (7640U), BIOS 03.18, board FRANMDCP05. Dock: StarTech DK31C2DHSPD (USB-C, DisplayPort Alt Mode + MST, not Thunderbolt), on the rear right port. Fedora-based bootc image, kernel 7.2.

**Symptom**: plugging the dock in with the system running brings up everything except the display. Mouse, keyboard, hubs, ethernet all enumerate. No DisplayPort connector ever reaches `connected`, forcing a re-probe of the GPU changes nothing, and the kernel logs nothing except:

```
ucsi_acpi USBC000:00: unknown error 0
ucsi_acpi USBC000:00: UCSI_GET_PDOS failed (-5)
```

Attached at boot, the display works every time.

**What the dock says**: its USB Billboard descriptor reports the DisplayPort alternate mode (SVID 0xFF01) as *"configuration not attempted"* (`bmConfigured = 01`) in the failing case, and *"configuration successful"* (`bmConfigured = 03`) whenever the display works. So from the dock's side the host never even tried to enter the mode. The kernel's typec sysfs exposes no altmode entries for the partner on this machine, which I assume is the same UCSI gap.

**Workaround, reproducible**: rebinding the driver makes the negotiation happen on the second try.

```
echo USBC000:00 > /sys/bus/platform/drivers/ucsi_acpi/unbind
sleep 2
echo USBC000:00 > /sys/bus/platform/drivers/ucsi_acpi/bind
```

Seven out of seven plug-ins on this machine, display up within about seven seconds each time, and the dock's Billboard then reads "successful", also with no monitor attached to it. Cost: everything on the dock drops for about a second. A data role swap on the port was tried first and is refused.

I have this wired to a udev rule on `typec` partner add events, gated on the dock's Billboard verdict so it never touches a dock that negotiated fine. Happy to share the script if useful.

Two things this does not cover, in case they matter for the root cause: the display stays black when the dock is attached during POST on 3.18, which #170 says is fixed from 3.19; and I have not yet tested 3.20 because of the bricking reports in the community thread.

taha-yassine, 2026-08-25:
@JohnAZoidberg sorry for the wild tag but you're the only one from the Framework team that ever replied here, and this issue has been open for more than 2 years now with new reports every now and then and no feedback from you guys. Can we expect the team to look into this any time soon?

alvorithm, 2026-08-31:
@JohnAZoidberg sorry but not sorry for the tag: can you clarify if Framework is interested at all in solving the EC issues, widely reported here and in other issues, as well as in endless threads in the forum? 

At this point what we need is not zombie issues and self-help groups in the Framework community forum, but a clear statement from you, Framework, that you will invest resources in this, or that you own the disaster that the (seemingly fantastic) 4xUSB-C port choice has ended up being, and have no intention to fix.

Anything but wasting more of our time here speculating that our Frameworks will be fully functional one day.

alvorithm, 2026-09-03:
Also, let's stop pretending that this is one more open-source project to which we dedicate unpaid work to help the community and ourselves. This is defective hardware (actually firmware) in a product we paid often upwards of 2000€ for. Meanwhile Framework found better things to do with our money, such as new products, or subsidising the fascist kapo of Omarchy, a clearly disadvantaged individual. 

We are left with worse that a closed issue: a zombie thread to vent our frustration with no perspective of fix whatsoever, and no closure.