FrameworkComputer/SoftwareFirmwareIssueTracker #4: Laptop Issue

nickolasclarke, 2024-11-19:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [x] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
Please provide the bios version. 

03.04

### DIY Edition information
If you are experiencing an issue on a DIY system, Please also fill out the memory and storage devices you are using. 

Memory:  Crucial 2x16 GB ([CT32G56C46S5](https://www.crucial.com/memory/ddr5/CT32G56C46S5))
Storage: WD Black SN770 NVME 1TB 

### Port/Peripheral information
If you are experiencing an issue with a peripheral or an expansion card/port please fill out the following information: 
* Peripheral vendor and name.
* Port the Peripheral was connected to. Please see the following for port numbering: [Laptop 16](https://knowledgebase.frame.work/expansion-card-slot-functionality-on-framework-laptop-16-rkUjGm7cn) or [Laptop 13](https://knowledgebase.frame.work/expansion-card-functionality-on-framework-laptop-13-amd-ryzen-7040-series-SkrVx7gAh)
* Device or expansion card attached to the **Adjacent port** to the port that is having the issue.
1. USB-C
2. USB-C
3. USB-C 
4. USB-C 

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug
A clear and concise description of what the bug is.

After updating first to 3.03 and then subsequently to 3.04 automatically via the firmware updater tool in Ubuntu 24.04, my system would not successfully boot, and the init LED codes suggested that DDR OK failed. I have 2x16GB ram installed, which was working without issue before this update. I pulled one of my Crucial ([CT32G56C46S5](https://www.crucial.com/memory/ddr5/CT32G56C46S5)) 16 GB RAM sticks and it successfully booted again. It does not appear to be an issue with the the individual stick or the channel. I booted successfully using each stick alone, but could not get it to boot using both sticks, regardless of which channel they were placed in.

My thought was to lower the speed of the ram in the BIOS and try both sticks, but afaik, that is not possible? Any other thoughts? I do find it odd that this memory worked previously, but failed after the bios update.

## Expected behavior
A clear and concise description of what you expected to happen.
My system to boot with both 16 GB sticks of RAM installed. 

OS: 
Linux nclarke-framework13 6.8.0-48-generic #48-Ubuntu SMP PREEMPT_DYNAMIC Fri Sep 27 14:04:52 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

DHowett, 2024-11-21:
Have you reached out to Framework Support?

nickolasclarke, 2024-11-21:
@DHowett no, thats why I posted it here, per instructions I found on the framework community thread for the latest firmware update. I am happy to reach out though.

DHowett, 2024-11-21:
Ah, did you not have this issue on 3.02?

nickolasclarke, 2024-11-21:
No, it started when I did the update to 3.03. I troubleshot it for awhile
on 3.03 and then noticed that 3.04 was also out. I did the next upgrade,
and still experienced the same issue.

nickolasclarke, 2024-11-21:
@DHowett I worked around this with the following: https://community.frame.work/t/framework-laptop-13-intel-core-ultra-series-1-bios-3-04-release/59579/22?u=nick_clarke

chrisb92, 2024-11-26:
I experienced the same issue upgrading from 3.01 to 3.03, after applying the update the laptop would no longer POST until I took out 1 of my 2 RAM sticks (Crucial 2x 16GB DDR5-5600 - CT2K16G56C46S5) and then it was able to POST. I then updated to 3.04 with only 1 RAM stick, once it was POSTing, I turned off the laptop and put the other RAM stick back in and it started POSTing fine again after about 30+ seconds.

Framework 13 Core Ultra 7 155H DIY running Linux, applied the update using `fwupdmgr`