FrameworkComputer/SoftwareFirmwareIssueTracker #11: Laptop Issue

ALKahn10, 2024-11-21:
## Device Information

### Framwork13, 7840u, 64GB RAM, matte non-2.8k display

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ X] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
BETA 3.06

Windows:
Open a command terminal as administrator:
SMBIOSBIOSVersion
03.06

Linux: 
Open a terminal and run the following command
```sudo dmidecode --string bios-version```

### DIY Edition information
If you are experiencing an issue on a DIY system, Please also fill out the memory and storage devices you are using. 

Memory: 64GB (2 x 32GB) DDR5-5600 Framework Branded

Storage: Samsung 990 Pro 2GB

### Port/Peripheral information
If you are experiencing an issue with a peripheral or an expansion card/port please fill out the following information: 
* Peripheral vendor and name.
* Port the Peripheral was connected to. Please see the following for port numbering: [Laptop 16](https://knowledgebase.frame.work/expansion-card-slot-functionality-on-framework-laptop-16-rkUjGm7cn) or [Laptop 13](https://knowledgebase.frame.work/expansion-card-functionality-on-framework-laptop-13-amd-ryzen-7040-series-SkrVx7gAh)
* Device or expansion card attached to the **Adjacent port** to the port that is having the issue.
1. USB-C
2. USB-C
3. USB-B
4. USB-B

The following are for Laptop 16 only. 
5. [ Fill In device / expansion card ]
6. [ Fill In device / expansion card ]

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [X ] No

## Describe the bug
After upgrading to BIOS 3.07 I'm unable to recognize the iGPU.  AMD adrenaline will not install/reinstall.

## Steps To Reproduce
Steps to reproduce the behavior:
1. I Updated to the beta BIOS 3.06 https://community.frame.work/t/framework-laptop-13-ryzen-7040-bios-3-06-release-beta/60732?utm_campaign=2024-11-21-fl13-amd-beta-software-update+%2801JD7XVD4RGY8RM8GX40QKXH3R%29&utm_medium=email&utm_source=Framework+Laptop+software+update+email+subscriptions+list&_kx=2NiDFe8m3o8ovB3yp1HJoTH66ZdzWVvfcW6-n3yBWkw.LNYsui
2. My iGPU no longer worked

## Expected behavior
my iGPU to be recognized by the system

## Screenshots
[image]

## Operating System (please complete the following information):
 - OS/Distribution: Windows 11 24H2

## Additional context
Add any other context about the problem here.

ALKahn10, 2024-11-21:
Looks like I fixed it by reenabling the drivers in windows and reinstalling the AMD Adrenalin drivers.

ALKahn10, 2024-11-21:
Looks like I fixed it by reenabling the drivers in windows and reinstalling the AMD Adrenelen drivers

tycho, 2024-12-04:
I'm running into similar problems. AMD graphics drivers report "code 43" in Device Manager after upgrading to BIOS 3.06 beta. Reinstalling drivers seems not to help.

The main reason I tried upgrading to 3.06 is because AMD 24.10 drivers don't seem to want to install properly when on 3.05 (stuck on black screen during install).

kiram9, 2024-12-05:
We have found several reports of the display driver not working when running 24.10 drivers from AMD WITH 64GB or more of RAM. 

We have reproduced this issue and are working with AMD to try to resolve this.

ALKahn10, 2024-12-05:
> We have found several reports of the display driver not working when running 24.10 drivers from AMD WITH 64GB or more of RAM.
> 
> We have reproduced this issue and are working with AMD to try to resolve this.

you sure?  After re-enabling the driver in windows it's seemed to work fine.  I upgraded my drivers today to 24.12.1 and it seems to work fine.

tycho, 2024-12-06:
> We have found several reports of the display driver not working when running 24.10 drivers from AMD WITH 64GB or more of RAM.
> 
> We have reproduced this issue and are working with AMD to try to resolve this.

Do you know of a workaround, or should I just stick with the AMD graphics driver in the official Framework driver package for the time being? Also, does the same issue exist in 24.12?

hawkejo, 2024-12-26:
> Do you know of a workaround, or should I just stick with the AMD graphics driver in the official Framework driver package for the time being? Also, does the same issue exist in 24.12?

That's been my experience with my FW 16 and 96 GB of RAM. I've read reports that others with 64 GB of RAM have been able to install the 24.12.1 driver successfully. I have had no such luck since 24.8.1.

aleksey-orekhov, 2025-01-05:
Same boat hear with FW 16 and 96 GB of ram with the 24.12.1 and 24.10.1 drivers. Lots of black screens. Gaming mode did not help. Upgrading bios from 3.03 to 3.05 did not help. 

The issue seemed to be primarily with the 780M, the RX 7700S appears to work fine with the newer drivers when I connect an external monitor directly to it using the port on the rear of the laptop. (As measured by running the game civilization 6 -> It runs smoothly on the external monitor, and very slowly on the laptop monitor, which I'm guessing goes through the 780M) Device manager also indicated an issue with the 780M but not the RX 7700S.

(I had tried doing multiple clean driver installs. using the amdcleanuputility.exe in safe mode, etc with no effect)

Downgrading to 24.8.1 AMD graphics drivers seems to have fixed the issue for now.

Thanks very much Framework team for working with AMD on a forward looking fix :)

J3tto, 2025-01-28:
Same here with my FW 16, 96GB RAM and RX 7700S, running 3.05. 

24.8.1 works without issues.

But anything later results in a solid white screen on boot for me. After a number of failed boots it finally starts with the driver disabled, which then allows me to uninstall and revert to 24.8.1. 

I also tried a full clean Win 11 instal on a spare SSD. same issue as soon as update past 24.8.1.

DrJKL, 2025-01-29:
+1

- FW16
- Windows 11 
- 96GB RAM
- 3.05 BIOS
- 7700S Expansion Card

Tried updating the driver to see if I could get ROCm working on WSL.
Twice cycled through [install](https://www.amd.com/en/resources/support-articles/release-notes/RN-RAD-WIN-24-12-1.html) &rarr; fail to boot properly &rarr; weep openly &rarr; safe mode &rarr; [DDU](https://www.wagnardsoft.com/display-driver-uninstaller-DDU-) &rarr; Try again
Gave up and went back to the [official bundle](https://knowledgebase.frame.work/framework-laptop-bios-releases-S1dMQt6F) for now.

deathlyrage, 2025-02-08:
Having this on both the FW16 and F13 on AMD with both on 96 GB of RAM

dattrax, 2025-02-15:
Any update from AMD on this?

hawkejo, 2025-02-19:
I've been able to reproduce this bug with AMD driver versions: 24.9.1, 24.10.1, 24.12.1, 25.1.1, 25.2.1, and PRO 24.Q2. The bug has been reproduced on a Framework 16 with 96 GB of RAM, with the graphics expansion bay and without it on Windows 11 24H2, _AND_ Windows 11 23H2.

Removing a stick of RAM to go down to 48 GB allows the driver to install and function normally. Installing the second stick after updating the driver (bringing the machine back to 96 GB) causes a black screen at boot with occasional artifacting.

CZEMacLeod, 2025-02-19:
For reference, I'm getting the same with any version (including inbox/WHQL windows update for win11 dev branch) of the drivers beyond the ones included with the framework drivers package.
Framework 16, 64GB ram, 7940HS/780M, 7700S, Windows 11 24H2 Insider Dev, BIOS 03.05.
Tried with and without 'gaming mode'.

I have to use the Windows Update Troubleshooting Pack `wushowhide.diagcab` to hide the AMD drivers each time a new WHQL driver is launched or windows automatically 'bricks' itself (Safemode, Bitlocker keys, Display Driver Uninstaller with turn off windows update mode and then reinstall the original driver required each time).

TBH, really wish you could just disable the 780M outright (at least when on external power) as there are a lot of glitches in the driver that works for it. (Artifacts, crashes, bad video decoding, etc.). 

I had really thought that AMD had got past the great hardware, poor software stage of its past, but apparently not.

I hope Framework launches an Intel based Framework 16 board in the future (with an Intel or NVidia dGPU) as I will _not_ be buying another AMD based one.

ALKahn10, 2025-02-19:
> For reference, I'm getting the same with any version (including inbox/WHQL windows update for win11 dev branch) of the drivers beyond the ones included with the framework drivers package. Framework [...]

Disabling outright for the Laptop 16 would be fine but this ticket was opened for the Laptop 13 which ONLY has a 780m

kiram9, 2025-02-27:
We got an updated driver 24.20.60 that fixes issues with the system not booting when using Windows 24H2 with over 64GB ram. We should be releasing this as a beta in the next few days. 
However we have been able to reproduce another bug, which is that resume from hibernate may result in a black screen. But there is a workaround to enabled gaming mode in the bios to prevent this. We will update this issue when we are ready to release the driver on our forum as a beta.

hawkejo, 2025-03-06:
I just managed to successfully install the 25.3.1 driver from AMD's website on my Framework 16 _without the dGPU._ Time will tell whether it'll be stable, but the fact that the system is actually running with the new driver makes me happy.

I'll probably shove the dGPU in the machine sometime to see if that changes anything.

OLRobert, 2025-03-06:
Dear @hawkejo , please keep us updated.

What is the RAM size of your machine?
It would be great, if it is above 64GB and everything works fine, even with the dGPU installed...

hawkejo, 2025-03-06:
I'm rocking 96 GB of Crucial RAM

OLRobert, 2025-03-06:
So does my machine.
Fingers crossed, the valley is crossed by installing the updated driver!

hawkejo, 2025-03-07:
So, I just shoved the dGPU in my Framework 16 (which was running the 25.3.1 driver). Again, the machine appears stable, which makes me happy. Given Framework's note on hibernate issues in one of the posts above, I'll try and test that later (I don't usually use hibernate, so it isn't an issue that really affects me).

Overall specs of my laptop, I'm running a Laptop 16 with the Ryzen 7 processor, 96 GB of RAM, and have tested this both with and without the dGPU (which implies that my sans-dGPU results should work for you Laptop 13 folks). I'll likely run some torture tests when I get home from work tonight just to see if I can crash the driver. Needless to say, if this issue is _finally_ resolved, it'll be nice to use Solidworks with the full allotment of RAM again.

deathlyrage, 2025-03-07:
So for my testing 25.3.1 i noticed that if the BIOS isnt set to 'gaming mode' it will still black screen but provided its in gaming mode its fine on FW 13. need to test FW 16 tho.

delsakov, 2025-03-08:
Tested FW13 (96/128GB) with 25.3.1. Hibernate works fine with "Auto" as well as "Gaming" mode.
Issue was with first few restarts though, while iGPU was auto-disabled. Reactivating and restarting helps.

_**Note**: Double boot broke hibernate for Win11 now. Previously (24.2.1) it was working fine._

dattrax, 2025-03-16:
I'm not sure I'm using hibernate.  I do have some random black screen issues when booting.  holding the power switch and trying again seems to get it back working again, so definitely an improvement from a month ago.  FW13 with 3.07BIOS and 96GB

wmigas, 2025-03-16:
> I'm not sure I'm using hibernate. I do have some random black screen issues when booting. holding the power switch and trying again seems to get it back working again, so definitely an improvement from a month ago. FW13 with 3.07BIOS and 96GB

have you seen Kiram's [comment](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/11#issuecomment-2686572135)? have you enabled gaming mode in bios. I'm using the latest drivers 25.3.1 and with gaming mode I do not have any issue with hibernate for the couple days (since I install these drivers). I have FW16 with 64GB

dattrax, 2025-03-16:
> have you seen Kiram's [comment](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/11#issuecomment-2686572135)? have you enabled gaming mode in bios. I'm using the latest [...]

I'm in gaming mode.  Its generally OK, but about 50% of the time it boots to a black screen.  Just a power cycle is enough to get it back going

delsakov, 2025-03-18:
> I'm in gaming mode. Its generally OK, but about 50% of the time it boots to a black screen. Just a power cycle is enough to get it back going

Confirming - have exactly that behavior on my wife's F13 (96GB) and on  my FW16 (128GB). Starting, getting black screen, rebooting, OK. If wait ~10-15 minutes, it started after black screen, but without ANY iGPU in Device Manager (_like it tries to initialize it and failed_).

delsakov, 2025-03-18:
> have you seen Kiram's [comment](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/11#issuecomment-2686572135)? have you enabled gaming mode in bios. I'm using the latest [...]

The issue is when using 96/128GB. With 64 it's OK.

dattrax, 2025-03-24:
My system auto installed 25.3.2 and its back to non-functional again.

Anyone else had this issue, or is it just me?

OLRobert, 2025-03-24:
It would be a god sign of Framework could get this situation solved.
More than 64GB RAM should not be the issue and they might be able fixing this by considering this for an upcoming BIOS update for FW 13/16.

There was romumours about a new driver pack...
I did not see a new one.

The message "the FW driver pack is just to get started, All new drivers will come from the manufacturers website." Does not work either...

Dear FW, please get this solved...

hawkejo, 2025-03-24:
I manually installed 25.3.2 on a clean Windows install on my Framework 16. Shockingly, I did _not_ have that issue with either the dGPU or the iGPU.

OLRobert, 2025-03-24:
So, where is the pattern?

Some FW 13 / 16 users are facing black screen issues, some don't.

Is it "just" the guys who installed more than 64GB RAM?

What is a working BIOS setup for 
RAM > 64 GB (So 96 or 128) and a working FW 13 plus a fully functional FW 16 without dGPU and with dGPU?

There must be a configuration thing...
(If it is not batch / generation limited, which would be a shame...)

hawkejo, 2025-03-24:
As I've noted before, I'm running 96 GB of RAM, so the issue is definitely inconsistent between devices.

kiram9, 2025-03-26:
We have an update from AMD this week, they have sent us a test driver which seems to resolve this issue, so we are waiting for their WHQL driver to be released in the next 2 weeks so we can release this as a beta driver update once they release to us.

OLRobert, 2025-04-03:
AMD offered / suggested installing  25.3.1 to my machine and it seems to be O.K. 
FR16, latest BIOS, Battery Extender is active and internal VGA is set to Gaming Mode (8GB RAM reserved). 

A new issue is a hanging login process if I do not log in "soon" after the login screen appears...

I am looking towards getting an official working driver package provided by framework and / or getting good support from AMD directly.

quinchou77, 2025-04-09:
New driver bundle release with the latest AMD graphic driver, which fixed the black/blank screen issue with more than 64GB of memory. You can get the driver from the forum below.

https://community.frame.work/t/framework-laptop-13-amd-ryzen-7040-series-driver-bundle-v2-04-2025-04-02-release-beta/67276

OLRobert, 2025-04-09:
That is great news. 
This seems to be a Beta / unofficial bundle, as it is not available on 
https://knowledgebase.frame.work/en_us/framework-laptop-16-bios-and-driver-releases-BkeqkVovp
or 
https://knowledgebase.frame.work/de/framework-laptop-16-bios-und-treiberversionen-BkeqkVovp

As I am using my FW16 as my "working device", I'll wait until that bundle gets "officially" uploaded.

quinchou77, 2025-04-11:
@OLRobert What is your system configuration? Both VGAs, you install Graphic module on the system?
The issue happened with the latest v2.04 beta or the original driver in the knowledgebase?

OLRobert, 2025-04-11:
Dear @quinchou77 , 
System Config is 
Both VGA installed. So integrated and additional VGA module are installed.
(Besides the situation of causing a strange fan curve, where the fan starts spinning loud even without having a high CPU, iGPU or dGPU, this setup works fine.)

Issue happened with AMD driver 25.3.1, AMD Chipset driver provided via AMD Software (Adrenaline¿) and the standard Windows 11 updates.

I realized issues in Zoom, where blueing the background or using virtual backgrounds was no longer possible. Disabling that feature made Zoom working again and now, javin just one VGA installed, the system is fine and Zoom works as designed.

I had to install the official drivers provided by Framework.

delsakov, 2025-04-12:
> This morning my FW16, 96GB RAM and "Game Mode" in Bios did no longer recognize both VGAs. Windows disabled them due to a misfunction.
> 
> Uninstalling all AMD Software, running DDU und shutting the system down, removing the dGPU, rebooting and re-installing the original but outdated FW Drivers from their homepage got the system back to normal.
> 
> I am really looking towards getting this situation fixed for FW13 & 16 and Systems with more than 64GB RAM....

I haveFW16 with eGPU (_NVidia in Razer Core_) and 128Gb (_Crucial_), but other then that - exact same issue. It's like 50/50 - will windows start or not... and if it failed few times, it disables Drivers and I have to unplug eGPU, restart few times, enable drivers, plug eGPU back.

Sometimes it's started with no issues and all working just great... until I reboot / restart - then it could work and could failed. Issue always happened on Win 11 starting (_hardware initialization?_).

dattrax, 2025-04-14:
The beta package is working better for me, however I’m seeing black screens in edge/chrome from some websites.

OLRobert, 2025-04-15:
Happy Tuesday.
I uninstalled the AMD drivers, as even without the dGPU the System acted strange.
Installing the old Framework drivers made the system running O.K. again.

Zoom still does not like blurred backgrounds but this might be a Zoom issue.

After all, I absolutely love the modularity of the FW 13/16 and the upcoming 12. 
The software seems to require further adjustment andy fingers are crossed swing this optimization in near future ....
FW13&16 were released some time back, so getting SW ready is an important step....

Companies want to have "working" machines. 
FW already offers great support for companies but the BIOS & Driver situation needs some enhancements.

Fingers crossed they don't forget about the "old" devices and just focus on the new / upcoming ones ....

new01, 2025-04-15:
Updated to 25.3.1, now my computer boots around 75% of the time. Sometimes, the machine boots up with both iGPU and dGPU disabled, but it would still output to the internal display, with reduced resolution. so I have to go into the device manager and manually re-enable them. For the iGPU, I usually have to enable/disable it twice in the device manager before it works properly. I do have gaming mode enabled in the BIOS. 

Specs:
Framework Laptop 16, 7840HS Mainboard
96GB RAM
RX 7700S Installed

delsakov, 2025-04-16:
> [@delsakov](https://github.com/delsakov) Have you used the new driver? Or, it is happened with the old driver?

Installed 25.3.1. Chipset and Video Driver.

delsakov, 2025-04-16:
> Happy Tuesday. I uninstalled the AMD drivers, as even without the dGPU the System acted strange. Installing the old Framework drivers made the system running O.K. again.
>
> Zoom still does not like [...]

totally agree. However I don't think the issue in BIOS as I have no issues in Ubuntu (have dual boot). All issues looks like at the moment Windows started initializing devices (I have Razer Huntsmen keyboard and it blinks at that exact moment and Win could start "good" (all devices available) or could start "bad" (Radeon iGPU disabled and issue with delay on login etc.) )
So I think the issue is with AMD chipset? driver... or graphics? drivers...

delsakov, 2025-04-16:
> Updated to 25.3.1, now my computer boots around 75% of the time. Sometimes, the machine boots up with both iGPU and dGPU disabled, but it would still output to the internal display, with reduced [...]

I decided to move back to 64Gb - all of those issues disappeared (_except black screen on login screen if left it for ~5 mins... where looks like system went to sleep and can't recover after_)

wmigas, 2025-04-16:
> I decided to move back to 64Gb - all of those issues disappeared (_except black screen on login screen if left it for ~5 mins... where looks like system went to sleep and can't recover after_)

can you try press Win + Ctrl + Shift + B when you have black login screen issue. It reset AMD drivers, maybe it will help. 
I have 64GB, only drivers 25.3.1 and everything works fine. I was considering increasing RAM to 96GB but I will wait until it will get stable

OLRobert, 2025-04-16:
Framework states the FW16 can be equipped with up to 64GB of RAM...
I guess, everything above is out of specs.
Is there a possibility of limiting the available RAM in Bios / UEFI to 64GB and back to "everything available"?
 
Spending something around 150€ for testing is a little high, as the device is company owned and I am a "tester".

We wanted to have devices with RAM over 64GB...

OLRobert, 2025-04-16:
I just checked the Framework compatibility list 
https://knowledgebase.frame.work/en_us/what-dram-memory-is-supported-by-framework-laptop-16-ryS2Xr3ch

And there is following message:
**Please note:** this is not officially supported by AMD on Ryzen 7040 Series

AMD states on their product page
https://www.amd.com/en/products/processors/laptop/ryzen/7000-series/amd-ryzen-9-7940hs.html

**System Memory Type**
DDR5 (FP7r2) , LPDDR5X (FP7-FP8)
**Memory Channels**
2
**Max. Memory**
256 GB
**System Memory Subtype**
UDIMM , SO-DIMM
""Max Memory Speed**
4x2R	DDR5-5600, LPDDR5x-7500
**ECC Support**
No

256GB of RAM seems to be the limit by using four double ranked 32GB modules...

OLRobert, 2025-04-25:
AMD AGESA ComboAM5PI 1.2.0.3c
got released and gets distributed via BIOS Updates.

According to
https://www.pcgameshardware.de/Mainboard-Hardware-154107/News/AMD-AGESA-Update-1-2-0-3-c-1471267/

It contains a fix for a [CVE](https://www.amd.com/en/resources/product-security/bulletin/amd-sb-7033.html) and more and faster DDR5 RAM:
"Diese adressiert insbesondere die kritische Sicherheitslücke CVE-2024-36347 alias "EntrySign", welche Ryzen- und Epyc-CPUs mit Zen 1 bis Zen 5 bedroht, und mehr schnellen DDR5-Arbeitsspeicher."

Fingers crossed this will reach Framework soon.

dattrax, 2025-05-07:
I'm on 25.4.1 with BIOS 3.07, it occasionally will black screen and not do the spinning circle for windows loading.  I just power cycle and start again and its fine the 2nd time.

OLRobert, 2025-05-08:
Good morning.
I updated to that version yesterday and am facing a black screen this morning.
Pressed the Power Button until it turned black.
Powered on again.
framework logo appears, Windows loading circle and login is available...

External screens are not recognized after that boot.

Device Manager shows VGA being disabled...

It is a disappointment 😞

OLRobert, 2025-05-08:
Good morning.
I yesterday updated the driver and the system did not boot up correctly and was stuck with a black screen.

Keeping the power button pressed until it was off, pushing it again made the framework logo and the Windows loading circle appear.

System came up but did not recognize the external screens.

VGA was disabled in device manager.
Activation caused AMD software recognizing an error and I submitted a report.

Rebooted the device via "reboot" of the start menu and it came up recognizing the external screens.

It would be great if there could be a fix...

quinchou77, 2025-05-09:
Driver 25.x.x is AMD's public driver, which we understand does not yet include the solution. Currently, the solution is available in driver 24.20.60.01, found in the latest driver bundle (v2.04). I recommend using this version. Please let us know if you encounter any issues, as we welcome your feedback.

OLRobert, 2025-05-09:
Same situation this morning 
System came up with black screen and fans winded up after some minutes.

Held the power button until the light was off.
Pushed it again, framework logo and Windows loading circle appeared, internal screen can be used, external screens are not recognized.

Logged on to Windows, checked Device Manager and had to see the GPU being disabled.

I removed all AMD Software, excerpt the Chipset drivers, DDUed (18.1.1.1 Clean and reboot) the AMD drivers and re-installed the old outdated framework drivers.

Interesting observation: after removing AMD software, which includes the VGA drivers, the VGA was activated and the external screens came up.

OLRobert, 2025-05-09:
Well, performing a last attempt of installing the new driver pack made the system showing a black screen for several minutes. 
Waiting patiently infront of the machine the screen came up.

The driver package got installed.

Did this take "forever" on your machines as well? 

The install script did not finish and I hit the Enter key, for sanity check.
Script finished and mentioned System will be restarted.
This was accepted and the reboot ended in a black screen with loud fans.

I powered the laptop off an on again.

Framework logo came up, windows circle as well, login appeared on all screens...

Let's see what the new package offers...

OLRobert, 2025-05-18:
Hi there.
I re-instaled the d-GPU and the Framework 16 started showing random short black screens while working. 

Did anyone's device show the same behaviour?