FrameworkComputer/SoftwareFirmwareIssueTracker #12: Suspiciously high power draw from the power supply during firmware update

brlin-tw, 2024-11-21:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [x] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION

3.05 → 3.06 BETA (`03.06`)

### DIY Edition information

Memory: ADATA AD5S560032G-B from Framework(32GiB*2)
Storage: Samsung SSD 970 PRO 512GB

### Port/Peripheral information

1. FW TYPE-C exp. card (empty)
2. FW TYPE-A exp. card (empty)
3. FW TYPE-C exp. card (connected to the SHARGEEK S100P 100W charger as power source)
4. FW DP exp. card (connected to Xiaomi G34WQi display)

### Standalone Operation

Is standalone mode enabled in the BIOS?

- [ ] Yes
- [x] No

## Describe the bug

It's not necessarily a problem, but I noticed that during the firmware update process, the laptop continuously draws 90W from the power supply with little variance in between. During the reboot right after the update, the power draw increased and stayed at 99W for a moment(~10s) before gradually lowering down.

Again, as the power draw is still within the power supply's specification and the battery charge capacity is indeed quite low, I wouldn't say this is a problem; the behavior is just peculiar enough for me to want to document it somewhere.

## Steps To Reproduce
Steps to reproduce the behavior:

1. Reduce battery charge capacity to around 30%
1. Connect the laptop to a 100W power supply with a wattage meter hooked up
1. Start the firmware upgrade and examine the power draw during the process.

## Expected behavior

Somewhat similar power draw pattern as in normal usage.

## Screenshots

I actually recorded most of the firmware upgrade process with my phone, will share the link here after it is uploaded to YouTube.

## Operating System (please complete the following information):

 - OS/Distribution: Ubuntu
 - Version: 24.04
- Linux Kernel Version: 6.8.0-48-generic

JohnAZoidberg, 2024-11-23:
Probably your battery was charging at around 60W + BIOS CPU frequency is pegged at max, which would be another 40+W.

UEFI firmware is not a proper OS, it does not have a scheduler or much CPU frequency control. So it's easy to run into these optimization issues.
It might be able to fix this, but it's not a priority because you need to have power connected anyways and the update process does not take very long.

closed 2024-11-23 by JohnAZoidberg

brlin-tw, 2024-11-23:
@JohnAZoidberg 

Thanks for the explanation, as long as it doesn't cause any problems I'm fine with that.