FrameworkComputer/SoftwareFirmwareIssueTracker #5: Linux "sudo lsusb -v"  fails

jcdutton, 2024-11-21:
## Device Information

### System Model or SKU

Please select one of the following
 [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
3.0.5

Windows:
N/A

Linux: 
Open a terminal and run the following command
```sudo dmidecode --string bios-version```
03.05

### DIY Edition information
Memory: Manufacture and SKU
Kingston Fury Impact: Part Number: KF556S40-32
2x making 64GB total.
Storage: Manufacture and SKU
Model Number:                       WD_BLACK SN850X 1000GB
Firmware Version:                   620361WD

### Port/Peripheral information
1. USB-C card, nothing plugged in.
2. Empty
3. Empty
4. Empty
5. USB-C card, FW16 PSU plugged in.
6. USB-A card, nothing plugged in.

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] No

## Describe the bug
The USB chips hang and reset themselves. I suspect USB chip firmware bugs.
EC console shows TIMEOUT and RESET messages.

In a Linux terminal window:
"sudo lsusb -v"
It hangs at the line:
Bus 002 Device 002: ID 05e3:0625 Genesys Logic, Inc. USB3.2 Hub
The EC then detects a timeout and resets the Genesys chip.

## Steps To Reproduce
Steps to reproduce the behavior:
In a Linux terminal window:
"sudo lsusb -v"
It hangs at the line:
Bus 002 Device 002: ID 05e3:0625 Genesys Logic, Inc. USB3.2 Hub

## Expected behavior
In a Linux terminal window:
"sudo lsusb -v"
It should continue to display all devices immediately and not hang.

## Screenshots
N/A

## Operating System (please complete the following information):
 - OS/Distribution: Linux/Ubundu
 - Version: 24.04
- Linux Kernel Version: `uname -a`  6.12.0  <- Mainline compiled kernel.

## Additional context
Add any other context about the problem here.

sinatosk, 2024-11-21:
I've experienced this with 03.04 too

JohnAZoidberg, 2024-11-22:
> The USB chips hang and reset themselves. I suspect USB chip firmware bugs.
EC console shows TIMEOUT and RESET messages.
>
>In a Linux terminal window:
"sudo lsusb -v"
It hangs at the line:
Bus 002 Device 002: ID 05e3:0625 Genesys Logic, Inc. USB3.2 Hub
The EC then detects a timeout and resets the Genesys chip.

Could you please post a full log of EC console and lsusb?

jcdutton, 2024-11-22:
In FW16 BIOS 3.0.3 there was EC console output when the "sudo lsusb -v" was used.
Now with FW16 BIOS 3.0.5 there is no each EC console output when the "sudo lsusb -v" was used.
I cannot downgrade to BIOS 3.0.3 to prove this as BIOS 3.0.5 is not downgradable.
Does anyone know what EC command I need to use to turn the USB chip resets debug back on?
The "sudo lsusb -v" still hangs for a while (about 60 seconds) before skipping the Genesys Logic part and continuing down the other components.
This problem should be 100% reproducible where you are in Linux.

jcdutton, 2024-11-22:
[lsusb-v.txt](https://github.com/user-attachments/files/17874126/lsusb-v.txt)
Interestingly, after the "sudo lsusb -v" the "Bus 002 Device 002: ID 05e3:0625 Genesys Logic, Inc. USB3.2 Hub" disappears permanently from "lsusb" output, until one power cycles the laptop.
No useful output on EC console.

JohnAZoidberg, 2024-11-23:
>  The EC then detects a timeout and resets the Genesys chip.

Why do you think this is happening?

> Port/Peripheral information
> 2. Empty
> 3. Empty
> 4. Empty

This hub is exactly connected to these three ports.
To make sure we align here: 2 is middle right, 3 is lower right, 4 is lower left.
The hub is also connected to the dGPU interposer. Have you got a dGPU installed?

Can you try to see if these ports are working for you? And whether lsusb is fine when something is connected vs nothing connected?

jcdutton, 2024-11-23:
@JohnAZoidberg 
I would be happier if you has a FW16 AMD and confirmed that you are able to reproduce the problem I am seeing.
If you follow the 5 easy steps to reproduce it, you should see the problem for yourself.
If you cannot reproduce the problem, I will probably need to RMA the mainboard as it is unique to me then.

jcdutton, 2024-11-23:
With all slot cards removed, no PSU power. No usb devices plugged into the slots.
Note the problem is also present if tested with the laptop keyboard removed.

1. Power off laptop.
2. Power on laptop.
3. Login to Ubuntu 24.04
4. Open terminal windows.
5.
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 005: ID 0e8d:e616 MediaTek Inc. Wireless_Device
Bus 001 Device 006: ID 32ac:0018 Framework Laptop 16 Keyboard Module - ISO
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 05e3:0625 Genesys Logic, Inc. USB3.2 Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

6.
sudo lsusb -v
....
Hangs at "Bus 002 Device 002: ID 05e3:0625 Genesys Logic, Inc. USB3.2 Hub" for about 60 seconds.

7.
lsusb
Initially might hang for a while also.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 005: ID 0e8d:e616 MediaTek Inc. Wireless_Device
Bus 001 Device 006: ID 32ac:0018 Framework Laptop 16 Keyboard Module - ISO
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

8.
Notice the line missing from lsusb:
"Bus 002 Device 002: ID 05e3:0625 Genesys Logic, Inc. USB3.2 Hub"
It stays missing until you reboot the laptop.

It is this disappearance of the Genesys Logic USB 3.2 hub that makes me believe that there is a bug in the firmware of the Genesys Logic device.

amstan, 2024-11-24:
```
Bus 002 Device 002: ID 05e3:0625 Genesys Logic, Inc. USB3.2 Hub
.... long wait...
Couldn't open device, some information will be missing
```

```
[114843.017542] xhci_hcd 0000:c4:00.3: Timeout while waiting for setup device command
[114848.350153] xhci_hcd 0000:c4:00.3: Timeout while waiting for setup device command
[114848.553427] usb 2-2: device not accepting address 5, error -62
[114853.896731] xhci_hcd 0000:c4:00.3: Timeout while waiting for setup device command
[114859.228956] xhci_hcd 0000:c4:00.3: Timeout while waiting for setup device command
[114859.432260] usb 2-2: device not accepting address 6, error -62
[114859.440639] usb usb2-port2: unable to enumerate USB device
```

I can reproduce as well. Thanks for the bug, I'll keep investigating.

kiram9, 2024-12-19:
Also can reproduce this. We will work with our vendor to see if they can fix this. We are going to continue to track this internally. 

This hub is internally connected to the bottom left, and mid/bottom right ports only.

ektus, 2025-03-17:
three month later, the problem is still there, tested  with OpenSuse Tumbleweed (kernel 6.13.6-1-default)
`Bus 002 Device 002: ID 05e3:0625 Genesys Logic, Inc. USB3.2 Hub` goes missing after `sudo lsusb -v`

ektus, 2026-01-18:
Likewise with Linux kernel 6.18.5-1.1 and latest BIOS 4.03