FrameworkComputer/SoftwareFirmwareIssueTracker #7: BIOS 3.05 Beta Install Failure

corndog2000, 2024-11-21:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [x] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
3.05 Beta

Windows:
Open a command terminal as administrator:
```wmic bios get SMBIOSBIOSVERSION```

```
SMBIOSBIOSVersion
03.05
```
### DIY Edition information
If you are experiencing an issue on a DIY system, Please also fill out the memory and storage devices you are using. 

Memory: Manufacture and SKU
[Kingston FURY Impact 64GB (2x32GB) 5600MT/s DDR5 CL40 Laptop Memory Kit of 2 | Lower Power Comsumption | Intel XMP 3.0 | Plug N Play | KF556S40IBK2-64](https://www.amazon.com/Kingston-Impact-2x32GB-Comsumption-KF556S40IBK2-64/dp/B0BRTJT5P2)

Storage: Manufacture and SKU
[SABRENT Rocket 2230 NVMe 4.0 1TB High Performance PCIe 4.0 M.2 2230 SSD [SB-2130-1TB]](https://www.amazon.com/SABRENT-Rocket-2230-Performance-SB-2130-1TB/dp/B0BQG6JCRP)

### Port/Peripheral information
If you are experiencing an issue with a peripheral or an expansion card/port please fill out the following information: 
* Peripheral vendor and name.
* Port the Peripheral was connected to. Please see the following for port numbering: [Laptop 16](https://knowledgebase.frame.work/expansion-card-slot-functionality-on-framework-laptop-16-rkUjGm7cn) or [Laptop 13](https://knowledgebase.frame.work/expansion-card-functionality-on-framework-laptop-13-amd-ryzen-7040-series-SkrVx7gAh)
* Device or expansion card attached to the **Adjacent port** to the port that is having the issue.
1. USB-C: Anker 568 USB-C docking station
2. HDMI : Empty
3. USB-A: Empty
4. USB-C: Razer Core X eGPU dock with Nvidia GTX 1070 GPU connected to Displayport monitor.

The following are for Laptop 16 only. 
5. USB-C: Pixel Watch 2 charger.
6. USB-A: Empty

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug
I tried installing the BIOS update and it got to about 40% on the screen with the spinning framework logo. Then the whole screen went black. After 15 minutes there had been no change. I force powered off the laptop with the power button. The laptop thankfully did boot again. I ran the BIOS update .exe again from Windows 11 and the second time the GUI update screen did get to 100% and said “Update completed”.

## Steps To Reproduce
Steps to reproduce the behavior:
1. Run BIOS 3.05 Beta installer from Windows 11 Pro as admin.

## Expected behavior
First time BIOS update runs it completes and reboots computer.

## Screenshots
Black screen on both the laptop monitor and the display connected to the eGPU.

## Operating System (please complete the following information):
 - OS/Distribution: Windows 11 Pro
 - Version: 24H2

corndog2000, 2024-11-21:
I’m also experiencing known issue #3. I’ve unplugged all devices and reconnected them without success. I’ve also powered off the laptop and unplugged the devices for a few minutes still no luck restoring USB3 functionality.

kiram9, 2024-11-21:
Just as a note, You were able to resolve known issue 3 by removing your dock, which was powering the laptop.

corndog2000, 2024-11-21:
That resolved known issue 3 but not by following the steps provided in the original forum post. Just removing the dock from the laptop did not fix the USB-C port/Dock. I had to unplug the dock and then power cycle the dock while it was unplugged.