FrameworkComputer/SoftwareFirmwareIssueTracker #14: fwupdmgr cannnot find new UEFI update 03.05 for Framework 16 Laptop in Linux

lpapadakos, 2024-11-22:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ X] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
Vendor: INSYDE Corp
Version: 03.03
Release Date: 03/27/2024
Address: 0xE0000
product: AMD Ryzen 7 7840HS w/ Radeon 780M Graphics
vendor: Advanced Micro Devices [AMD]
physical id: 4
bus info: cpu@0
version: 25.116.1

### DIY Edition information
If you are experiencing an issue on a DIY system, Please also fill out the memory and storage devices you are using. 

Yes this is a DIY Edition laptop

Memory: Crucial CT2K16G56C46S5 2x16GB
Storage: Lexar NM790 4TB

### Port/Peripheral information

Unrelated to USB peripherals or expansion cards.

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [X] No

## Describe the bug

I got an email about a BIOS update (that's pretty cool, never seen that with other OEMs) an it directed me to this community post:
https://community.frame.work/t/framework-laptop-16-ryzen-7040-bios-3-05-release-beta/60735

Naturally I quickly ran in my terminal:

```
fwupdmgr refresh
fwupdmgr get-updates
```

But there were no updates.

Alright so it turns out we need to enable `lvfs-testing`:

Let's do that:
```
fwupdmgr enable-remote lvfs-testing
```

Then I ran
```
fwupdmgr refresh -- force
fwupdmgr get-updates
```

I can see that both `lvfs` and `lvfs-testing` are being downloaded, but alas, still no updates.

Let's run them with sudo:

```
sudo fwupdmgr refresh -- force
sudo fwupdmgr get-updates
```

Now I still see no UEFI updates, but I got one for the fingerprint sensor! Alright, I installed it with `sudo fwupdgr update`

Let's try a UI solution.
I opened `gnome-firmware`

Under the `System Firmware`  device it's correctly listing the two versions available in testing, 03.04 and 03.05.
(Though it -probably intentionally- only lists them if AC is connected)
Success!

At that point I retry the fwupdmgr commands with and without sudo, but I still see 0 updates.

I completed the installation though gnome-firmware and it went smoothly, the UEFI update capsule was put in /EFI/UpdateCapsule and it ran on reboot (gnome-firmware prompted me to reboot)

The bug is that fwupdmgr would NOT show the UEFI update not matter what I did.
And also it is weird that the fingreprint sensor update only showed up with sudo.

## Steps To Reproduce

Steps to reproduce the behavior:
1. Have a Framework 16 Laptop
2. Make sure you have `lvfs-testing` enabled
3. Run the  commands (refresh -> get-updates)
4. Observe there are no updates

## Expected behavior

I expected the update to show up as it did in `gnome-firmware`.

## Operating System (please complete the following information):

Up-to-date Arch Linux with kernel 6.12.0

`fwupdmgr --version | grep 'org.freedesktop.fwupd'`

```
Idle…: 0%
compile org.freedesktop.fwupd 2.0.1
runtime org.freedesktop.fwupd-efi 1.7
runtime org.freedesktop.fwupd 2.0.1
```

ctsdownloads, 2024-11-22:
I replicated this environment for testing, had no problem getting from 3.04 to 3.05:

- Arch Linux

- fwupd 2.0.1

- fwupdmgr enable-remote lvfs-testing

- fwupdmgr refresh --force

- fwupdmgr get-updates

- fwupdmgr update

- Follow the prompts, reboot. 

In your instance, you're on 3.03 going to 3.04 and then to 3.05, but this would work the same ideally.

JohnAZoidberg, 2024-11-23:
> Though it -probably intentionally- only lists them if AC is connected

Yes, that's what fwupdmgr does.
Was your battery fully charged? There's an issue that when the system is fully charged, it doesn't charge anymore, so fwupdmgr doesn't want to show updates or update.

lpapadakos, 2024-11-23:
The battery was really low and when I started this, the battery was around 30-31% and charging

ctsdownloads, 2024-12-06:
> The battery was really low befire al lthis, and by the time I started the commands, the battery was around 30-31% and charging

In my testing, I was about 80% battery, AC connected.

b4shful, 2024-12-07:
By the way if anyone reading this is on 3.04 beta and fwupdmgr isn't finding the update to 3.05 (it was moved to stable yesterday), try disabling the `lvfs-testing` remote with `fwupdmgr disable-remote lvfs-testing`.

On my machine it couldn't find 3.05 stable until I removed the `lvfs-testing` remote, but immediately after doing that it was able to find and install the update from 3.04 beta to 3.05 stable without issues.

JohnAZoidberg, 2025-02-20:
> On my machine it couldn't find 3.05 stable until I removed the lvfs-testing remote, but immediately after doing that it was able to find and install the update from 3.04 beta to 3.05 stable without issues.

Hmm that might be a bug in fwupd then.

Maybe you saw this issue https://github.com/fwupd/fwupd/issues/8243
Or a similar one?
A couple issues like this were included in recent releases.

Anybody still seeing this issue with fwupd version 2.0.4 or higher?

JohnAZoidberg, 2025-04-22:
Closing this, as it was due to a buggy fwupd version.

closed 2025-04-22 by JohnAZoidberg