FrameworkComputer/SoftwareFirmwareIssueTracker #15: virtual machine failure on bios 3.0.6 beta

learningandmemorylab, 2024-11-22:
AMD 13" DIY
Virtualbox worked fine before upgrading to the 3.0.6.beta BIOS. The error message is:

VirtualBox can't enable the AMD-V extension. Please disable the KVM kernel extension, recompile your kernel and reboot (VERR_SVM_IN_USE).
Result Code:
NS_ERROR_FAILURE (0x80004005)
Component:
ConsoleWrap
Interface:
IConsole {6ac83d89-6ee7-4e33-8ae6-b257b2e81be8}

I tried reinstalling 2 versions of vbox, tested several VMs, three kernels (6.11.x, 6.12.x, 6.5.x) all to no avail. cat /proc/cpu lists "svm" for each CPU.

jcdutton, 2024-11-22:
Are you running kvm or docker also?  You might need to rmmod kvm and stop any docker instances.

JohnAZoidberg, 2024-11-23:
As @jcdutton indicated (thanks!) it might be a configuration issue.

As your error message says "Please disable the KVM kernel extension, recompile your kernel and reboot".
Have you tried that?

brlin-tw, 2024-11-23:
@learningandmemorylab 

> VirtualBox can't enable the AMD-V extension. Please disable the KVM kernel extension, recompile your kernel and reboot (VERR_SVM_IN_USE).

This error message seems to indicate that the hardware virtualization support is being used by another application(e.g. KVM or VMware workstation), please ensure none of those are running and try to reproduce this issue.

You can also try booting into a live system and reproduce this problem as it won't have these kinds of things installed.

learningandmemorylab, 2024-11-23:
Thank you all for your good advice.  rmmod kvm_amd let the VMs work again. Superm1's advice was right on too about the change in 6.12 behavior.

closed 2024-12-02 by JohnAZoidberg