FrameworkComputer/SoftwareFirmwareIssueTracker #16: Framework 16 BIOS 3.0.5 Power Profiles with DGPU Possibly Incorrect

howvenger, 2024-11-25:
## Device Information

### System Model or SKU
Framework 16 with Ryzen 7 7840HS
Radeon RX 7700S

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [x] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
03.05

### DIY Edition information
If you are experiencing an issue on a DIY system, Please also fill out the memory and storage devices you are using. 

Memory: Crucial DDR5 5600MHz 32GB (2x16) CT2K16G56C46S5
Storage: Crucial T500 CT2000T500SSD8 2TB/Sabrent Rocket Q4 2230 2TB SB-213Q-2TB

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug
Updated BIOS to 3.0.5 and began noticing strange power issues with dGPU. Didn't verify CPU yet. But the dGPU is boosting to almost 100 watts even in balanced. Before it would stay at 80 watts when in balanced mode. Confirmed when booting into windows and using Adrenalin software, total tdp remains unchanged between performance and balanced mode for APU/dGPU tdp.

## Steps To Reproduce
Steps to reproduce the behavior:
1. Connect 180W charger to laptop
2. Enable either MangoHUD or use Adrenalin software. Not official from Framework, but driver from AMD.
3. TDP will boost to 90 watts + on balanced mode.

## Expected behavior
TDP for dGPU should remain at 80 Watts

## Screenshots
If applicable, add screenshots to help explain your problem.

## Operating System (please complete the following information):
 - OS/Distribution: Windows 11 Professional/Fedora 41
- Linux Kernel Version: Linux yuno 6.11.8-300.fc41.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Nov 14 20:37:39 UTC 2024 x86_64 GNU/Linux

howvenger, 2024-11-25:
Confirmed that CPU TDP works as expected and power saving puts it at a max of ~30W. dGPU TDP set to ~80 watts

knipp30, 2024-12-21:
I had thought this was a change from PPD to Tuned - but yes, I have the same issue.