FrameworkComputer/SoftwareFirmwareIssueTracker #22: FW16 won't boot if Lenovo TB dock is connected.

henris42, 2024-12-09:
FW16 won't boot if Lenovo TB dock is connected. Nothing happens, only power button light turns on. After a while fan might spin up.
If you hit again the power button it turn off immediately.

Nothing happens means that no screen starts, no backlight, no any messages - no memory test, no logo. Just blank. Does not matter if lid is open or closed. This does not seem to be OS related. (?)

If you unplug the dock it works fine.

Lenovo dock is TB4 Universal Dock. (40B00135EU) It works perfectly fine if you plug it in after the FW logo shows.

Tried this again today with 3.05 BIOS, as bad as ever.

This is extremely annoying, as you cannot reboot the OS without pulling out the dock cable. OS in Win11 Pro 23H2, if that matters.

Spunkie, 2024-12-13:
I've been seeing the same thing on a fw16(`03.03` bios) with a [monoprice usb-c dock](https://www.monoprice.com/product?p_id=44465) but it only started happening after the most recently windows 11 pro 24H2 update.

Spunkie, 2024-12-13:
Potentially related community forum post: https://community.frame.work/t/windows-update-24h2-issue-docking-station-and-external-usb-device-problems/60708

jameshartig, 2025-01-27:
Hmm, I'm not having this issue at all and I have the same dock. What do you have plugged into your dock? What FW is the dock?

quinchou77, 2025-02-13:
> Hmm, I'm not having this issue at all and I have the same dock. What do you have plugged into your dock? What FW is the dock?

@jameshartig Do you also use Framework Laptop 16 (AMD Ryzen™ 7040 Series)? What is the BIOS version on your system?