FrameworkComputer/SoftwareFirmwareIssueTracker #23: Framework Laptop 13 AMD 7040 BIOS 3.06 iGPU setting issue

kennebnonedz, 2024-12-11:
## Device Information
Framework Laptop 13 (AMD Ryzen™ 7040 Series) DIY 64GB RAM + 2.8k screen
Windows 11 Pro 23H2
AMD GPU driver 24.12.1

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [x] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
3.06 BETA

### DIY Edition information
Memory: Kingston FURY 64GB (2x32GB) 5600MHz CL40 Impact
Storage: Samsung 990 PRO 4TB

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug
BIOS description of iGPU setting seems incorrect/incorrect amount of vRAM assigned

Installed BIOS 3.06 BETA having GAMING mode iGPU config enabled; on the first reboot screen went white but after a while and a forced reboot (press and hold power button for a few seconds) Windows booted fine
Windows 11 shows AMD Radeon(TM) 780M has 8GB vRAM (as opposed to 4GB stated in bios description and to what I had while on 3.05)

When I try to switch to AUTO iGPU mode, windows boots up with a white screen (boot up jingle is audible so it's a video mishap), reboot does not help

## Steps To Reproduce
Steps to reproduce the behavior:
1. Go to BIOS advanced settings
2. Set iGPU mode to AUTO
3. Save and exits
4. See Windows 11 not booting correctly

## Expected behavior
Windows 11 boots normally 

## Operating System (please complete the following information):
 - Windows 11 Pro
 - Version: 23H2

quinchou77, 2025-04-08:
This issue is likely related to the AMD graphics problem of blank/black screens with more than 64GB of memory.
AMD has released a new driver with the fixed and we will release a new driver bundle v2.04 beta this week, which includes the new AMD graphic driver..

quinchou77, 2025-04-09:
There is the new driver bundle release with the latest AMD graphic driver, which fixed the black/blank screen issue with more than 64GB of memory. You can get the driver from the forum below.

https://community.frame.work/t/framework-laptop-13-amd-ryzen-7040-series-driver-bundle-v2-04-2025-04-02-release-beta/67276

quinchou77, 2025-12-30:
We have transitioned to the AMD Public Driver version for this release. Please update to the latest public driver from AMD's website  or download the latest driver bundle from the [knowledge article](https://knowledgebase.frame.work/zh_tw/framework-laptop-bios-and-driver-releases-(amd-ryzentm-7040-series)-r1rXGVL16), and verify if the issue persists. This move ensures better long-term support and alignment with AMD's latest software optimizations.