FrameworkComputer/SoftwareFirmwareIssueTracker #25: Freezes requiring reboots and general flakyness on dGPU Access after 03.05 upgrade

Popkornium18, 2024-12-11:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [x] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
03.05

Windows:
Open a command terminal as administrator:
```wmic bios get SMBIOSBIOSVERSION```

Linux: 
Open a terminal and run the following command
```sudo dmidecode --string bios-version```

### DIY Edition information
If you are experiencing an issue on a DIY system, Please also fill out the memory and storage devices you are using. 

Memory:

```
  *-memory
       description: System Memory
       physical id: 15
       slot: System board or motherboard
       size: 32GiB
     *-bank:0
          description: SODIMM Synchronous Unbuffered (Unregistered) 5600 MHz (0,2 ns)
          product: KF556S40-16
          vendor: Kingston
          physical id: 0
          serial: FF066189
          slot: DIMM 0
          size: 16GiB
          width: 64 bits
          clock: 1305MHz (0.8ns)
     *-bank:1
          description: SODIMM Synchronous Unbuffered (Unregistered) 5600 MHz (0,2 ns)
          product: KF556S40-16
          vendor: Kingston
          physical id: 1
          serial: FF0660EC
          slot: DIMM 0
          size: 16GiB
          width: 64 bits
          clock: 1305MHz (0.8ns)
```
Storage:
```
           *-nvme
                description: NVMe device
                product: WD_BLACK SN850X 2000GB
                vendor: Sandisk Corp
                physical id: 0
                bus info: pci@0000:05:00.0
                logical name: /dev/nvme0
                version: 620361WD
                serial: 24021W802373
                width: 64 bits
                clock: 33MHz
                capabilities: nvme pm msi msix pciexpress nvm_express bus_master cap_list
[... 219 lines cut ...]
Dez 11 21:50:10 deepthought kernel:  drm_client_modeset_commit+0x25/0x40
Dez 11 21:50:10 deepthought kernel:  __drm_fb_helper_restore_fbdev_mode_unlocked+0x85/0xd0
Dez 11 21:50:10 deepthought kernel:  drm_fb_helper_set_par+0x30/0x40
Dez 11 21:50:10 deepthought kernel:  fbcon_init+0x2c6/0x660
Dez 11 21:50:10 deepthought kernel:  visual_init+0xea/0x180
Dez 11 21:50:10 deepthought kernel:  do_bind_con_driver.isra.0+0x241/0x3c0
Dez 11 21:50:10 deepthought kernel:  ? fbcon_startup+0x202/0x2e0
Dez 11 21:50:10 deepthought kernel:  do_take_over_console+0x17b/0x1f0
Dez 11 21:50:10 deepthought kernel:  ? __schedule+0x434/0x1660
Dez 11 21:50:10 deepthought kernel:  do_fbcon_takeover+0x80/0x150
Dez 11 21:50:10 deepthought kernel:  fbcon_register_existing_fbs+0x53/0xb0
Dez 11 21:50:10 deepthought kernel:  process_one_work+0x176/0x330
Dez 11 21:50:10 deepthought kernel:  worker_thread+0x252/0x390
Dez 11 21:50:10 deepthought kernel:  ? __pfx_worker_thread+0x10/0x10
Dez 11 21:50:10 deepthought kernel:  kthread+0xcf/0x100
Dez 11 21:50:10 deepthought kernel:  ? __pfx_kthread+0x10/0x10
Dez 11 21:50:10 deepthought kernel:  ret_from_fork+0x31/0x50
Dez 11 21:50:10 deepthought kernel:  ? __pfx_kthread+0x10/0x10
Dez 11 21:50:10 deepthought kernel:  ret_from_fork_asm+0x1a/0x30
Dez 11 21:50:10 deepthought kernel:  </TASK>
Dez 11 21:50:10 deepthought kernel: ---[ end trace 0000000000000000 ]---
Dez 11 21:50:10 deepthought kernel: Console: switching to colour frame buffer device 320x100
```

</details>

Popkornium18, 2024-12-11:
Two important realizations:

1. This does not seem to be dGPU related, as the freeze also happens without passing `DRI_PRIME=...`. 
2. It seems to be fullscreen related, as I get control back after force quitting fullscreen via kwin. The content is flickering very quickly, though.

kiram9, 2024-12-11:
I have sent this to review for our internal team to try to replicate.

Popkornium18, 2024-12-12:
I tried it today on a Ubuntu 24.04 live system and had no issues. It does seem to be also related to the kernel version. Somone on the [Framework forum](https://community.frame.work/t/fullscreen-games-freeze-on-plasma-6-with-dgpu/61088/18) reported very similar issues to mine. They did not mention firmware though.

Popkornium18, 2024-12-15:
Setting `amdgpu.dcdebugmask=0x400` in the Kernel cmdline fixes this or rather works around this. This means that this is not a firmware issue and me noticing it only after the firmware upgrade was just bad timing.

closed 2024-12-15 by Popkornium18