FrameworkComputer/SoftwareFirmwareIssueTracker #27: Laptop Issue

Obasav, 2024-12-14:
## Device Information

### System Model or SKU 

Framework 16 7940HS + 7700S

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [x] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION 3.05

Windows:
Open a command terminal as administrator:
```wmic bios get SMBIOSBIOSVERSION```

Linux: 
Open a terminal and run the following command
```sudo dmidecode --string bios-version```

### DIY Edition information
If you are experiencing an issue on a DIY system, Please also fill out the memory and storage devices you are using. 

Memory: Kingston KF556F40-32 x2
5600 MT, CL 40 1.1v 64 GB (2x32 GB)
Storage: 2280: Solidigm P44 Pro 2TB
2230: Solidigm P41 Plus 1TB

### Port/Peripheral information
If you are experiencing an issue with a peripheral or an expansion card/port please fill out the following information: 
* Peripheral vendor and name.
* Port the Peripheral was connected to. Please see the following for port numbering: [Laptop 16](https://knowledgebase.frame.work/expansion-card-slot-functionality-on-framework-laptop-16-rkUjGm7cn) or [Laptop 13](https://knowledgebase.frame.work/expansion-card-functionality-on-framework-laptop-13-amd-ryzen-7040-series-SkrVx7gAh)
* Device or expansion card attached to the **Adjacent port** to the port that is having the issue.

1. USB-C 
2. USB-A
3. USB-A
4. USB-C (with delta 240W PD 3.1 charger connected)

The following are for Laptop 16 only. 
5. USB-A
6. USB-A

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [X] No

## Describe the bug
DPU isn't using full 100W consistently with Delta 240W PD 3.1 charger. It will briefly use it when battery is full, but when it's charging/discharging it doesn't reach the 100W limit.

## Steps To Reproduce
Steps to reproduce the behavior:
1. Discharge framework 16 battery and open a GPU intensive game. I used total war warhammer 3 (proton experimental)

2. Open a monitor software and look how much power DGPU uses while charging.

3. When DGPU has hit charging cap, look at average and see it has increased. It will potentially hit 100W for a brief moment and then drop back down to around 75W when the battery changes charging state.

## Expected behavior
While laptop is charging and connected to 240W PD 3.1 charger, more power should be available to GPU to let it hit 100W and stay there.

## Screenshots
[image]

Image 1 is laptop gaming while battery is from 80% dropping to 70% and system is unplugged.

[image]

Image 2 is system plugged in to 240W delta PD 3.1 charger, gaming, while battery is recharging from 70% to 80%

[image]

Image 3 is system charged to 80% while connected to the 240W delta charger.

## Operating System (please complete the following information):
 - OS/Distribution: Ubuntu
 - Version: 24.10
- Linux Kernel Version: `6.13-rc2`

## Additional context
Add any other context about the problem here.

wizzlebippi, 2024-12-17:
I can recreate this on windows.  With the Framework 180W PSU, I can consistently run the dGPU at 80-90W.  On the Delta 240W PSU, the dGPU is constantly throttling at 30W.  

Framework 16
Ryzen 7 7840HS
Radeon RX 7700S
64GB RAM
2TB SSD
Windows 11 24H2
BIOS 3.05
2024-10-2 driver pack installed

Originally had BIOS configured to limit battery charge to 90%.  Suspected that BIOS might be limiting dGPU power draw to charge the battery, but since the battery could never reach fully charged my performance suffered.  Set max charge to 100%, disabled adaptive battery saver, no change.  Only switching to OEM 180W PSU with Balanced power mode seems to allow dGPU to run at up to 100W.

kiram9, 2024-12-19:
Thanks! We will investigate this and try to fix in the next release.

knipp30, 2024-12-20:
@kiram9 - Same issue as OP - when below 90% I notice EXTREME frame pacing issues. My dGPU shows its using 75-100w (in fluctuations) and I think that is why the frame pacing is happening. I tried with different GPU power modes - but automatic is the most stable - as the other options are worse.

Same bios - 3.05

If any logging is needed, please let me know!

knipp30, 2024-12-20:
Video of the frame pacing happening when playing a game.
https://drive.google.com/file/d/1bw0l7X3jTmHzgbyk40W47PSKG0ax3BaQ/view?usp=sharing

psierra117, 2024-12-20:
I wrote about that issue first on 12/1/24 in the Framework Forum and tried a Fault analyzes with overwriting Smartshift and STT/stapm Values with Smokeless_UMAF and even tried reinstalling Windows and installing bazzite OS. It doesn't depend on the installed OS. [https://community.frame.work/t/framework-laptop-16-ryzen-7040-bios-3-05-release/60735/70?u=psierra117](url) I Suspect some recent changes to the EC Configs to be responsible, as the STT/Stapm Settings do not correspond to any changes I made with Smokeless_UMAF so the Bios/Uefi is not controlling the Power Management. And also many People already have told, that with 3.05 the Framework 16 is draining battery in "Balanced" Mode on the 180w Charger which was not happening in 3.04 or 3.03. So changes introduced in 3.05 seem to be responsible.

knipp30, 2025-01-31:
To add to this. as it seems that the issue still relates to power negation

when using the 240w charger, when gaming in performance or balanced mode, if the battery discharges below 100%, the games start frame pacing issue I shared in the video above.

Right now, to game anything close to consistent, I have to game in power save mode.

Obasav, 2025-02-12:
Any update to the progress of resolving this?

I have some more information:

My current system configuration is set up with Linux mint 22.1 using kernel 6.14-rc2. 

The issue still persists with the battery disconnected from the system while it is in performance mode.

[image]
[image]
[image]

Without the battery, it also exhibits the same issue in the balanced power mode:

[image]
[image]

In power saver mode, with the battery disconnected, the frame rate is significantly better averaged and the GPU memory clock speed doesn't change for the entire duration of the benchmarch, only the SCLK.

[image]
[image]
[image]
[image]

SepticFuddy, 2025-02-15:
After a lot of frustration with throttling on the 180W charger I got myself a Delta 240W. In Balanced mode, I can go for 2-3hrs in a demanding game until suddenly performance/GPU power throttles and the battery is at 95%. Whenever I check the battery earlier than that it reads 100%, so it's as if it's just deciding to let the battery drain 5% at some point and then the GPU power throttles as a result until it's recharged. Far from ideal when you're in the middle of playing with friends and suddenly the game is unplayable, but the alternative is the 180W charger letting the battery drain like crazy and trying to keep it from going too far.

As someone who basically never uses a laptop on battery in the first place, allowing battery conditioning logic to repeatedly tank my experience is a complete waste. If I could rollback to a previous BIOS I would, but the forum thread for that doesn't record anyone doing so successfully. Please resolve this ASAP. Thanks

Obasav, 2025-02-15:
> After a lot of frustration with throttling on the 180W charger I got myself a Delta 240W. In Balanced mode, I can go for 2-3hrs in a demanding game until suddenly performance/GPU power throttles and [...]

Unfortunately, Framework has stated that you cannot roll back the bios version from 3.05...

"Important Note

After updating to this version you will not be able to downgrade to an earlier version."

[image]

Obasav, 2025-02-28:
Another update to add to this, for visibility and having some extra data points...

First off, I'd like to thank everyone for taking time to contribute their data to this issue. It feels like an eternity waiting. 

I've made some more changes to my system since the last update, and switched to using Nobara 41 (Kernel 6.13.5) for these tests, along with Mesa 25.1 and Proton Experimental. The rest of my system is still the same, and these are the current bios settings I have:

[image]

As you can see, I have my iGPU memory set to auto and the Battery Extender enabled.

For these tests, I'm using LACT to raise the dGPU power limit to 120W.

Picture 1: Performance mode. 79.2 avg framerate & significant fluctuations in framerate.

Picture 2: Balanced mode. 79.4 avg framerate & fewer larger fluctuations in framerate.

Picture 3: Power saver mode. 11.4 avg framerate. dGPU power never exceeded 13W of power. 

[image]
[image]
[image]

As you can see in the above photos, Balanced power mode without curve optimizer has a significantly more stable framerate, and a slightly higher average as a result. Power saver mode does what it's supposed to do, and the dGPU doesn't go to any faster power states, or use more than 13W of power (**This occurs across all tests, and as a result I will be leaving it out of the rest of the post.**)

The second set of tests I conducted included adding a -30 all core curve optimizer, while keeping Battery Extender enabled and the power limit for the dGPU set to 120W.

Picture 1: Performance mode w/ -30 CO-all. 79.6 avg framerate & lots of fluctuations in framerate.

Picture 2: Balanced mode w/ -30 CO-all. 79.9 avg framerate & few fluctuations in framerate.

[image]
[image]

As seen above, CO on

Obasav, 2025-03-01:
Some more data about performance mode:

Approximately every 10 seconds the VRAM power state goes from level 3 down to level 0, with a large dip in the clock speed of the dGPU. Note that in the first photo, the dGPU is pulling short bursts of 150W+ of power. At some points, the battery charging indicator switches from discharging to charging.

[image]
[image]

[image]
[image]

Obasav, 2025-03-02:
Compared to my previous tests with the 240W adapter, here are some tests showing how the performance is using the 180W charger. Using the same settings as the curve optimizer + power limit adjustments:

[image]
[image]

knipp30, 2025-03-18:
@kiram9 

Just checking in to see if this was reproduced by you guys as well?

Are any additional logs needed?

Any further information?

I was able to see using LACT that the main issue seems to be two fold:
1. The laptop will drop to 95% when under load (gaming)  - which is fine except for..
2. Once the laptop hits 95%, the system cannot seem to determine the correct VRAM and GPU power states - causing it to jump constantly between power states.

Ideally, with the 240w charger, the system should never even try to discharge.

I am happy to reproduce and provide any necessary logs - but the current situation really makes it difficult to use this computer for an extended period of time for anything that is GPU intensive. (And falling back to the 180w charger is just as much an issue, as it will discharge and drop to a lower power state within moments of use anyway.)

jcdutton, 2025-05-07:
Hi, 

If you have a 240W PSU/Charger, please can you do the following ectool commands.
It will tell me whether the charger is in bypass mode or not. A 240W charger should be in bypass mode, but might not be due to a bug somewhere.

sudo ectool chargestate param 3
interfaces:0xffffffff
comm_init_dev being used /dev/cros_ec
49168 (0xc010)       # chg_status

sudo ectool chargestate param 4
interfaces:0xffffffff
comm_init_dev being used /dev/cros_ec
42926080 (0x28f0000)       # chg_option

knipp30, 2025-05-07:
@jcdutton :
$ sudo ectool chargestate param 3
114704 (0x1c010)       # chg_status
$ sudo ectool chargestate param 4
50534497 (0x3031861)       # chg_option`

chiefmigizi, 2025-11-20:
I just upgraded to 4.02 and things look to be fixed now. I ran the same benchmarks as before.

Balanced Mode
68 FPS

Performance Mode
72 FPS, no more stuttering.

quinchou77, 2025-12-30:
Please test v4.03 Beta if you encountered issues with the previous version. We identified several incorrect settings in the Power Table within v4.02 and have corrected them in this latest release. Our goal is to move v4.03 to the stable channel next week, pending successful community feedback.