FrameworkComputer/SoftwareFirmwareIssueTracker #30: Can't enter Secure Boot setup mode

pjungkamp, 2024-12-24:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [x] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

```
Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Framework
	Product Name: Laptop 13 (Intel Core Ultra Series 1)
	Version: A5
	Serial Number: <REDACTED>
	UUID: <REDACTED>
	Wake-up Type: Power Switch
	SKU Number: FRANDPCP05
	Family: Laptop
```

### BIOS VERSION

```
Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: INSYDE Corp.
	Version: 03.04
	Release Date: 10/09/2024
```

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug

"Erase all Secure Boot Settings" leaves EFI variables in an unexpected state that is described in multiple posts on the forum:

> I’ve tried restarting with “Erase all Secure Boot Settings”, “Restore Secure Boot to Factory Settings”, and both “Erase” and “Restore” set to enabled in the BIOS menu, but on booting and running sbctl status it shows setup mode as disabled.

See https://community.frame.work/t/cant-enable-secure-boot-setup-mode/57683

> When I put secureboot into setup mode via:
> 
> Firmware settings → “Administer Secureboot” → “Erase all Secureboot Settings” → Enable → F10
> 
> Several efi-based tools stop working on subsequent boots.

See https://community.frame.work/t/putting-secureboot-into-setup-mode-breaks-several-efi-tools/62033

## Steps To Reproduce

Steps to reproduce the behavior:
1. Go into UEFI firmware "Administer Secure Boot"
2. Select "Erase all Secure Boot Settings"

The EFI firmware enters a state where several EFI variables are just missing.

## Expected behavior

"Erase all Secure Boot Settings" should enter a proper Secure Boot Setup Mode state.

## Operating System (please complete the following information):
- OS/Distribution: NixOS
- Version: nixos-unstable 2024-12-17
- Linux Kernel Version: `6.5.7`

## Additional context

- https://github.com/Foxboron/sbctl/issues/407
- https://community.frame.work/t/cant-enable-secure-boot-setup-mode/57683
- https://community.frame.work/t/putting-secureboot-into-setup-mode-breaks-several-efi-tools/62033

quinchou77, 2025-03-26:
When set "Erase all Secure Boot Settings" to "Enabled", it will erase all secure boot setting and clear PK, KEK, DB, and DBX key after reboot the system. Then, it enters the setup mode, so you can enroll your key.

pjungkamp, 2025-07-06:
I just updated to 3.05 and it's still affected.

The "Erase all Secure Boot Settings" options still cause many efi vars to disappear entirely...

mariuszste, 2025-10-15:
3.06, can reproduce, how is this still not fixed?