FrameworkComputer/SoftwareFirmwareIssueTracker #32: Laptop Issue

ilc, 2024-12-25:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [X] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
Please provide the bios version. 

3.03 -> 3.04

Windows:
Open a command terminal as administrator:
```wmic bios get SMBIOSBIOSVERSION```

Linux: 
Open a terminal and run the following command
```sudo dmidecode --string bios-version```

[nix-shell:~]$ sudo dmidecode --string bios-version
03.04

### DIY Edition information
If you are experiencing an issue on a DIY system, Please also fill out the memory and storage devices you are using. 

Memory: Manufacture and SKU:  Crucial CT2K48G56C46S5 x 2
Storage: Manufacture and SKU: Seagate WDS800T2X0E 

### Port/Peripheral information
If you are experiencing an issue with a peripheral or an expansion card/port please fill out the following information: 
* Peripheral vendor and name.
* Port the Peripheral was connected to. Please see the following for port numbering: [Laptop 16](https://knowledgebase.frame.work/expansion-card-slot-functionality-on-framework-laptop-16-rkUjGm7cn) or [Laptop 13](https://knowledgebase.frame.work/expansion-card-functionality-on-framework-laptop-13-amd-ryzen-7040-series-SkrVx7gAh)
* Device or expansion card attached to the **Adjacent port** to the port that is having the issue.
1. [ Fill In device / expansion card ]
2. [ Fill In device / expansion card ]
3. [ Fill In device / expansion card ] 
4. [ Fill In device / expansion card ]

The following are for Laptop 16 only. 
5. [ Fill In device / expansion card ]
6. [ Fill In device / expansion card ]

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [X] No

## Describe the bug
A clear and concise description of what the bug is.

I wen t to update my Framework 13 from 3.03 to 3.04 with fwupdmgr and it didn't sequence the update right.  I can not execute the port retimer updates, because my laptop is on 3.04 now.

## Steps To Reproduce
Steps to reproduce the behavior:

Use fwupdmgr to go from 3.03 -> 3.04.

## Expected behavior
A clear and concise description of what you expected to happen.

I expect the update to work.

## Screenshots
If applicable, add screenshots to help explain your problem.

## Operating System (please complete the following information):
 - OS/Distribution: NixOS unstable
 - Version: As of today.
- Linux Kernel Version: Linux bear 6.6.66 #1-NixOS SMP PREEMPT_DYNAMIC Sat Dec 14 19:00:23 UTC 2024 x86_64 GNU/Linux

## Additional context
Add any other context about the problem here.

I'm unclear on the impacts this will have on the machine.  :(

zquestz, 2025-01-17:
> I'm not sure if it's the exact same issue you're facing, but I had issues with the fwupdmgr and the retimer updates when updating to 3.04. I was able to successfully run `sudo fwupdmgr update` after fully disconnecting all power and letting the board sit for a couple minutes before restarting.
> 
> See the related issue [#39](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/39) .

This suggestion worked for me. =)

JohnAZoidberg, 2025-01-29:
> Linux Kernel Version: Linux bear 6.6.66
> Framework Laptop 13 (Intel® Core™ Ultra Series 1)

For Ultra Series 1 you should use kernel 6.9 or newer or several things like suspend won't work correctly.

> I can not execute the port retimer updates, because my laptop is on 3.04 now.

From GNOME Firmware you can always reinstall an update. `fwupdmgr install` also lets you install a custom version of any firmware. For next time :)

>  was able to successfully run sudo fwupdmgr update after fully disconnecting all power and letting the board sit for a couple minutes before restarting.

Retimer update is complicated and sometimes they don't properly reset after the update, which is why removing power and letting it sit for a minute makes everything work again.

closed 2025-01-29 by JohnAZoidberg