FrameworkComputer/SoftwareFirmwareIssueTracker #35: Sleep/Hibernate issue with Windows 11

CDRXavier, 2025-01-03:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [x] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
Please provide the bios version. 

03.05 (Oct 2 2024)

### DIY Edition information

#### Memory: Manufacture and SKU
* Crucial CT16G56C46S5.M8G...

#### Storage: Manufacture and SKU
* SAMSUNG MZ9LQ512HBLU-00B00
* WDC PC SN530 SDBPPNPZ-512G-1006

### Port/Peripheral information
1. USB 4
2. USB-A
3. USB-A
4. USB 4
5. USB-A
6. USB-A

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug

Windows 11 (and 10) will experience different levels of crash when entering/exiting sleep (state with power button slowly fading on and off).

### Windows 11:
* Windows 11 will automatically restart after a unspecified amount of time, into a cold boot.
[image]

### Windows 10:
Windows 10 will be:
* unable to wake from sleep. Pressing power button, activating lid switch, performing action on keyboard.numpad/trackpad does nothing.
* (after mannual trigger) Wakes up into cold boot.
[image]

## Steps To Reproduce
Steps to reproduce the behavior:

I have lid switch set to "do nothing" in the Windows power profile, so just press the power button once. And then leave.

The chance of success seem to occur more often when on battery, when Windows actually is willing to sleep. Compared to when plugged in, when it doesn't really try to.

## Expected behavior

The will 

## Operating System (please complete the following information):
 - OS/Distribution: Windows 11, Windows 10 (respectively)
 - Version: 24H2, 22H2 (respectively)

## Additional context

Since Framework does not provide Windows 10 drivers, I have gotten the chipset driver from AMD.
Windows 11 runs on the 5.06.29.310 drivers in the Oct 2, 2024 package. Windows 10 should be running on the 6.10.17.152 chipset package, and the same 23.40.18.02 graphics driver packaged from Framework.

CDRXavier, 2025-01-03:
Note the dual m.2 carrier card has not been installed yet. I will be doing this shortly.

Should I post as separate issue, or continue to update this issue?

PureKrome, 2025-01-03:
[Hardware_Drivers_Report.csv](https://github.com/user-attachments/files/18299430/Hardware_Drivers_Report.csv)

I've attached my hardware drivers and their OS's.

BIOS: 03.05 (Oct 2 2024)

I can provide my ORDER NUMBER so you can reference the exact model/specs/manufacturing run, etc?

OS:
Edition	Windows 11 Pro
Version	24H2
Installed on	‎26/‎11/‎2024
OS build	26100.2605
Experience	Windows Feature Experience Pack 1000.26100.36.0

---
Extra Info:

I swapped out my stock WIFI card for an Intel(R) Wi-Fi 6E AX210 160MHz
 
Lastly I sometimes have "Bluetooth paired" my Bose Q35 Wireless Headphones II and an external speaker called MEGABOOM 3. I usually connect the Q35 II's and rarely connect the Megaboom 3. I had not connected that recently when I had the crashes.

PureKrome, 2025-01-16:
> > BIOS: 03.05 (Oct 2 2024)
> 
> I assume you are having the same issues?

Correct - I am having the same issues.

CDRXavier, 2025-01-18:
Right.
Here's what happens when I do run m.2s in the enclosure.

[image]

It's worse, and frankly ... There's nothing plugged in to the USB-A cards, is there?

And there's no bloody circuitry, it's literally dumb wires

CDRXavier, 2025-03-05:
Did something in Windows 10 that allowed it enter hibernation mode.

CDRXavier, 2025-04-01:
So. A few things.
After reading this (funny thing with self-encrypting drives and firmware), I have decided to turn off the hard drive passwords.
https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/42

I have password on all my drives -- Windows drive, Linux drive, my data drive. It's simpler than Bitlocker, a lot more reliable.
This way I can decide which drive to "connect" to the selected OS to boot up. This is after the Linux-Windows incident. (where the Ubuntu 2404 installer wiped the Windows install disk).
And then my Linux install worked immediately. I have since moved to Windows, and for a very long time I straight up don't get any problem waking up.