FrameworkComputer/SoftwareFirmwareIssueTracker #36: Long BIOS boot times between Power On and GRUB menu.

jcdutton, 2025-01-03:
## Device Information

### System Model or SKU
 [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)
No dGPU.

### BIOS VERSION
3.0.5

Windows:
N/A

Linux: 
Open a terminal and run the following command
```sudo dmidecode --string bios-version```
03.05

### DIY Edition information
Memory: Manufacture and SKU
Kingston Fury Impact: Part Number: KF556S40-32
2x making 64GB total.
Storage: Manufacture and SKU
Model Number:                       WD_BLACK SN850X 1000GB
Firmware Version:                   620361WD

### Port/Peripheral information
1. USB-C card, nothing plugged in.
2. Empty
3. Empty
4. Empty
5. USB-C card, FW16 PSU plugged in.
6. USB-A card, with "Adafruit SHT45 Trinkey - USB Temperature and Humidity Sensor" plugged in.

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] No

## Describe the bug
The BIOS takes 60 seconds to get from Power on to GRUB menu when the "Adafruit SHT45 Trinkey - USB Temperature and Humidity Sensor" is plugged into port 6.
The USB chips hang and reset themselves. I suspect USB chip firmware bugs.
Note, this is just the BIOS boot time from power on to GRUB menu. It is before the OS is loaded.

## Steps To Reproduce
Steps to reproduce the behavior:
1. Start from a powered off laptop. (It has to start from a powered off laptop to reproduce the problem)
2. Plug the "Adafruit SHT45 Trinkey - USB Temperature and Humidity Sensor" into port 6
3. Power on laptop
4. Time how long it takes to get to the GRUB menu. It takes 60 seconds.

## Expected behavior
It should get to the GRUB menu in about 7 seconds.

## Screenshots
N/A

## Operating System (please complete the following information):
 - OS/Distribution: Linux/Ubuntu
 - Version: 24.04
- Linux Kernel Version: `uname -a`  6.12.7  <- Mainline compiled kernel.

## Additional context
Add any other context about the problem here.

JohnAZoidberg, 2025-02-03:
> The USB chips hang and reset themselves. I suspect USB chip firmware bugs.

What does that mean? Which chips and how can you tell what they are doing?

amstan, 2025-04-09:
I think it'll be easier for us to buy one of [those](https://www.adafruit.com/product/5896). More importantly though, what firmware is are you running on it?
I have worked with mcu firmware that did usb device in the past, and I have managed to hang my computer. You're right though, i feel like usb controllers shouldn't hang (i'm assuming that's what you mean by "USB chips hang and reset themselves. I suspect USB chip firmware bugs") regardless of what usb devices are doing and how they're misbehaving.