FrameworkComputer/SoftwareFirmwareIssueTracker #37: feat: Resizeable Bar Support (via Thunderbolt or USB 4)

ChihweiLHBird, 2025-01-07:
Many people are working with an external GPU (eGPU) enclosure to run a discrete graphics card with the laptop.

There are some recent launched graphic cards like Intel Arc series which can utilize the Resizable Bar (ReBAR) feature to get a significant performance improvement.

The BIOS and chipset used by Framework laptops should have the ReBAR option under the hood but not exposed as a configurable option in the BIOS UI.

There is also a long-waiting feature request in the Framework Community for this feature: https://community.frame.work/t/rebar-support/22392

Implementing it looks very simple, just to add another entry in the BIOS UI for enabling it. I wonder if there is any chance we may get this feature in the foreseeable future. Thanks!

Vashiru, 2025-01-11:
Very much this. I bought a framework to have a compact PC that if needed I can take with me, I bought an eGPU enclosure so I can game with it at home, but I've not pulled the trigger on the GPU yet because it turns out Framework doesn't have the option to enable ReBAR. ReBAR is required for performance from the new Intel GPUs and those are really appealing at the moment.

The chipset supports it, people have enabled it with hacks and it works, it's just a matter of we don't currently have the option for it.

amstan, 2025-01-11:
Sounds like a useful thing, sure!

> The chipset supports it, people have enabled it with hacks and it works

Do you have any references there? Might be useful if it's just an acpi table modification.

ReedGraf, 2025-01-21:
I also have been waiting for this to come to FW13 laptops for quite a while! I've looked at the BIOS mod for the FW13 but 13th gen didn't have a public BIOS at the time. Plus, it seems too risky for how important my laptop is to my daily life. I'd really appreciate it if Framework released a BIOS update that gave ReBAR as an option! ❤

CDRXavier, 2025-02-03:
> There are some recent launched graphic cards like Intel Arc series which can utilize the Resizable Bar (ReBAR) feature to get a significant performance improvement.

The reason is because Intel Arc have silicon-level bug. When the GPU runs, sometimes it would "cache miss" (think like CPU cache, but for GPU as well). Specifically the silicon bug will prevent the core that triggered the "cache miss" to stall (wait around doing nothing), instead of trying to figure out something else to do.

This is why anything that cuts down latency (e.g., ReBar) significantly increase performance, as it reduce the time cache stalls.

onix331, 2026-08-18:
While playing around with an eGPU (Intel ARC A770 16GB) on my FW13 Pro (Intel) i also just noticed that ReBAR does not seem to be enabled when using Linux (Fedora 44).

 
```
03:00.0 VGA compatible controller: Intel Corporation DG2 [Arc A770] (rev 08) (prog-if 00 [VGA controller])
        Subsystem: Intel Corporation Device 1020
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupts: unknown pin routed to IRQ 189, MSI(X) routed to IRQ 189
        IOMMU group: 23
        Region 0: Memory at a4000000 (64-bit, non-prefetchable) [size=16M]
        Region 2: Memory at 8010000000 (64-bit, prefetchable) [size=256M]
```

I tried enabling it via `setup_var.efi` but fell into the trap of `WRITE_PROTECTED (The device cannot be written to)` which is really sad.

I would be very pleased if this is a feature that will make it into the UEFI someday since the option is clearly there