FrameworkComputer/SoftwareFirmwareIssueTracker #38: Framework 16 incompatible with JMS583 based NVMe SSD enclosures

Niko-O, 2025-01-10:
## Device Information

### System Model or SKU
Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
```
C:\Windows\system32>wmic bios get SMBIOSBIOSVERSION
SMBIOSBIOSVersion
03.05
```

### DIY Edition information
I ordered DIY but the laptop came mostly pre-assembled. (Not complaining, just noting down.)

Memory: I ordered  "DDR5-5600 - 32GB (2 x 16GB)", it was already installed and I never replaced it.
Storage: WD Black SN770M 1TB (2230 slot)
Storage: When running Windows 10: Samsung 990 Pro 2TB (2280 slot)
Storage: When running Windows 11: Samsung 970 Evo Plus 2TB (2280 slot)

### Port/Peripheral information
https://www.amazon.de/dp/B08C2G7CQF
AGPTEK m.2 NVMe SSD to USB 3.2 Gen 2 (10Gbit/s) enclosure. Uses the JMicron JMS583 chip internally.

> * Port the Peripheral was connected to
> * Device or expansion card attached to the **Adjacent port** to the port that is having the issue

See section "Details" below.

### Standalone Operation
No

## Describe the bug
When the SSD enclosure is connected directly to the laptop, such that it connects with USB 3.2 Gen 2 (10Gbit/s), and files are copied to/from the SSD, then:
* Most of the times, the copy speed quickly drops to zero, stays there for a few seconds, and then the operation fails with a 0x800701B1 error code. 
* But sometimes the speed stays at zero for minutes without completely disconnecting.  And sometimes the copy continues at wildly inconsistent (i.e. slow) speeds. 
* Sometimes I don't even have to start the copy, and simply opening the drive in Explorer causes a disconnect.
* And very rarely it appears to be stable at consistent and plausible speeds.

It seems that different slots also have different behavior... sometimes? It's hard to tell.

## Steps To Reproduce
* Put an m.2 NVMe SSD into the enclosure.
* Connect the enclosure directly to the laptop.
* Open the drive in Windows Explorer.
* Copy files off the drive.

## Expected behavior
The copy should happen at consistent and high speeds and complete successfully.

## Details

### Tests

* The problematic m.2 NVMe to USB-C enclosure connects with USB 3.2 Gen 2 (10Gbit/s). I do not have any other devices that feature such a USB connection.
* The issue is **not specific** to the SSDs in the problematic enclosure. I tried different SSDs.
* The issue does **not persist** when using an m.2 SATA to USB-C enclosure that connects with USB 3.1 Gen 2 (10Gbit/s).
* The issue is **not specific** to whether I use a USB-C to USB-C cable connected to a USB-C expansion module, or USB-C to USB-A cable connected to a USB-A expansion module.
* With the USB-C to USB-A cable and a USB-A expansion module, I can do the "half insert trick" (insert the USB-A plug partially at first, wait for the device to show up, then insert it fully). This way, the problematic enclosure connects with USB 2.1 and the issue **does not persist**.
* The issue is **not specific** to the used cable. I tried different ones. I also tried skipping the expansion card and connecting directly to the recessed USB-C port, but the issue **does persist**.
* The issue is **not specific** to the presence and usage of other expansion modules. The issue persists with nothing else connected to the laptop, with regular use, and with intense use on all modules.
* The issue is **not specific** to whether I read from or write to the drive.
* The issue does **not persist** when using a USB stick that connects with USB 3.2 Gen 1 (5Gbit/s).
* The issue does **not persist** if I use a (passive) USB hub that connects with USB 3.0 (5Gbit/s) and connect the problematic enclosure to that. It does not matter whether I first connect the enclosure to the hub, or the hub to the expansion module.
* The issue does **not persist** if I use an (active) USB-C dock that connects with 5Gbit/s (I forgot to write down the USB version and I don't have that dock anymore, but it definitely was 5Gbit/s) and connecting the problematic enclosure to that.
* The issue **does persist** when using a USB-A interposer board with beefy capacitors on the power rails and additional externally supplied 5V.

At some point I got an overcurrent warning when testing on slot 3. I had to reboot to get the port working again.
[image]

[... 35 lines cut ...]
## Screenshots

Two examples of copy operations with inconsistent speeds (but no disconnects):
[image]
[image]

Here's a video that demonstrates the issue:
https://github.com/user-attachments/assets/ba50bd61-6504-42fd-8e3f-e55aeae53fa3
* 00:00 At first, the enclosure is connected to an unpowered hub, which in turn is connected to slot 2.
* 00:03 The screen shows the SSD in the enclosure (P drive) in the top right, one of the internal SSD (C drive) in the bottom right, and USB Device Tree Viewer on the left.
* 00:12 I copy a virtual machine (21.5GiB in total) from P to C. Note how the transfer speed is perfectly stable at 370MiB/s (~3.1Gbit/s).
* 00:23 The transfer finishes successfully. I plug the enclosure directly into slot 2.
* 00:46 (After deleting the previously copied files) I copy the same virtual machine from P to C again.
* 00:51 The transfer speed reaches 0 Bytes/s.
* 01:03 The transfer speed momentarily goes up a few times while I wait, but always quickly goes back down to 0.
* 01:10 I cancel the transfer and wait for a while before giving up and ending the video.

## Operating System:
Happens both on Windows 11 23H2 and Windows 10 22H2 with up-to-date drivers and after a mainboard reset.

## Additional context
Add any other context about the problem here.

On 2024-11-24 I contacted Framework Support, asking whether there is any progress on this issue, as it was brought up to Framework support already, according to https://community.frame.work/t/amd-framework-and-nvme-ssd-enclosure-compatibility-investigation/41775/12.
On 2025-01-03, after a long and frustrating merry-go-round of ignoring my questions or giving non-sequitor answers, asking questions that I have already answered, giving me unclear and sometimes flat-out wrong instructions, I was instructed to file a new issue for this here.

CDRXavier, 2025-01-16:
Can confirm that it is also flaky on my Framework 16, but only on the DP/PD capable ports. Connected to the USB 3.2 ports there are less issues.

And mine are straight USB connections to the laptop (root hub), with nothing in between.

> About the USB-interposer board: It's a USB 2 hub with USB 3 passthrough (https://www.amazon.de/dp/B09241T3CS):

This stuff is straight-up out-of-spec. Do not expect any connected 3.0 device to work.

jcdutton, 2025-02-02:
Have you seen this page:
https://knowledgebase.frame.work/en_us/expansion-card-slot-functionality-on-framework-laptop-16-rkUjGm7cn

Specifically read the paragraph about how much power each port can output.
Most nvme enclosures need 3A option, so best to use slot 1 or 4 for the nvme enclosure.
I use slot 1,4 for fast devices, 2 or 5 for PSU, and hardly use slot 3 or 6.

Niko-O, 2025-02-02:
@jcdutton Thanks. I have seen the graphic, but not the information about the current limitation. That explains part of it.

Niko-O, 2025-04-28:
I got a mainboard replacement (slot 4 on mine had a physically damaged USB-C port, covered under warranty), and the new mainboard has the same problem.

jcdutton, 2025-04-29:
You can probably only plug in one USB SSD at a time. Due to the power limits.
It assigned 15W to the first item plugged in, and then only 9W to other devices, Some SSDs need about 10W, so don't work if you plug 2 in at the same time.

Niko-O, 2025-04-29:
@jcdutton I assume your intention is simply to add this as information, but just in case you are trying to point out a possible reason for why this might be happening for me: This issue also happens with just the charger connected, and even with nothing else connected to the laptop at all.

CDRXavier, 2025-06-04:
> Due to the power limits.
This is true, but also un-true. If you *don't* run a hub (at least, un-powered hub), you should be able to plug a 1-SSD enclosure into any of the 6 ports.
> Some SSDs need about 10W
Can't say I have tested those. But I imagine the enclosure run them at lower speeds (and power consumption) anyway.

Niko-O, 2026-04-26:
Can't attach arbitrary files here, so a description will have to do:
A 18392*512+1024=9417728 bytes long file, starting with 18392*512=9416704 zero-bytes, followed by the following 1024 bytes:
`37-6E-48-F1-74-00-00-00-00-00-00-00-00-00-00-00-00-E4-63-DF-BE-7D-D1-E4-C9-93-3B-CC-98-14-93-4B-7A-FE-F9-E7-82-39-CD-CD-CD-79-F7-F7-EB-D7-3F-D1-7B-FF-FD-F7-77-58-7F-D7-5D-77-25-EA-87-94-95-E5-BD-6B-EF-DE-86-E0-AC-AF-BD-FA-6A-87-3D-83-2F-BE-38-51-3F-61-C2-84-BC-F7-FD-F3-9F-6B-82-7D-5B-36-6F-8E-D2-6A-6D-6D-0D-EE-6E-3A-E9-A4-93-BA-2C-97-D4-D8-F8-55-F4-E0-83-0F-E6-F5-5B-49-93-4B-7A-E5-95-F9-C1-9C-78-67-BE-06-0D-1E-9C-E8-8D-7F-0B-00-00-00-00-00-00-00-00-00-00-00-00-74-AD-83-07-0F-1E-BE-8B-26-57-AE-64-CA-94-29-5D-92-4B-EA-DF-FF-82-C4-8C-49-93-26-15-D4-FF-C6-1B-AF-27-FA-7B-F7-E9-93-B3-36-CE-3B-65-9E-39-CE-BA-14-22-33-77-73-DD-75-A3-72-D6-D6-FE-EB-5F-C1-BE-CD-9B-37-15-B4-6F-D8-B0-8A-44-FF-B4-69-D3-A2-B4-EE-B9-E7-9E-C4-AC-E1-C3-2F-8F-7A-F4-EC-59-74-2E-A9-AD-AD-2D-9A-35-EB-AF-39-7F-2B-0F-3C-F0-40-97-E4-92-32-33-5E-7F-BE-F3-CF-05-F5-2F-5E-BC-28-D1-5F-DA-A3-47-D4-DE-DE-5E-F0-39-00-00-00-00-00-00-00-00-00-00-00-00-C8-ED-91-47-1E-C9-9A-31-E9-DE-BD-7B-B4-68-D1-5B-87-6B-8A-CD-25-7D-FE-F9-E7-C1-8C-15-CB-97-17-3D-A3-B6-B6-36-6B-ED-CA-15-2B-82-DA-B8-BF-10-F1-F9-32-67-B4-B4-B4-64-AD-7D-F2-C9-27-13-75-71-06-A8-50-99-33-06-5C-74-51-94-C6-A2-B7-DE-0A-CE-1D-FB-4D-AF-5E-45-E7-92-A6-4F-9F-9E-33-93-B4-70-E1-9B-87-6B-8A-CD-25-FD-E7-3F-FB-82-19-F1-33-15-3B-63-ED-DA-0F-0B-9A-01-00-00-00-00-00-00-00-00-00-00-00-40-E1-B9-A4-51-A3-AE-8F-1A-1A-1A-8E-D4-14-9B-4B-AA-A9-59-18-CC-68-6E-FE-6F-C1-67-3D-A7-6F-DF-C4-8C-A7-9F-7E-3A-6B-DD-C3-0F-3F-9C-A8-EB-D7-AF-7F-C1-BB-9A-9A-9A-82-33-AF-5A-F5-4E-D6-DA-8A-8A-E1-89-BA-F1-E3-C7-17-BC-6F-FD-BA-75-C1-BE-A3-BF-83-7C-EC-DB-B7-2F-2A-29-29-49-CC-D8-B0-61-C3-E1-FF-9D-D1-BB-F7-31-C9-25-C5-B9-A3-3D-F5-F5-47-6A-8A-CD-25-2D-5D-FA-76-30-E3-AB-AF-BE-2A-F8-AC-17-0E-18-90-98-F1-D8-E3-8F-15-3C-03-00-00-00-00-00-00-00-00-00-00-00-80-DC-26-4F-9E-7C-24-BB-31-A4-AC-2C-5A-FA-F6-DB-41-4D-B1-B9-A4-47-1F-7D-34-D1-7F-EE-EF-7F-9F-EA-AC-D7-8F-1E-9D-98-33-62-C4-88-AC-75-95-95-95-89-BA-71-E3-C6-A5-DA-D7-BB-4F-9F-C4-9C-29-53-A6-64-AD-2B-2D-2D-4D-D4-55-57-57-17-BC-AB-B1-B1-B1-E8-7B-82-46-8E-1C-99-E8-BF-F3-AE-3B-8F-FC-EF-AC-DF-FE-B6-E8-5C-D2-D4-A9-53-8F-F4-FF-71-E0-C0-AC-E7-2B-36-97-34-63-C6-8C-44-7F-9C-A7-4A-E3-96-5B-6E-49-CC-29-2F-2F-4F-35-07-00-00-00-00-00-00-00-00-00-00-00-80-EC-6E-BD-F5-D6-A8-CF-99-67-46-8B-16-E5-CE-C0-14-9B-4B-AA-AA-AA-4A-F4-5F-75-D5-55-45-67-A8-E2-4F-9C-B5-C9-E6-37-BD-7A-25-EF-CA-79-2C-DD-5D-39-65-43-87-26-E6-DC-70-C3-0D-41-CD-17-5F-7C-11-BC-9F-25-4B-16-17-BC-AB-AD-AD-2D-98-93-EB-3E-A8-6C-66-CF-9E-1D-F4-1F-ED-EC-73-CE-29-3A-97-34-71-E2-C4-A8-D7-E9-A7-1F-BE-FF-2A-97-62-73-49-63-C7-8E-4D-F4-5F-7A-E9-A5-51-57-DC-03-D6-F3-B4-D3-52-CD-01-00-00-00-00-00-00-00-00-00-00-00-20-BB-3D-F5-F5-9D-D6-14-9B-4B-FA-C3-85-17-26-FA-EF-BD-F7-DE-54-67-8D-73-3A-99-67-69-6D-6D-4D-D4-FC-F8-E3-8F-41-CD-FC-F9-F3-52-ED-1B-33-66-4C-62-4E-7C-47-50-A6-BA-BA-BA-60-DF-A6-4D-B5-A9-F6-65-DE-69-74-F7-DD-77-E7-D5-F7-F1-C7-E1-19-32-BF-D7-F3-CE-3B-AF-E8-5C-D2-DE-BD-0D-9D-D6-14-9B-4B-CA-CC-82-8D-1F-3F-3E-4A-63-EE-DC-39-C1-59-F6-EF-DF-9F-6A-16-00-00-00-00-00-00-00-00-00-00-00-00-E9-14-9B-4B-EA-D1-B3-67-A2-FF-A9-A7-9E-4A-75-8E-D7-5E-7D-35-38-4B-4B-4B-4B-A7-F7`

I tried various random blocks of 9416704 bytes instead of all zeros, and they all still cause a disconnect.
I tried over a hundred-thousand different random blocks of 1024 bytes instead of the 1024 bytes above, but none of those cause a disconnect.
I also tried all 1024*8=8192 ways of flipping a single bit in the 1024 bytes, but none of those cause a disconnect either.

If the file is shorter even by 1 byte, the disconnect does not happen.
If the file is longer, the disconnect happens before reaching the end.

It only happens on my Framework 16 on slots 3 and 5, and only if connected as USB3. If I only partially insert the USB-A plug, such that the superspeed-lines don't make contact, the drive connects with USB2 and then the disconnect doesn't happen.
Slots 1, 2 and 4 are fine. (Drive doesn't show up on slot 6 at all.)
It doesn't happen on other computers (4 different devices tested).

The disconnect happens when copying the file by dragging it from windows Explorer, when programmatically writing the file directly onto the drive in one go, and when programmatically writing the file in 512 byte blocks by opening the file in OpenOrCreate-mode, seeking to the end, writing the next 512 bytes, closing the file, and then repeating for the next block. Writing programmatically causes csrss.exe (a Windows process) to show one top-most "Delayed write failed" error message box on the desktop for the block that caused the disconnect and for every block that was attempted to write after that:

https://github.com/user-attachments/assets/780ec5c6-aa82-4aa5-9009-01c45ec87a6a

It also breaks Windows' shutdown menu:

https://github.com/user-attachments/assets/28f7047e-ba9e-4334-a005-a56be03c0964

Framework. Truly the last laptop I'll buy.

jcdutton, 2026-04-26:
Hi,
In case it is linked.
On linux, doing the following command has a similar effect:
sudo lsusb -v

Essentially, it kills the USB 3 chip on the FW mainboard, until the next reboot.