FrameworkComputer/SoftwareFirmwareIssueTracker #39: Firmware Update Inconsistency - 3.04 Intel® Core™ Ultra Series 1

zquestz, 2025-01-16:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [x] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
03.04

### DIY Edition information
Memory: 96GB RAM (from Framework)
Storage: 4TB NVME (from Framework)

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug
I updated my BIOS using the following link: https://downloads.frame.work/bios/Framework_Laptop_13_Intel_Core_Ultra_Series1_capsule_signed_allsku_3.03_3.04_EFI.zip

This was successful and told me all updates were applied, I saw no errors.

However, my system still thinks the update is available via fwupdmgr. When I run `sudo fwupdmgr get-upgrades` it still shows the updates as available. Here is the output.

```
Devices with no available firmware updates: 
 • Rubix24
 • System Firmware
 • TPS DMC Family
 • TPS6598X PD#1
 • USB4 Retimer
 • WDS400T3X0C-00SJG0
Devices with the latest available firmware version:
 • Fingerprint Sensor
 • Intel Management Engine
 • TPS6598X PD#0
 • UEFI dbx
 • USB3.0 Hub
Framework Laptop 13 (Intel Core Ultra Series 1)
│
├─UEFI Device Firmware:
│ │   Device ID:          93c5f795ba1fe8224d57c20ad3857ee56b403b1e
│ │   Summary:            UEFI System Resource Table device (Updated via capsule-on-disk)
│ │   Current version:    0x000000cf
│ │   Vendor:             Framework (DMI:INSYDE Corp.)
│ │   Update State:       Success
│ │   GUID:               c57fd615-2ac9-4154-bf34-4dc715344408
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
│ │ 
│ └─Laptop Intel Core Ultra Retimer01 Controller Update:
│       New version:      0x00000270
│       Remote ID:        lvfs
│       Release ID:       99280
│       Summary:          Framework Laptop Retimer 01 Firmware for Intel Core Ultra Mainboards
│       License:          Proprietary
│       Size:             2.3 MB
│       Created:          2024-09-11
│       Urgency:          High
│         Tested:         2024-10-31
│         Distribution:   ubuntu 24.10
│         Old version:    0x000000cf
│         Version[fwupd]: 1.9.24
│         Tested:         2024-09-20
│         Distribution:   pop 22.04
│         Old version:    0x00000270
│         Version[fwupd]: 2.0.0
│       Vendor:           Framework
│       Duration:         2 minutes
│       Release Flags:    • Trusted metadata
│                         • Is upgrade
│                         • Tested by trusted vendor
│       Description:      
│       This release makes the following improvements:
│       
│       • This update adds certified Thunderbolt support.
│       • The firmware is signed by Intel
│       Checksum:         e283e9d4b9fd083bb7b592dc2f58353687ab6e45518328ad3a607e3764af5f93
│     
└─UEFI Device Firmware:
  │   Device ID:          cbbeb6b445a4e88670e0fff8637e5fa32bfdefcd
  │   Summary:            UEFI System Resource Table device (Updated via capsule-on-disk)
  │   Current version:    0x000000cf
  │   Vendor:             Framework (DMI:INSYDE Corp.)
  │   Update State:       Success
  │   GUID:               bdffce36-809c-4fa6-aecc-54536922f0e0
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │   Device Requests:    • Message
  │ 
  └─Laptop Intel Core Ultra Retimer23 Controller Update:
        New version:      0x00000270
        Remote ID:        lvfs
        Release ID:       99281
        Summary:          Framework Laptop Retimer 23 Firmware for Intel Core Ultra Mainboards
        License:          Proprietary
        Size:             2.3 MB
        Created:          2024-09-11
        Urgency:          High
          Tested:         2024-10-31
          Distribution:   ubuntu 24.10
          Old version:    0x000000cf
          Version[fwupd]: 1.9.24
          Tested:         2024-09-20
          Distribution:   pop 22.04
          Old version:    0x00000270
          Version[fwupd]: 2.0.0
        Vendor:           Framework
        Duration:         2 minutes
        Release Flags:    • Trusted metadata
                          • Is upgrade
                          • Tested by trusted vendor
        Description:      
        This release makes the following improvements:
        
        • This update adds certified Thunderbolt support.
        • The firmware is signed by Intel
        Checksum:         ec74f1bf929f49c5cee6a70d25acc35d6c713d6a87c02bbdb8e26cac88dfa28f
```

However when I go to try and apply the upgrades, they fail, and i get an EFI Error 07. I assume it is because the updates have already been applied!

## Steps To Reproduce
Steps to reproduce the behavior:
1. Apply the firmware update via the EFI updater (https://downloads.frame.work/bios/Framework_Laptop_13_Intel_Core_Ultra_Series1_capsule_signed_allsku_3.03_3.04_EFI.zip)
2. Allow all the updates to be applied, they should be successful.
3. Boot your machine normally after BIOS is updated.
4. run `sudo fwupdmgr refresh && sudo fwupdmgr get-updates && sudo fwupdmgr update`.
5. Notice the firmware updates are listed! They have not been applied!
6. Agree to update the firmware.
7. Allow machine to reboot.
8. Watch firmware update fail.

## Expected behavior
I should either:
A) Not see any updates available, as they have already been applied.
B) The updates should be successful from fwupdmgr.

## Operating System (please complete the following information):
 - OS/Distribution: Arch Linux
 - Version: Rolling
- Linux Kernel Version: Linux seras 6.12.9-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 10 Jan 2025 00:39:41 +0000 x86_64 GNU/Linux

zquestz, 2025-01-16:
It seems Retimer01 and Retimer23 didn't update correctly, as those are the only two updates that show.

7heMC, 2025-01-16:
I was having the same issue with the Retimer23 update on Fedora after running the Bios update. In addition to this, the modules on the left side of laptop completely stopped working, which brought me to [another recent Framework Bios Update post](https://community.frame.work/t/12th-gen-intel-core-bios-3-08-release/43244) for a different mainboard. However, I tried the following tip that was mentioned:

> Troubleshooting:
> If you experience ports not working after your update. Please shutdown, unplug all power sources, wait 90 seconds, and then power on again.

When I restarted, I ran the `sudo fwupdmgr update` command again, and it worked.

zquestz, 2025-01-17:
> I was having the same issue with the Retimer23 update on Fedora after running the Bios update. In addition to this, the modules on the left side of laptop completely stopped working, which brought [...]

THIS WORKED!

Turned off my laptop, disconnected the battery, then waited a few minutes. Then put the battery back in, booted into Arch, then ran fwupdmgr and it worked!

closed 2025-01-17 by zquestz