FrameworkComputer/SoftwareFirmwareIssueTracker #41: FW16 Freeze then Reboot  (FTR)

jcdutton, 2025-02-02:
## Device Information

### System Model or SKU
 [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)
No dGPU.

### BIOS VERSION
3.0.5

Windows:
N/A

Linux: 
Open a terminal and run the following command
```sudo dmidecode --string bios-version```
03.05

### DIY Edition information
Memory: Manufacture and SKU
Kingston Fury Impact: Part Number: KF556S40-32
2x making 64GB total.
Storage: Manufacture and SKU
Model Number:                       WD_BLACK SN850X 1000GB
Firmware Version:                   620361WD

### Port/Peripheral information
1. USB-C card, nothing plugged in.
2. Empty
3. Empty
4. Empty
5. USB-C card, FW16 PSU plugged in.
6. USB-A card, nothing plugged in.

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] No

## Describe the bug
Occasionally, about once a month I get a random crash/freeze then about 20 seconds later a reboot.
There is some details in this community thread:
https://community.frame.work/t/frwk16-random-crash-then-reboots/62411/31

## Steps To Reproduce
Steps to reproduce the behavior:
1. Start from a powered off laptop.
2. Power on laptop
4. Wait a random amount of time. Play videos, netflix, youtube etc.
5. System freezes for about 20 seconds and then reboots itself.

## Expected behavior
It should not randomly freeze then reboot.  (FTR)

## Screenshots
N/A

## Operating System (please complete the following information):
 - OS/Distribution: Linux/Ubuntu
 - Version: 24.04
- Linux Kernel Version: `uname -a`  6.12.7  <- Mainline compiled kernel.

## Additional context
Add any other context about the problem here.

jcdutton, 2025-02-02:
There has been an interest in understanding the port80 codes.
On the FW16 AMD, these are 32bit values, and not 8bit values.

I attach some files, that contain the PORT80 codes output during a normal reboot and a normal cold-boot. They have been sorted and de-duped to shorten the list of codes.

It would be helpful if FW support could identify what each code means.
When the problem next occurs, I will be capturing the PORT80 codes and posting them here.

[port80-reboot-de-dupe1.txt](https://github.com/user-attachments/files/18632432/port80-reboot-de-dupe1.txt)
[port80-cold-boot-de-dupe1.txt](https://github.com/user-attachments/files/18632431/port80-cold-boot-de-dupe1.txt)

jcdutton, 2025-02-02:
People are seeing 3 different scenarios:

1. an instant power off. Needing power button to switch on.
2. a system pause/blank screen, for about 20 seconds, then a reboot.

In both cases, no logs or crash dumps to help diagnose the problem.

3. the screen goes blank, but does not power off and does not reboot.

All three are difficult to reproduce.
I see (2) about once a month.
If people are seeing (3), that is just a gpu driver bug, so gather crash traces and report to amd.
If people see (1), please raise a different Issue, and don't mix it with this one.
This issue is only for "Freeze then Reboot"  issues. Not "Freeze then power off".

jcdutton, 2025-02-02:
As an aid to people trying to understand the EC console output.
E.g. [66390.711600 HC 0x0115 err 1]
#define EC_CMD_PD_GET_LOG_ENTRY 0x0115

Please see attached file for the definition of each HC (Host command to EC) 

[ec_commands.txt](https://github.com/user-attachments/files/18632535/ec_commands.txt)

jcdutton, 2025-02-03:
When comparing multiple port80 traces, from say, repeating a "reboot", and then comparing the sequence of port80 values.  It appears to be that the capture of port80 values by the EC is not perfect. There a bit errors and missed values.
I.e. Sometimes the value is absent in the sequence compared to a different reboot.
Also, sometimes some bits are set in the 32bit value, and other times it is not set. This would hint at bit errors being introduced.
e.g.
ef008888
vs
00008888
vs
b0008888
The "ef" has bits set when maybe they should not have been set.
The "b0" has bits set when maybe they should not have been set.

I will see if there is some bug in the EC code or something else causing the problem.

PureKrome, 2025-02-03:
[image]

jcdutton, 2025-02-04:
During a normal boot (I.e. not a FTR):
Some of the port80 codes are interesting:
Here it is using lots of Port80 writes to say:
"Entering Vbios Boot Load%r"
Also, with what looks like two devices trying to write to port80 at the same time.
Notice the "%" instead of the "e"

00000145 | 5f535452 | _STR
00000146 | 0000000a | ....
00000147 | 00000045 | ...E
00000148 | 0000006e | ...n
00000149 | 00000074 | ...t
0000014a | 00000065 | ...e
0000014b | 00000072 | ...r
0000014c | 00000069 | ...i
0000014d | 0000006e | ...n
0000014e | 00000067 | ...g
0000014f | 00000020 | ...
00000150 | 00000056 | ...V
00000151 | 00000062 | ...b
00000152 | 00000069 | ...i
00000153 | 0000006f | ...o
00000154 | 00000073 | ...s
00000155 | 00000020 | ...
00000156 | 00000042 | ...B
00000157 | 0000006f | ...o
00000158 | 0000006f | ...o
00000159 | 00000074 | ...t
0000015a | 00000020 | ...
0000015b | 0000004c | ...L
0000015c | 0000006f | ...o
0000015d | 00000061 | ...a
0000015e | 00000064 | ...d
0000015f | 00000065 | ...e
00000160 | e825fe00 | .%..
00000161 | 00000072 | ...r
00000162 | 00000020 | ...
00000163 | 0000002e | ....
00000164 | 0000002e | ....
00000165 | 0000002e | ....
00000166 | 0000000a | ....
00000167 | 5f454e44 | _END

jcdutton, 2025-02-07:
The following file contains the Port80 codes during a reboot of the Linux operating system normally. I.e. User clicked on "reboot".  The purpose of this is to
1) Share them
2) Maybe discover what some of them mean.
3) There are some comments in the spreadsheet, of the values I know already.
4) Once I see a FTR, I can then compare that bad one with this one.

[port80-reboot10.ods](https://github.com/user-attachments/files/18710596/port80-reboot10.ods)

jcdutton, 2025-02-09:
The changes I have made to the EC and ectool are here:
https://github.com/jcdutton/ectool/commits/main/
https://github.com/jcdutton/EmbeddedController/commits/fwk-lotus-azalea-19573
https://github.com/jcdutton/zephyr/commits/lotus-zephyr/

jcdutton, 2025-02-10:
My updates to the ECTool.efi  reflash     program, to make it slightly safer to use.
It defaults to doing nothing unless the command line arguments are exactly right.
https://github.com/jcdutton/FrameworkHacksPkg/commits/main/

I only ever write to the RW section, never to the RO section, so that if something goes wrong, it falls back to the FW provided EC firmware.

PureKrome, 2025-02-11:
@sinatosk just to confirm, you DON'T HAVE the AMD RX7700S in the extension port thiny? This is all occurring on your iGPU?

sinatosk, 2025-02-11:
> [@sinatosk](https://github.com/sinatosk) just to confirm, you DON'T HAVE the AMD RX7700S in the extension port thiny? This is all occurring on your iGPU?

correct

jcdutton, 2025-02-11:
Another observation.
One can force an OS crash using "echo c >/proc/sysrq-trigger".
On my system (Ubuntu) that causes the system to freeze the screen and halt the system.  No reboot happens (I waited 120 seconds and no reboot). 
FTR is doing freeze, wait about 20-30 seconds, then reboots.
So, the FTR is not following the configured Linux crash configuration for me.
I can only assume from this that the FTR is not OS related and does the whole Freeze and reboot without the OS knowing about it, otherwise, on my system, it should not have rebooted.

jcdutton, 2025-02-11:
@sinatosk 
Thank you for the information. I will try setting my system to powersave, 165Hz display in the hope that I will see the FTR more often.
Up to now, I have been on "balanced" power mode.

sinatosk, 2025-02-11:
> Up to now, I have been on "balanced" power mode.

try it in DC mode ( battery ) too please

jcdutton, 2025-02-12:
I have some more info that might help track this down. The quote is from someone at AMD:
Quote starts----
Another thing that is really useful is that there is a register in FCH
called S5_RESET_STATUS.  If you can get the value from it when this
fails it can point you at where the issue is.  I don't think this patch
landed, but you can see if it works for your system to print the info.

https://marc.info/?l=linux-i2c&m=168089982408414

There should be some public documentation on interpreting
S5_RESET_STATUS somewhere, but it's slipping my mind where it is.
---Quote ends.

So, If anyone wishes to compile their own kernels and help with this problem, its very worth while applying the above patch.

sinatosk, 2025-02-12:
I'll try that on the Saturday, busy tomorrow and Friday

sydneymeyer, 2025-02-12:
I have a 13" FW AMD and a longstanding issue with similar symptoms (20s freeze -> reboot, no logs), although mostly after resume from suspend.

I now have 6.12.13 compiled with the proposed patch and after a reboot ( and an unrelated hard reset in between :), i find:

```
Feb 13 00:05:18 fw kernel: S5_RESET_STATUS = 0x00080800
-- Boot dc7cb0856fbb40cfa5171be88a9547ba --
Feb 13 00:07:57 fw kernel: S5_RESET_STATUS = 0x00200800
-- Boot 3106ceac8bb34769bc02d093bcc8dcf4 --
-- Boot 495ba047b880414f87d5461dbeffa2bb --
-- Boot bb72a980dcb24cf883187281825ecaf6 --
Feb 13 00:15:43 fw kernel: S5_RESET_STATUS = 0x00080800

```

Would this (output after of the next freeze/reboot cycle, oc) be of any help, albeit a FW 13 AMD?

jcdutton, 2025-02-13:
Some BIOS Port codes can be found here:
They are from an older CPU, but some might be appropriate.
https://github.com/jcdutton/amd_firmware_binaries/blob/main/picasso/PSP/AblPostCode.h.txt
It covers 32bit Port codes from the PSP ABL  with prefix: 0xEA00XXXX   where XXXX is the code listed in the link above.

PureKrome, 2025-02-13:
[image]

sydneymeyer, 2025-02-14:
FWIW, perhaps this can help narrow down the cause or help others being plagued by this issue.

I experienced this (FTR) issue since i received the laptop (dec 23) every 3-5 days and around September 2024, i contacted FW support, who suggested to perform a "mainboard reset" by the instructions following below.

After this "mainboard reset", which i suppose resets some kind of nvram on the board/ec (obviously, i don't understand all this low level stuff well enough), but since this reset to this day (around 4-5 months), i had around 3-4 FTR cycles so far (which could as well be an whole different issue), but the regular, every few days, FTR issue was/is, apparently/hopefully gone.

Considering this, it may take a while for this to happen again here, but if it does, i'll remember and post the value in this issue.

> According to wiki here the AMD 7840HS CPU is a phoenix CPU.

It's a 7840u here. I'd consider this to be a different layout of the same architecture.

Quote Instructions from Framework Support:

Please perform a full mainboard reset. This will clear up any states that are saved. Kindly follow the instructions below and see how it goes.

Plug in the system to AC.
Remove the Input Cover.
Press the chassis open switch in the center of the Mainboard 10 times, you must press it slowly, so press for 2 seconds. Release, wait for the red blink on the Mainboard LEDs. repeat.
Press the power button to boot the system
BIOS settings will be reset to defaults.

YummYume, 2025-02-15:
@sydneymeyer So do you suggest people experiencing this issue to try this as well ? It will only reset the BIOS settings ?

sydneymeyer, 2025-02-15:
@YummYume This is what Framework Support is suggesting people experiencing this issue to do and what had the effect for me i described above.

> It will only reset the BIOS settings ?

Well, it resets the BIOS to factory defaults, e.g. if you have set up Secure Boot with your own keys, you might have to re-enroll them. Perhaps you also might have to restore the UEFI-Boot Entries, can't remember. But other than that, yes.

YummYume, 2025-02-15:
@sydneymeyer Thanks, I'll probably try doing that tomorrow and see if it fixes anything. It's weird because I haven't touched my BIOS much anyway other than changing the max charging power for the battery.

jcdutton, 2025-02-16:
Just a comment. This issue ticket is aimed at finding the cause of the problem. So hiding it with a main board reset etc. is not the aim of this issue ticket. I.e. we are looking for a root cause.

sinatosk, 2025-02-16:
Hi, didn't get a chance to do it yesterday as I said I would. I'll apply the patch tomorrow ( after linux 6.14-rc3 is released )

I'll then try to trigger the issue ( if possible ) over the week or two

sydneymeyer, 2025-02-19:
> Just had a screen freeze didn't go black but it also didn't restart so I held power button to force reboot

@sinatosk This is explicitly not what @jcdutton kindly asked about.

Please read:

https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/41#issuecomment-2629402351

jcdutton, 2025-02-19:
Yes, the screen freeze but didn't restart is probably a different bug. But still useful to hear that @sinatosk is still trying.
I think I am going to try to link up a sysrq key to force a AMD GPU reset. I.e. the equivalent of that "cat" command that forces a reset.

sydneymeyer, 2025-02-21:
Just had my obilgatory FTR cycle.

`Feb 21 12:33:11 fw kernel: S5_RESET_STATUS = 0x08000800`

jcdutton, 2025-02-21:
Ah, that is helpful, a new S5_RESET_STATUS code we have not seen before: S5_RESET_STATUS = 0x08000800

jcdutton, 2025-02-21:
Some explanations from AMD:

Normal values:
Cold Power off / Power on:
S5_RESET_STATUS = 0x00200800
Answer: System exited S0i3, ACPI S3, ACPI S4, or ACPI S5.

Warm reboot:
S5_RESET_STATUS = 0x00080800
Answer: System was reset by CF9 = 0x06.

Keeping your finger of the power button for 10 seconds to force the power off.
S5_RESET_STATUS = 0x00000800
Answer: True "Cold boot".

Suspend:
Nothing output.

Value seen during a Freeze-then-reboot (FTR):
S5_RESET_STATUS = 0x08000800
Answer: Sync flood.

If it helps, the above S5_RESET_STATUS = 0x08000800
was seen on a:
Framework Laptop 13 powered by AMD Ryzen™ 7040 Series Processors:
Ryzen 7 7840U and Ryzen 5 7640U

Explanation from AMD:
Bit 27 means this is a reboot due to "sync flood".  Here's a description:

A drastic method of reporting errors in which a device transmits Sync
packets continuously until reset occurs. Each device on the chain that
detects the Sync flood repeats the pattern on all links. A reset is
required to recover from a Sync flood. Sync flood is analogous to SERR#
(System Error) on a PCI bus.

Debugging this will need to be done with a custom BIOS that disables
reboot on Syncflood so that the full state of the system could be
analyzed to characterize it.

In short; you unfortunately won't be able to debug this any further
yourself; it needs to be done by Framework BIOS team.

The best thing you as an end user could do is find the pattern of
activity that's tripping this (if there is one) so that Framework is
able to reproduce it.

sydneymeyer, 2025-02-21:
@jcdutton It would be good to have some form of communication from Framework themselfs, if they are aware of the issue or even working on it. Framework Support insisted that i'll record the issue on video, which meant for me having to film my laptop everytime i opened it. Tons of logs and multiple requested tests, just to get told (_after_ i submitted everything requested, of course) that NixOS is not on the list of supported linux distributions. Quite a disappointing experience, to be frank. If it leads _somewhere_, e.g. like this effort here, i'd be willing to work with Framework Support again, but i simply cannot invest the amount of time atm without knowing that someone is indeed working/acknowledging this issue. Perhaps they delayed the upcoming bios update because of this issue. Who would know.

PureKrome, 2025-02-21:
@sydneymeyer oh my - so dissapointing (from FW). Surely the "NixOS not supported" is a seriously lame excuse hiding a bigger problem -- we don't have the resources and as such we'll use that excuse.

wow! do i want to start rethinking my use of purchasing more FW lappies? If they will struggle to fix it - fine. I'm down with that. But I _just feel like_ this is not a transparent and honest answer to this situation.

I hope i'm just cynical and misreading this situation.

jcdutton, 2025-02-21:
But, just to be clear. We now only have an error sample size of 1.
I would very much like more reports than just one.

jcdutton, 2025-02-21:
@sydneymeyer 
Did you keep note of which cards were in what slots at the time of the FTR, and also what you were doing at the time.
Anything that helps us reach an easier way for the FW team to see the problem for themselves will be helpful.

sydneymeyer, 2025-02-21:
The issue occurs exclusively after resume from suspend. The Laptop is mostly connected to a USB-C powered dock with HDMI, Ethernet and a Apple Magic Trackpad 2 attached. FTR cycles have occured while connected to a dock (connected before suspend, while sleeping and when resuming), after disconnecting the dock while sleeping and resuming standalone, and completely standalone (i.e. suspend/resume cycles without attached dock).

Next boot, the last logged line in the system logs is usually from suspending the laptop (via systemd), nothing else. "ectool panicinfo" gives no "no panic data".

OS is NixOS with respective stable kernel. Programs open are usually sway (vulkan), Firefox (w/ vaapi enabled), Thunderbird, mpv (vulkan), foot, vim, cmus with pipewire, networkd, iwd.. More or less.

Unfortunately, i have no way of reproducing the issue or have recognized a pattern other than what's described above. 

- Framework 13 AMD 7840U DIY
- 2x16 GB DDR5-5600 Crucial Memory (CT2K16G56C46S5)
- 2TB WD SN850X (FW 620361WD)
- Slot 1 USB-C (back left)
- Slot 2 USB-A (front left)
- Slot 3 USB-C (back right)
- Slot 4 USB-A (front right)
- Original Framework 13 USB-C charger
- BIOS 3.05
- Secure Boot enabled
- TPM not used
- 60% Battery charge limit
- "Gaming Mode" enabled (4GB VRAM alloc.)

jcdutton, 2025-02-22:
I just had a FTR.
S5_RESET_STATUS = 0x08000800

sydneymeyer, 2025-03-02:
Mar 02 15:45:20 fw kernel: S5_RESET_STATUS = 0x08000800, like usual, i.e. while resuming from suspend, no logs at all.

helpimnotdrowning, 2025-03-03:
I was linked here from the forums (https://community.frame.work/t/sudden-reboot-on-wake-with-no-logs/65434 ) and I'm experiencing a similar issue with my Framwork 13 / R5 7640U. I applied the S5_RESET_STATUS patch ([with light modification](https://community.frame.work/t/sudden-reboot-on-wake-with-no-logs/65434/6 )), and got a seemingly-new status code `0x00800800` that hasn't been mentioned here before. Is there anything more I can do on my end?

jcdutton, 2025-03-03:
I have found a message that contains something helpful.
https://lore.kernel.org/all/80dbe1de-c71c-4556-817d-3f06e67f38ba@amd.com/

In that message, there is a link to this URL:
https://www.amd.com/content/dam/amd/en/documents/processor-tech-docs/programmer-references/55901_B1_pub_053.zip

In that zip file there are some PDFs.
In the PDF:
55901_B1_pub_5.pdf
There is an explanation regarding each bit in the S5_RESET_STATUS
Extract included here:
PMx000000C0 (FCH::PM::S5_RESET_STATUS)
Reset: 0000_0800h.
_aliasHOSTLEGACY; PMx000000C0; PM=FED8_0300h
Bits Description
31: sw_sync_flood_flag. Read-write,Read,Write-1-to-clear. Reset: 0. PMxC0[31] will be set if sw_sync_flood (PMx88[10]) trigger reset. Write 1 to clear. Bit[31] and Bit[28:16] except bit[20] will be cleared by Last reset event except the associated bit will be set.
30: sdp_parity_err. Read-write,Read,Write-1-to-clear. Reset: 0.
Description: When there is 'parity error', Sync_Flood reset will occur and PMxC0[27] will be set if enabled, in order to distinguish 'parity error' and 'CPU sync flood', PMxC0[30] will be set when there is SDP parity Error, thus software can distinguish 'parity error' and 'CPU sync flood'.
This bit will not be cleared by other reset event, software need write 1 to clear. SDP parity error will not clear any status bit in this register.
29: mp1_wdtout. Read-write,Read,Write-1-to-clear. Reset: 0. This bit will be set to 1 when MP1_Watchdog timer time out (this indicates there was a failed warm reset handshake between SMU and FCH). This bit will not be cleared by other reset event, software need write 1 to clear. MP1_Watchdog timer time out will not clear any status bit in this register.
28: Reserved.
27: sync_flood. Read-write,Read,Write-1-to-clear. Reset: 0. system reset was caused by a SYNC_FLOOD event which was due to an UE error( when PMx74[18]=1). Write 1 to clear. Bit[31] and Bit[28:16] except bit[20] will be cleared by Last reset event except the associated bit will be set.
26: remoteresetfromasf. Read-write,Read,Write-1-to-clear. Reset: 0. system reset was caused by a remote RESET command from ASF. Write 1 to clear. Bit[31] and Bit[28:16] except bit[20] will be cleared by Last reset event except the associated bit will be set.
25: watchdogissuereset. Read-write,Read,Write-1-to-clear. Reset: 0. system reset was caused by MICROSOFT WatchDog Timer. Write 1 to clear. Bit[31] and Bit[28:16] except bit[20] will be cleared by Last reset event except the associated bit will be set.
24: failbootrst. Read-write,Read,Write-1-to-clear. Reset: 0. system reset was caused by AMD Fail boot timer. Write 1 to clear. Bit[31] and Bit[28:16] except bit[20] will be cleared by Last reset event except the associated bit will be set.
23: shutdown_msg. Read-write,Read,Write-1-to-clear. Reset: 0. system reset was caused by a SHUTDOWN command from CPU (when PMx08[20]=1 and PMx74[17]=1). Write 1 to clear. Bit[31] and Bit[28:16] except bit[20] will be cleared by Last reset event except the associated bit will be set.
22: kb_reset. Read-write,Read,Write-1-to-clear. Reset: 0. system reset was caused by assertion of KB_RST_L.. Write 1 to clear. Bit[31] and Bit[28:16] except bit[20] will be cleared by Last reset event except the associated bit will be set.
21: sleepreset. Read-write,Read,Write-1-to-clear. Reset: 0. Reset status from Sleep state (Power saving mode, S3, 4, or 5) transition. Write 1 to clear. Bit[31] and Bit[28:16] except bit[20] will be cleared by Last reset event except the associated bit will be set.
20: do_k8_full_reset. Read-write,Read,Write-1-to-clear. Reset: 0. Description: system reset was caused by CF9 = 0x0E. Write 1 to clear.
[Note] Write CF9=0xE will set this bit=1, but write CF9=0xE will generate SLpRst later which will set bit[21]=SleepReset. In order to keep this bit =1, this bit will not be cleared by hardware, software need to write 1 to clear this bit. Bit[31] and Bit[28:16] except bit[20] will be cleared by Last reset event except the associated bit will be set.
19: do_k8_reset. Read-write,Read,Write-1-to-clear. Reset: 0. system reset was caused by CF9 = 0x06. Write 1 to clear. Bit[31] and Bit[28:16] except bit[20] will be cleared by Last reset event except the associated bit will be set.
18: do_k8_init. Read-write,Read,Write-1-to-clear. Reset: 0. system reset was caused by CF9 = 0x04. Write 1 to clear. Bit[31] and Bit[28:16] except bit[20] will be cleared by Last reset event except the associated bit will be set.
17: soft_pcirst. Read-write,Read,Write-1-to-clear. Reset: 0. system reset was caused by writing to PMIO 0xC4[0] (PciReset). Write 1 to clear. Bit[31] and Bit[28:16] except bit[20] will be cleared by Last reset event except the associated bit will be set.
16: usrrstb. Read-write,Read,Write-1-to-clear. Reset: 0. Last reset was caused by BP_SYS_RST_L assertion. Write 1 to clear. Bit[31] and Bit[28:16] except bit[20] will be cleared by Last reset event except the associated bit will be set.
15:14: pmeturnofftime. Read-write. Reset: 0h.
Description: 
00: 1ms
01: 2ms
10: 4ms
11: 8ms

13:10: Reserved.
9: intthermaltrip. Read-write,Read,Write-1-to-clear. Reset: 0. system was shut down due to an internal ThermalTrip event. Write 1 to clear 
8:5: Reserved.
4: remotepowerdownfromasf. Read-write,Read,Write-1-to-clear. Reset: 0. SOC has received a remote Power Off command from ASF. Write 1 to clear.
3: Reserved.
2: shutdown. Read-write,Read,Write-1-to-clear. Reset: 0. system was shut down due to ShutDown event (SHUTDOWN# pin). Write 1 to clear.
1: pwrbtn4second. Read-write,Read,Write-1-to-clear. Reset: 0. system was shut down due to 4s PwrButton event. Write 1 to clear.
0: thermaltrip. Read-write,Read,Write-1-to-clear. Reset: 0. system was shut down due to BP_THERMTRIP_L assertion. Write 1 to clear.

jcdutton, 2025-03-03:
So, S5_RESET_STATUS = 0x00800800  means:
Bit 11 set. - Reserved
Bit 23 set - 23: shutdown_msg. Read-write,Read,Write-1-to-clear. Reset: 0. system reset was caused by a SHUTDOWN command from CPU (when PMx08[20]=1 and PMx74[17]=1). Write 1 to clear. Bit[31] and Bit[28:16] except bit[20] will be cleared by Last reset event except the associated bit will be set.

So, this implies the CPU caused the shutdown.
My guess therefore, is its due to a software bug, rather than anything more serious.

jcdutton, 2025-03-03:
Also from the PDF document linked above:

Shutdown (message from CPU):
Triple faults in CPU will cause an internal SHUTDOWN message
broadcasted. FCH will generate a reset to S0 logic; configurable to warm or cold reset.

SYNC_FLOOD (message)
Internal data fabric logic detects an error (eg. parity error) and broadcasts an internal SYNC_FLOOD message. FCH will generate a reset to S0 logic; configurable to warm or cold reset.

helpimnotdrowning, 2025-03-05:
> > configurable to warm or cold reset.
> 
> Does this (warm reset) mean the system, i.e. the EC/Kernel/CPU (obviously, i don't get it), can/could recover from this happening?

It probably refers to https://en.wikipedia.org/wiki/Reboot#Cold_versus_warm_reboot , the difference seeming to be whether the system performs a complete [POST](https://en.wikipedia.org/wiki/Power-on_self-test) sequence or just a small part. Doesn't sound recoverable unfortunately .

CDRXavier, 2025-03-05:
This seems vaguely familiar ...
[Sleep/Hibernate issue with Windows 11](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/35)

Since I am running primarily on Windows, extensive Linux testing is unlikely. However, if there are things I can do to help either of those issues.
 
Is there a way for me to like, tap into the bus? There is a EC header, and I have logic analyzers. I can solder wires to onboard headers and monitor it externally.

jcdutton, 2025-03-05:
@CDRXavier 
Although the symptoms are similar when sleeping, the cause is very different.
1) Random FTR not related to sleep results in a 0x08000800 <- Sync Flood <- More hardware / BIOS related.  I have been told by AMD that we, users, cannot do anything further with this. It needs special BIOS code and board analysis equipment to diagnose it further.
2) Reboots after sleep appear to be different, resulting is a 0x00800800 <- Shutdown msg -- Triple Fault - More software related.
These ones need debugging the kernel code to track down the triple fault.
Note, triple faults are notoriously difficult to track down.
We are only focusing on (1) in this ISSUE ticket.
We would need a new ISSUE ticket for discussing (2)

PureKrome, 2025-03-06:
> We are only focusing on (1) in this ISSUE ticket.
We would need a new ISSUE ticket for discussing (2)

@jcdutton  Would you like me to make a new ticket for (2) because that was the cause of me creating the conversation in the support forums which has had a lot of convo in it. I felt like (2) is more common pain point for everyone.

WillNilges, 2025-03-06:
@PureKrome I would join you in that ticket. I am almost certain that my issue is (2), but since installing 6.13.5 with the recommended patch yesterday I haven't been able to repro. I've seen it a lot between 6.12 and 6.13.4. Anyone seen it on 6.13.5 yet?

In case I haven't mentioned it yet I have a FW 13 Ryzen 7 7840U with these cards:
1. USB-C
2. USB-A
3. USB-C
4. HDMI

WillNilges, 2025-03-07:
CONFIRMED! My issue is (2)

https://community.frame.work/t/spontaneous-and-ungraceful-restarts-upon-closing-the-lid/64806/13?u=will_nilges

jcdutton, 2025-03-07:
Hi All,
I am the original poster of this ISSUE.

1) Random FTR not related to sleep results in a 0x08000800 <- Sync Flood <- More hardware / BIOS related.  I have been told by AMD that we, users, cannot do anything further with this. It needs special BIOS code and board analysis equipment to diagnose it further.
2) Reboots after sleep appear to be different, resulting is a 0x00800800 <- Shutdown msg -- Triple Fault - More software related.
These ones need debugging the kernel code to track down the triple fault.
Note, triple faults are notoriously difficult to track down.

We are only focusing on (1) in this ISSUE ticket.
I have only seen (1) on my FW16. I have not seen (2).

Please raise a new issue for discussing (2)

WillNilges, 2025-03-08:
@PureKrome @jcdutton I've done it: https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/50

Can we work on centralizing folks around this issue? Please spread the word. I would really like to hear back from a Framework engineer because I cannot use my laptop because of this.

sydneymeyer, 2025-03-15:
> This S5_RESET_STATUS patch should apply to kernel 6.13.5 kernel:
> 
> [S5_RESET_STATUS-6.13.5.txt](https://github.com/user-attachments/files/19091396/S5_RESET_STATUS-6.13.5.txt)

FWIW, i'm now running 6.14-rc6 with this patch.

sydneymeyer, 2025-04-08:
Should we continue to report every instance of this (S5_RESET_STATUS = 0x08000800) happening, or is there anything else we can do regarding this issue?

To be frank, i don't see Framework taking _any_ interest in this at all (except from what appears to be "managing the community"),  and i have no desire to compile the kernel from scratch everytime i'm doing updates forever until Framework feels like it.. or perhaps not. Who would know.

Besides, there appears to be nothing new to report, always the same, already reported many times, circumstances.

sydneymeyer, 2025-05-26:
@jcdutton This [1] one might prove to be immensely helpful. IIRC it has gone in for 6.16.

[1] https://patchwork.kernel.org/project/platform-driver-x86/patch/20250410200202.2974062-5-superm1@kernel.org/

jcdutton, 2025-05-27:
@sydneymeyer 
Yes, it just a little more friendly version of the S5_RESET_STATUS one we are already using.

sydneymeyer, 2025-05-27:
@jcdutton Yes, but when it is included in mainline, it will give the issue much more visibility, considering the amount of "sudden reboot" issues i've read about with Framework laptops so far.

After xx-many emails with Framework support i apparently have not yet even been able as to convey the simple fact to them that it is not me rebooting/hard resetting the computer, but that the laptop resets itself all on its own. So, i'm not too confident on that front.

WillNilges, 2025-06-09:
> [@jcdutton](https://github.com/jcdutton) Yes, but when it is included in mainline, it will give the issue much more visibility, considering the amount of "sudden reboot" issues i've read about with [...]

If you haven't already, you can explicitly ask for your ticket to be escalated to engineering. They might be able to dig into it better. I would be curious to read your transcripts, as would the folks on https://linustechtips.com/forum/98-lmg-sponsor-discussion/ >:)

sinatosk, 2025-09-20:
Framework 16 AMD Ryzen 7 7840HS using Radeon 780M
RAM/Memory: [128GB ( 2x64GB ) - Crucial 128GB Kit (64GBx2) DDR5-5600](https://uk.crucial.com/memory/ddr5/ct2k64g56c46s5)
NVME 2280: [Western Digital SN850X 2TB - Firmware 620361WD](https://shop.sandisk.com/en-gb/products/ssd/internal-ssd/wd-black-sn850x-nvme-ssd?sku=WDS200T2X0E-00BCA0)
NVME 2230: [Western Digital SN770M 2TB - Firmware 731120WD](https://shop.sandisk.com/en-gb/products/ssd/internal-ssd/wd-black-sn770m-nvme-ssd1?sku=WDBDNH0020BBK-WRSN)
BIOS: 3.07
Gentoo Linux 2.17 ( Linux 6.17-rc6 mainline realtime, compiled by clang 21.1.1 march and mtune set to znver4)
KDE Plasma 6.4.5 Wayland

Well, with BIOS 3.06 and now 3.07 ( earlier today ), I'm still getting FTR issues

> 86/amd: Previous system reset reason [0x08000800]: an uncorrected error caused a data fabric sync flood event

The last three FTR's I've had ( one each, BIOS 3.05, 3.06 and 3.07 ) all happened after resuming ( battery or power adapter ).

While not related to this issue, I'm still getting [FTH's](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/58), which seem to happening during usage regardless of the load and performance profile

jcdutton, 2025-12-31:
I have made a new observation.
When the FTR happens, the Mediatek Wifi card fails to function well after the reboot.
This makes me thing that maybe the Wifi card might be causing the FTR.

Please can whoever is seeing this problem, also mention which Wifi card make/model they are using.
Now, I know the FW16 automatically comes with the Mediatek wifi card. But it would be interesting to hear from anyone who has changed the wifi card, whether they see FTR or not.

jcdutton, 2026-01-12:
Another bit of info that might make a difference. Please can people add feedback for these:
1) Do you have a dGPU or only an iGPU ?

2) 
```
sudo dmidecode |grep Version
	Version: 03.05
	Version: A7
	Version: A7

```
I am interested in whether the Version says A7 or AG.

3) In amdgpu_top,  does Tctl ever reach 100 C, when running stress-ng or something to use the CPU a lot.

sydneymeyer, 2026-01-13:
1. iGPU only (780M)
2.
```
sudo dmidecode |grep Version
	Version: 03.17
	Version: A7
	Version: A7
	Version: A7
	Version: AMD Ryzen 7 7840U w/ Radeon  780M Graphics     
	Firmware Version: Unknown
	Firmware Version: Unknown
	SBDS Version:      
	String: Compiler Version: VC 9.0
	Specification Version: 2.0
```
3. Not that i'm aware of. (~95 C° at ~ 32-35W until fan ramps up to ~6.7K RPM / FW 13 AMD 7840U)

@jcdutton N.B: Last time i hit this (0x08000800) was in may last year. Literally not a single occurance since then (fingers crossed).

jcdutton, 2026-01-14:
There is a new question regarding a possible cause of this problem. That is RAM temperatures?
The reports seem to highlight that the problem occurs often while playing videos. For me, it has only occurred during playing videos. As my FW16 AMD 7840HS is an APU, where the iGPU uses the CPU RAM for the display. It is fairly obvious, therefore, that when playing a video, the RAM is being used a lot for each and every video frame displayed, while the CPU use tends to be low at the time.
So, a theory now, is that the problem is either the RAM chips getting hot, or the data path between the RAM chips and the CPU causing the problem.
To this end, I have started logging to a .csv, the RAM chip temperatures, in the hope that when the FTR next happens, I will have some data showing that the RAM temp was just before the FTR.
So far, I have not managed to get the RAM temperature above 60C, but others might succeed.

Summary: please capture RAM temperature, just before or just after, when the FTR happens.

[17 more replies]