FrameworkComputer/SoftwareFirmwareIssueTracker #42: Cannot boot from partially locked self-encrypting drives

callahad, 2025-02-04:
👉 Ported from Framework Community post [Unable to boot from write-locked OPAL SED SSD](https://community.frame.work/t/unable-to-boot-from-write-locked-opal-sed-ssd/52813)

## Device Information

### System Model or SKU

**All models likely affected.** Specifically confirmed on:

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [x] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [x] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION

**All versions likely affected.** Specifically confirmed on:

- FW13: 3.08
- FW16: 3.03, 3.05

### DIY Edition information

**All TCG Opal 2.0 Self-Encrypting Drives likely affected.** Specifically confirmed on:

Storage: Samsung 990 Pro, Samsung 980 Pro, Samsung PM961

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug

TCG Opal drives can selectively lock different regions of the disk. For example, the `/boot` partition can be left in cleartext, while the `/` partition is encrypted.

In these setups, sufficient data is available to boot the system, but the Framework BIOS refuses to even attempt to boot, instead trapping the user at a "Harddisk Security" prompt. Laptops from other manufacturers _are_ able to boot from the same drive in the same configuration.

## Steps To Reproduce

Alex DeLorenzo's blog post [Encrypted Root with LUKS and Opal](https://alexdelorenzo.dev/articles/cryptsetup-luks-self-encrypting-drives) is a thorough discussion of how to create this kind of configuration, but basically:

1. Fully erase and reset a Self-Encrypting Drive: `cryptsetup luksErase --hw-opal-factory-reset /dev/nvme1n1`
2. Create two partitions: A 512 MB EFI System Partition (ESP) to be mounted at `/boot`, and a root partition occupying the rest of the space on the drive.
3. Encrypt the root partition: `sudo cryptsetup luksFormat /dev/nvme1n1p2 --hw-opal-only`
4. Unlock the root partition: `sudo cryptsetup open /dev/nvme1n1p2 root`
5. Format and mount the ESP (`/dev/nvme1n1p1`) and the unlocked root partition (`/dev/mapper/root`)
6. Install Linux per usual
7. Verify that the global locking range (LR0) is disabled, and only a locking range for the root partition is enabled: `sudo sedutil-cli --listLockingRanges $PASSWORD /dev/nvme1n1`
8. Power off the laptop.
9. Power on the laptop.

Note: It may be necessary to invoke `sudo sedutil-cli --setMBREnable off /dev/nvme1n1` to disable the shadow MBR. Check status with `sudo sedutil-cli --query /dev/nvme1n1`.

## Expected behavior

The system should attempt to boot. Whether or not it _succeeds_ is another matter, but it should _try._

## Actual behavior

At power on, the BIOS immediately throws up a "Harddisk security" prompt indicating that the drive is in "(Lock)" status.

If the drive is selected, the BIOS prompts for a decryption key and refuses to proceed.

It does not attempt to boot.

Booting from a Live USB stick shows that the boot partition can be mounted and fully read without unlocking the drive.

Placing the drive in a less sophisticated laptop (a 5th Generation ThinkPad X1 Carbon from 2017) boots up just fine.

## Screenshots

(Taken from https://community.frame.work/t/solved-stuck-on-harddisk-security-at-boot/41088, but the prompts are identical for this issue)

[image]

[image]

## Operating System (please complete the following information):
- Linux (all)

## Additional context

This seems like the Framework BIOS is trying to be **too smart** when it handles TCG Opal drives: If it sees _any_ locked ranges, it refuses to proceed, even if the partition it needs to boot from is not locked.

If there was a way to disable or bypass attempts from the BIOS to handle unlocking, that should solve this issue.

CDRXavier, 2025-03-26:
Interesting.
I have a .. Intel ... SSDPEKNU512GZ, which also support Opal. I think.

However I don't use Linux, so I use simple full-disk encryption. The BIOS is unable to update the disk lock status until a full powe cycle and cold boot.

Might be related.

iandrewt, 2025-10-21:
@JohnAZoidberg is it worth trying again with the 4.01 beta?

RoootTheFox, 2025-12-30:
hi, i'd like to jump in and add that i am having the same issue (FW13, AMD 7040U) using a Samsung 990 PRO (2TB) drive and BIOS version 3.17

my setup: 1GB unencrypted EFI partition followed by a 1.8TB OPAL-encrypted root partition set up through `cryptsetup` - i first thought simply skipping the (incorrectly placed) password prompt should just work (as the boot partition is unencrypted), however this causes the firmware to simply label the *entire* drive as unbootable.
i even tried to disable the firmware's handling of OPAL devices by setting "TCG Storage Action" to "Disable_BlockSIDFunc" in Security > Storage Password Setup Page - however, this option doesn't persist/get saved when changing it.

i am willing to work with the firmware developers to fix this issue, as i'd really like to use partial SED encryption. (if needed, my contact methods are linked on my github profile; matrix, discord and email are preferred)

plabadens, 2025-12-30:
I've run into this issue today too, it would be nice if the firmware could actually just implement the OPAL specification instead of trying to be unnecessarily smart about things. At the very least, an option to not prompt for passphrase at startup would be nice.

quinchou77, 2026-01-13:
We have been working closely with Insyde, and the current behavior is consistent with the Insyde BIOS. 
Following our discussion, we will provide a test version to verify the expected behavior.

quinchou77, 2026-01-21:
We have a new test BIOS available for the Framework Laptop 16 (AMD Ryzen™ 7040 Series). If anyone is interested in helping us validate this build, please let us know. I will follow up with you via email to provide the BIOS files and testing instructions.

KellenRenshaw, 2026-02-04:
Can confirm that the test 3.18 BIOS fixed the issue on my FW13. Still have to press Esc to bypass the unlock screen, but it then boots from the unencrypted part of the disk and allows the OS to unlock the locked portion of the disk.

zraexy, 2026-02-05:
@quinchou77 I appreciate your work on this.

My only concern is:

> Still have to press Esc to bypass the unlock screen

This prevents automatically booting into the initrd for remote unlocking via SSH.
(It doesn't make much sense for a laptop to be used as such, but for a Desktop or laptop mainboard in Standalone Mode, where there may not be a keyboard directly connected, it is a valuable capability to have.)

For that reason, it is still important to have a BIOS setting which completely disables the password prompt.

Daft-Wullie, 2026-02-06:
Happy to test immediately when/if a release is available for the FW 16 Ryzen AI 300 @quinchou77 .
Thanks for the willingness to engage on this!

ThaCheeseBun, 2026-09-26:
Just got my FW 13 Pro Intel Core Ultra Series 3 and ran into this issue aswell. Pressing escape at the unlock prompt does not allow me to boot from the unencrypted partition either :(.