FrameworkComputer/SoftwareFirmwareIssueTracker #46: Laptop Issue

stinson6016, 2025-03-05:
## Device Information
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)

### BIOS VERSION
3.04

## Describe the bug
A clear and concise description of what the bug is.

usb c ports on right hand side gone after bios update to 3.04
got the laptop on friday feb 28, on sunday march 2nd I upgraded the bios. when the 3.04 bios upgrade finished the usb c ports stopped working. had my usb c dock plugged into the right hand side and once the bios upgrade finished the monitors and key board went away. I can swap the ports from the left to the right and everything works on the left side only, nothing works when plugged into the right side.

JohnAZoidberg, 2025-03-05:
Could you please shut your system down, unplug all cables (especially power adapter/dock), let it reset for a minute and then power back on.
Best also if you go into F2 BIOS menu and check BIOS, EC and PD versions.

My guess is that the PD firmware might be broken and you could probably repair it, by running the BIOS update again.

Are you using Windows or Linux?

stinson6016, 2025-03-05:
I am running windows 11 on the laptop.
I tried to rerun the 3.04 bios update on monday or tuesday to see if that would fix anything. I will try the power down in a little bit and check the ec, pd versions.

jcdutton, 2025-03-06:
Hi, I am just another user here.
When upgrading the BIOS, do so without any slot cards plugged in. (i.e. disconnect the dock, USB-A, USB-C, etc.)  Just the FW power brick should be plugged in to the side (using a single usb-c slot card)
Also, see if the EC, PD versions are all at the right levels. Compare the BIOS screens with the BIOS release notes.

Try the BIOS upgrade from the FW power brick plugged into the right had side, and then try again with the FW power brick plugged into the left side.
If that has still not helped.
Unplug the power, turn off the laptop and leave it off, and unplugged for more that 60 seconds.
The power it back on.  This resets the EC as well.