FrameworkComputer/SoftwareFirmwareIssueTracker #47: Can't update Retimer firmware

VickyFrenzy, 2025-03-06:
## Device Information

### System Model or SKU

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [x] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
3.03

### DIY Edition information

Memory: Crucial 2x32Go 5600MHz (CT2K32G56C46S5)
Storage: Samsung SSD 990 PRO 4TB

### Standalone Operation
- [ ] Yes
- [x] No

## Describe the bug
Cannot do step 2 of the 3.04 BIOS upgrade process (step 1 was successful).
```
Update Failed!!!
EFI ERROR CODE: 07
Firmware update failed!!!
```

## Steps To Reproduce
1. Ran Step2_Framework_Laptop_13_Intel_Core_Ultra_Series1_Retimer_port01_624.exe

## Expected behavior
I expected the update to be successful.

## Screenshots
[image]

## Operating System (please complete the following information):
 - OS/Distribution: Windows 11
 - Version: 22631.4890

JohnAZoidberg, 2025-03-07:
What expansion cards have you got plugged in on the right side while doing the update?
Could you try again, unplugging them before you start the update?

VickyFrenzy, 2025-03-10:
> What expansion cards have you got plugged in on the right side while doing the update? Could you try again, unplugging them before you start the update?
I had no expansion cards plugged except for the one for power. I tried both sides.

quinchou77, 2025-03-11:
Do you update the retimer firmware based on BIOS 3.03?

VickyFrenzy, 2025-03-11:
> Do you update the retimer firmware based on BIOS 3.03?

Since I did step one successfully, my bios is 3.0.3

quinchou77, 2025-03-12:
It is the known issue in previous BIOS version. System may report the device error when update the retimer. 
If you want to do the retimer update, we suggest to use the latest version 3.09, which has the fixed for retimer update.
Since the BIOS version is 3.03 on your system, you need to update to 3.06 first, then update to 3.09.

However, the retimer update is optional, if you don't have any problem with tbt ports, you can ignore the retimer update.

VickyFrenzy, 2025-03-12:
> It is the known issue in previous BIOS version. System may report the device error when update the retimer. If you want to do the retimer update, we suggest to use the latest version 3.09, which has [...]

There are no such thing as 3.06 and 3.09 for [Framework Laptop 13 Intel Core Ultra Series 1](https://community.frame.work/t/framework-laptop-13-intel-core-ultra-series-1-bios-3-04-release/59579).
3.04 is the latest available BIOS firmware as seen [here](https://knowledgebase.frame.work/framework-laptop-bios-and-driver-releases-intel-core-ultra-series-1-H1nZQdxYR).

Also the instructions are clear:
> Run the following updates in sequence. After each file is run your system will automatically restart.
> 1. Step1_Framework_Laptop_13_Intel_Core_Ultra_Series1_BIOS_3.03.exe
> 2. Step2_Framework_Laptop_13_Intel_Core_Ultra_Series1_Retimer_port01_624.exe
> 3. Step3_Framework_Laptop_13_Intel_Core_Ultra_Series1_Retimer_port23_624.exe
> 4. Step4_Framework_Laptop_13_Intel_Core_Ultra_Series1_BIOS_3.04.exe

---

> However, the retimer update is optional, if you don't have any problem with tbt ports, you can ignore the retimer update.

Since upgrading to 3.03, I'm having an issue where the power led blink between white and orange while plugged in.

I want to know from the Framework team if it's safe to upgrade to 3.04 directly if the step 2 and 3 are failing _(Retimer updates that won't work)_ and we can't figure out any way to make the Retimer updates work.

VickyFrenzy, 2025-03-12:
I also posted about my issue here: https://community.frame.work/t/framework-laptop-13-intel-core-ultra-series-1-bios-3-04-release/59579/33?u=vicky

quinchou77, 2025-03-17:
Sorry, I mixed up the product categories.  The issue you reported is for Ultra Series 1. Let me check again, then have the update for you.

quinchou77, 2025-03-17:
@VickyFrenzy Could you help me to check the PD version in BIOS setup after you update to 3.03?

VickyFrenzy, 2025-03-17:
> [@VickyFrenzy](https://github.com/VickyFrenzy) Could you help me to check the PD version in BIOS setup after you update to 3.03?

Well I'm already on 3.03 like I said and I can't reinstall 3.03 from Windows as it's saying it's already installed.

The only way I found I could attempt to reinstall 3.03 is by rewriting the `startup.nsh` EFI script to just start the 3.03 update (since by default the script will install 3.04 if the current version is 3.03) which I haven't tried yet except for starting the retimers updates to see if that would make a difference if it's not started from Windows but it had the exact same result.

Since I'm unsure if it's safe to skip the retimers update before updating to 3.04, I preferred to wait for advice on my situation (but most importantly truly fix my issue with the retimers updates that are failing).

Anyway @quinchou77 the BIOS is reporting version **0.0.08** for both PD controllers.

[image]

[image]

quinchou77, 2025-03-20:
I follow the same steps to update BIOS and retimer from BIOS 3.01.

Step1_Framework_Laptop_13_Intel_Core_Ultra_Series1_BIOS_3.03.exe **Success with error**
Step2_Framework_Laptop_13_Intel_Core_Ultra_Series1_Retimer_port01_624.exe **Complete without error**
Step3_Framework_Laptop_13_Intel_Core_Ultra_Series1_Retimer_port23_624.exe **Complete without error**
Step4_Framework_Laptop_13_Intel_Core_Ultra_Series1_BIOS_3.04.exe **Success with error**

Retimer f/w update is optional, so you still can update to 3.04 with PD version 0.0.0A, then update the retimer firmware if you want.

I also do the below steps to make sure the BIOS and retimer update work from BIOS 3.01.
Step1 Framework_Laptop_13_Intel_Core_Ultra_Series1_BIOS_3.03.exe **Success with error**
Step2 Framework_Laptop_13_Intel_Core_Ultra_Series1_BIOS_3.04.exe **Success with error**
Step3 Framework_Laptop_13_Intel_Core_Ultra_Series1_Retimer_port01_624.exe **Complete without error**
Step4 Framework_Laptop_13_Intel_Core_Ultra_Series1_Retimer_port23_624.exe **Complete without error**

VickyFrenzy, 2025-03-20:
Okay, I've updated to 3.04 now.
Retimer updates still won't work (same error) but at least I'm on the latest bios firmware and version 0.0.0A of the PD controllers.

I still have that weird issue with the LED blinking between white and orange but maybe it's a bug with the new battery feature I enabled alongside with the old charge limiter.

quinchou77, 2025-04-02:
Do you still want to update the retimer f/w? Or, update to 3.04 is fine for you?

VickyFrenzy, 2025-07-15:
Hey, I just tried the latest EFI firmware package and everything but retimers were updated successfully.

[image]

```
ESRT Table
  ResourceCount:        4
  ResourceCountMax:     4
  ResourceVersion:      1
ESRT Entry 0
  GUID:                 C57FD615-2AC9-4154-BF34-4DC715344408
  GUID:                 MtlRetimer01
  Type:                 DeviceFirmware
  Version:              0xCF (207)
  Min FW Version:       0x0 (0)
  Capsule Flags:        0x0
  Last Attempt Version: 0x0 (0)
  Last Attempt Status:  Success
ESRT Entry 1
  GUID:                 32D8D677-EEBC-4947-8F8A-0693A45240E5
  GUID:                 MtlCsme
  Type:                 DeviceFirmware
  Version:              0x9D6 (2518)
  Min FW Version:       0x0 (0)
  Capsule Flags:        0x0
  Last Attempt Version: 0x0 (0)
  Last Attempt Status:  Success
ESRT Entry 2
  GUID:                 72CECB9B-2B37-5EC2-A9FF-C739AABAADF3
  GUID:                 MtlBios
  Type:                 SystemFirmware
  Version:              0x305 (773)
  Min FW Version:       0x0 (0)
  Capsule Flags:        0x0
  Last Attempt Version: 0x0 (0)
  Last Attempt Status:  Success
ESRT Entry 3
  GUID:                 BDFFCE36-809C-4FA6-AECC-54536922F0E0
  GUID:                 MtlRetimer23
  Type:                 DeviceFirmware
  Version:              0xCF (207)
  Min FW Version:       0x0 (0)
  Capsule Flags:        0x0
  Last Attempt Version: 0x0 (0)
  Last Attempt Status:  Success
```

benjarobin, 2025-08-20:
I had exactly the same issue, to solve it, I had to reset secure boot with default keys: Restore Secure Boot to factory Settings.

I had enrolled my own keys, but without including OEM firmware's built-in certificates.

VickyFrenzy, 2025-08-20:
> I had exactly the same issue, to solve it, I had to reset secure boot with default keys: Restore Secure Boot to factory Settings.
> 
> I had enrolled my own keys, but without including OEM firmware's built-in certificates.

Sounds weird... I don't believe I have enrolled my own keys since I'm currently using Windows on this laptop but I can certainly try that...

benjarobin, 2025-09-03:
Well, the bug is still there, and still need to be fixed. It just doesn't happen when the BIOS uses the default settings. The problem is that I don't know which BIOS setting causes this bug, and since the update was done on my laptop I cannot reproduce it anymore...

maxtheaxe, 2026-01-29:
Hi @JohnAZoidberg. I also have this issue and am now unable to display on the left side of my computer as well. I used `fwpdmgr` to update from `3.04` to `3.06` and then from `3.06` to `3.19`. I'm using the Intel 12th generation (`i7-1260P`) of the Laptop 13, though. All the other updates worked—I actually didn't realize this one hadn't until I kept trying to do the update and every time I booted the update was still available. Didn't see it attempt to update and fail as the others seem to have seen.