FrameworkComputer/SoftwareFirmwareIssueTracker #49: Source code for the BIOS?

Autofire, 2025-03-08:
This is related to another issue I opened, #48, but I think it'd make sense to keep this as a separate issue.

When I had the idea for the issue above, my first thought was that I wanted to try my hand at modifying the BIOS code myself. It's not a high-priority issue by any stretch, but something I'd find useful.

With all of the hardware being open source, I figured the BIOS would be as well. However, based on this issue tracker's existence, I'm assuming it's probably not open source. Is there a reason for this?

All that being said, I realize it might be a bit stupid for me to even think about modifying my BIOS. I would assume custom BIOS work could very easily lead to a bricked laptop, if I mess anything up.

JohnAZoidberg, 2025-03-08:
Charging is actually controlled from the embedded controller, which we do actually open source the code for.
You can theoretically build and flash that yourself, which some customers have done. But as you mention, it is associated with a lot of risk.
https://github.com/frameworkComputer/embeddedcontroller

> All that being said, I realize it might be a bit stupid for me to even think about modifying my BIOS. I would assume custom BIOS work could very easily lead to a bricked laptop, if I mess anything up.

Yes.

brlin-tw, 2025-03-08:
> Is there a reason for this?

Note that the BIOS is (majority assumed to be) an intellectual property of [Insyde Software](https://www.insyde.com/) and is very likely proprietary and non-redistributable by Framework Computer Inc.

It may be possible to access the source code if there's ever a [coreboot](https://www.coreboot.org/) port of the system firmware, but I won't give high hopes for that.

AkechiShiro, 2025-03-09:
@brlin-tw You'll be surprised to see that an experimental port has already landed in Coreboot 24.05 for the AMD Laptop 7040 series, I don't think this port is intended for end users and I don't think it is possible/easy to test it, I haven't heard of anyone trying it.

I believe this port was done by an engineer who works at AMD on firmware.

[image]

There were also 3 laptops that have been apparently given to some Coreboot devs : 
> We've handed three systems that can boot unsigned bootloaders to folks in the coreboot community. Our plan in the near term is to help them create a shim loader that can be signed to run on any Framework Laptop, which then enables anyone to do further coreboot development.

However, on Intel's side, I haven't seen much news AFAIK.

Sources : 
- https://news.ycombinator.com/item?id=31097434
- https://community.frame.work/t/responded-coreboot-on-the-framework-laptop/791/498
- https://blogs.coreboot.org/blog/2024/05/23/coreboot-24-05-release/