FrameworkComputer/SoftwareFirmwareIssueTracker #50: FW13 Return from s2idle causes reboot (S5_RESET_STATUS = 0x00800800) <- Triple Fault

WillNilges, 2025-03-08:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [X] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
Please provide the bios version. 

03.05

Windows:
Open a command terminal as administrator:
```wmic bios get SMBIOSBIOSVERSION```

Linux: 
Open a terminal and run the following command
```sudo dmidecode --string bios-version```

### DIY Edition information
If you are experiencing an issue on a DIY system, Please also fill out the memory and storage devices you are using. 

Memory: CRUCIAL CT32G56C46S5
Storage: SAMSUNG MZVL21T0HDLU-00BLL

### Port/Peripheral information
If you are experiencing an issue with a peripheral or an expansion card/port please fill out the following information: 
* Peripheral vendor and name.
* Port the Peripheral was connected to. Please see the following for port numbering: [Laptop 16](https://knowledgebase.frame.work/expansion-card-slot-functionality-on-framework-laptop-16-rkUjGm7cn) or [Laptop 13](https://knowledgebase.frame.work/expansion-card-functionality-on-framework-laptop-13-amd-ryzen-7040-series-SkrVx7gAh)
* Device or expansion card attached to the **Adjacent port** to the port that is having the issue.
1. USB-C
2. USB-A
3. USB-C 
4. HDMI

The following are for Laptop 16 only. 
5. [ Fill In device / expansion card ]
6. [ Fill In device / expansion card ]

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug
A clear and concise description of what the bug is.

Often (at least 1/3 times) when I come out of s2idle, my laptop screen will suddenly go black, then the machine will start up as if coming from a cold boot. Sometimes it happens slowly enough that I can unlock and get a browser open, sometimes it happens before I'm even done opening the lid.

I applied a kernel patch (see below) and the reset code is

```
Mar 08 10:40:44.529328 willardpad kernel: S5_RESET_STATUS = 0x00800800
```

## Steps To Reproduce
Steps to reproduce the behavior:

1. Boot up
2. Log in
3. Go to s2idle
4. Wake up from s2idle
5. Unlock
6. Open Firefox

You should see the crash by Step 4

## Expected behavior
A clear and concise description of what you expected to happen.

I can use the laptop and sleep and wake as many times as I want without it crashing.

## Screenshots
If applicable, add screenshots to help explain your problem.

## Operating System (please complete the following information):
 - OS/Distribution: Arch Linux
 - Version: March 2025
- Linux Kernel Version: Linux willardpad 6.13.5-arch1-1-custom #1 SMP PREEMPT_DYNAMIC Wed, 05 Mar 2025 23:40:06 +0000 x86_64 GNU/Linux

I have applied [this patch](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/41#issuecomment-2701195650) to check my reset codes.

## Additional context
Add any other context about the problem here.

This issue started around Feb 14, 2025. At the time I was running Linux LTS and it had upgraded to 6.12 when the issues started. I was on 6.6 since around December and it was fine.

I was directed to open this issue here: https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/41

This is my thread: https://community.frame.work/t/spontaneous-and-ungraceful-restarts-upon-closing-the-lid/64806

I saw this thread about what seems to be the exact same issue: https://community.frame.work/t/framework-13-amd-return-from-sleep-causes-reboot/64743/16

Possibly these folks as well: https://community.frame.work/t/responded-inconsistent-resume-from-suspend-linux-amd/46587

It would be nice to reach out to everybody in these threads and see if we could get some more information. I am also going to open a support ticket with Framework directly because they keep ignoring me on the forums.

jcdutton, 2025-03-08:
Hi.
Thank you for raising an issue ticket for this.
Some things worth trying:
1) 
`sudo echo N > /sys/module/printk/parameters/console_suspend`
This makes the console stay awake even during suspend.
The advantage of this is that you get to see some debug messages that you might not have otherwise seen.
They might disappear very quickly though, so you probably need to video it with your phone, and then slow-mo through the video until you find what you need.
2)
Narrowing down what is causing the fault:
Unplug all your expansion slot cards and try a suspend / resume cycle and see if the problem remains.
3)
You don't say in the description whether you have a dGPU or not, please add that to the description at the top.
If you have one, try removing the dGPU and see if the problem remains.
4)
See if the SSD is the cause, try removing the SSD and boot with a USB stick with Linux on it and try to reproduce the problem.
5)
Try blacklisting the wifi card driver, so that it does not need to go through the suspend / resume cycle. This should discount it without needing to remove it.

Context:
Triple faults are notoriously difficult to track down, but at least they are software faults, so once the bug is found, it can be cured.

WillNilges, 2025-03-08:
Thanks for the ideas!
For 2) I can verify that even with all my cards yanked, the issue persists.

I'll try yanking the SSD and wifi card when I have the time. It could take days of sleep/wake to find the issue.

One pattern I notice is that it happens when I resume after sleeping for a long time (like 8 hours overnight) and then I can usually get it to happen very shortly after the first time, too.

I don't have a dGPU, this is a FW13 and I mostly use it on the go. I'll update the issue.

It would be nice if we could bisect the kernel to figure out if this was happening. From what I have seen online, it doesn't seem to be happening on 6.11, and was introduced sometime in 6.12. I guess I could try compiling the latest 6.11 kernel, and maybe the earliest 6.12 LTS kernel (because I know for a fact it happened there)

helpimnotdrowning, 2025-03-12:
Dropping by to add that I've been experiencing a similar issue (described here: https://community.frame.work/t/sudden-reboot-on-wake-with-no-logs/65434 )

I've experienced this on several kernels, being from Debian stable 6.1, backports 6.9 and 6.12, Liquorix 6.12, and mainline [6.14 rc4](https://github.com/torvalds/linux/commit/1e15510b71c99c6e49134d756df91069f7d18141) (though I haven't checked what patches the Debian team is applying on their kernels, maybe they are backporting some bad patches?)

I've heard that others with somewhat-similar issues on Arch could fix by downgrading Mesa to 24.x, but Debian stable ships 22, and updating to 24 did not help (crashes still occur)

I *have* noticed that I almost always prevent reboots by switching to a tty right as I open my laptop. I usually keep it there for a few seconds before switching back to tty1 with Plasma/Wayland.

WillNilges, 2025-03-18:
> I have noticed that I can almost always prevent reboots by switching to a tty right as I open my laptop. I usually keep it there for a few seconds before switching back to tty1 with Plasma/Wayland.

Sometimes I get reboots the instant I open my lid :(

WillNilges, 2025-03-18:
> sudo echo N > /sys/module/printk/parameters/console_suspend
> This makes the console stay awake even during suspend.
> The advantage of this is that you get to see some debug messages that you might not [...]

@jcdutton Can you say more about this? How would I see these messages if I'm not getting past my lockscreen before a resume? Would they appear in a TTY? Would they appear in my dmesg?

jcdutton, 2025-03-18:
@WillNilges 
Search the web if you want more info about the "sudo echo N > /sys/module/printk/parameters/console_suspend"
Or just try it and see what extra information it gives during a successful suspend/resume cycle, and then compare it to a failed suspend / resume cycle.

CDRXavier, 2025-03-27:
This is related?
https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/35

PureKrome, 2025-03-27:
@CDRXavier sounds very related, putting aside the OS's of choice/used.

Nice seeing more and more people mention this and in various hardware setup/flavors, too!

jcdutton, 2025-03-28:
Now, I don't think the Triple fault is coming from the Linux kernel.
I am now wondering if there some other BIOS or system management firmware that is running behind the scenes that could cause this S5_RESET_STATUS triple fault ?
If so, what is that piece of code called, and is that code x86 or ARM or something else?

Alternatively, could this be caused by bit-flips, maybe going undetected due to lack of ECC in some data paths?

I am just trying to think of some other things to investigate to track this down.

sydneymeyer, 2025-03-28:
> I am now wondering if there some other BIOS or system management firmware that is running behind the scenes that could cause this S5_RESET_STATUS triple fault ?

> Alternatively, could this be caused by bit-flips, maybe going undetected due to lack of ECC in some data paths?

Funny enough, i'm thinking the same. This is either something SMM, something the on-die ecc doesn't/can't catch, or, i can't even tell why, the mediatek wireless card freaking out with power management. The completely random pattern suggests the memory theory, i suppose. i.e. an electrical issue.

helpimnotdrowning, 2025-03-29:
I can safely (hopefully) say that, since the 14th, I am no longer experiencing this issue. I have no idea why. I am still using my self-built 6.14 rc4 kernel that I mentioned in the forum post.

This is the contents of my /var/log/dpkg.log of all my package changes from that last triple-fault to today: 

I had no changes on the 14th itself, so this starts on the 15th.
I imagine that the changes closest to the start of this log would be the most important?

<details>
2025-03-15 13:43:46 startup packages remove
2025-03-15 13:43:46 status installed libdrm-nouveau2:i386 2.4.123-1~bpo12+1
2025-03-15 13:43:47 remove libdrm-nouveau2:i386 2.4.123-1~bpo12+1 <none>
2025-03-15 13:43:47 status config-files libdrm-nouveau2:i386 2.4.123-1~bpo12+1
2025-03-15 13:43:47 status not-installed libdrm-nouveau2:i386 <none>
2025-03-15 13:43:47 startup packages configure
2025-03-15 13:43:47 status installed libc-bin:amd64 2.36-9+deb12u9
2025-03-15 16:50:11 startup archives unpack
2025-03-15 16:50:12 upgrade rustdesk:amd64 1.3.2 1.3.8
2025-03-15 16:50:14 startup packages configure
2025-03-15 16:50:14 configure rustdesk:amd64 1.3.8 <none>
2025-03-15 16:50:15 status installed rustdesk:amd64 1.3.8
2025-03-15 16:50:15 status installed hicolor-icon-theme:all 0.17-2
2025-03-15 16:50:15 status installed mailcap:all 3.70+nmu1
2025-03-15 16:50:15 status installed desktop-file-utils:amd64 0.26-1
2025-03-15 19:03:49 startup packages remove
2025-03-15 19:03:49 status installed vlc:amd64 3.0.21-0+deb12u1
2025-03-15 19:03:50 remove vlc:amd64 3.0.21-0+deb12u1 <none>
2025-03-15 19:03:50 status config-files vlc:amd64 3.0.21-0+deb12u1
2025-03-15 19:03:50 status not-installed vlc:amd64 <none>
2025-03-15 19:03:50 startup packages configure
2025-03-15 19:03:50 status installed mailcap:all 3.70+nmu1
2025-03-15 19:03:50 status installed desktop-file-utils:amd64 0.26-1
2025-03-15 19:04:19 startup packages remove
2025-03-15 19:04:19 status installed mpv:amd64 0.35.1-4
2025-03-15 19:04:19 remove mpv:amd64 0.35.1-4 <none>
2025-03-15 19:04:19 status config-files mpv:amd64 0.35.1-4
2025-03-15 19:04:19 startup packages configure
2025-03-15 19:04:19 status installed hicolor-icon-theme:all 0.17-2
2025-03-15 19:04:20 status installed man-db:amd64 2.11.2-2
2025-03-15 19:04:20 status installed mailcap:all 3.70+nmu1
2025-03-15 19:04:20 status installed desktop-file-utils:amd64 0.26-1
2025-03-15 19:04:25 startup packages remove
2025-03-15 19:04:25 status installed libsixel1:amd64 1.10.3-3
2025-03-15 19:04:26 remove libsixel1:amd64 1.10.3-3 <none>
2025-03-15 19:04:26 status config-files libsixel1:amd64 1.10.3-3
2025-03-15 19:04:26 status not-installed libsixel1:amd64 <none>
2025-03-15 19:04:26 status installed vlc-plugin-access-extra:amd64 3.0.21-0+deb12u1
2025-03-15 19:04:26 remove vlc-plugin-access-extra:amd64 3.0.21-0+deb12u1 <none>
2025-03-15 19:04:26 status config-files vlc-plugin-access-extra:amd64 3.0.21-0+deb12u1
2025-03-15 19:04:26 status not-installed vlc-plugin-access-extra:amd64 <none>
2025-03-15 19:04:26 status installed libvncclient1:amd64 0.9.14+dfsg-1
2025-03-15 19:04:26 remove libvncclient1:amd64 0.9.14+dfsg-1 <none>
2025-03-15 19:04:26 status config-files libvncclient1:amd64 0.9.14+dfsg-1
2025-03-15 19:04:26 status not-installed libvncclient1:amd64 <none>
2025-03-15 19:04:26 status installed libxpresent1:amd64 1.0.0-2+b10
2025-03-15 19:04:26 remove libxpresent1:amd64 1.0.0-2+b10 <none>
2025-03-15 19:04:26 status config-files libxpresent1:amd64 1.0.0-2+b10
2025-03-15 19:04:26 status not-installed libxpresent1:amd64 <none>
2025-03-15 19:04:26 status installed vlc-bin:amd64 3.0.21-0+deb12u1
2025-03-15 19:04:26 remove vlc-bin:amd64 3.0.21-0+deb12u1 <none>
2025-03-15 19:04:26 status config-files vlc-bin:amd64 3.0.21-0+deb12u1
2025-03-15 19:04:26 status not-installed vlc-bin:amd64 <none>
2025-03-15 19:04:26 status installed vlc-l10n:all 3.0.21-0+deb12u1
2025-03-15 19:04:26 remove vlc-l10n:all 3.0.21-0+deb12u1 <none>
2025-03-15 19:04:26 status config-files vlc-l10n:all 3.0.21-0+deb12u1
2025-03-15 19:04:26 status not-installed vlc-l10n:all <none>
2025-03-15 19:04:26 status installed vlc-plugin-notify:amd64 3.0.21-0+deb12u1
2025-03-15 19:04:26 remove vlc-plugin-notify:amd64 3.0.21-0+deb12u1 <none>
2025-03-15 19:04:26 status config-files vlc-plugin-notify:amd64 3.0.21-0+deb12u1
2025-03-15 19:04:26 status not-installed vlc-plugin-notify:amd64 <none>
2025-03-15 19:04:26 status installed vlc-plugin-skins2:amd64 3.0.21-0+deb12u1
[... 1069 lines cut ...]
2025-03-23 21:32:51 configure libfftw3-dev:amd64 3.3.10-1 <none>
2025-03-23 21:32:51 status installed libfftw3-dev:amd64 3.3.10-1
2025-03-23 21:32:51 configure libboost-date-time1.74-dev:amd64 1.74.0+ds1-21 <none>
2025-03-23 21:32:51 status installed libboost-date-time1.74-dev:amd64 1.74.0+ds1-21
2025-03-23 21:32:51 configure libboost-thread1.74-dev:amd64 1.74.0+ds1-21 <none>
2025-03-23 21:32:51 status installed libboost-thread1.74-dev:amd64 1.74.0+ds1-21
2025-03-23 21:32:51 configure libboost-thread-dev:amd64 1.74.0.3 <none>
2025-03-23 21:32:51 status installed libboost-thread-dev:amd64 1.74.0.3
2025-03-23 21:32:51 configure libboost-locale1.74-dev:amd64 1.74.0+ds1-21 <none>
2025-03-23 21:32:51 status installed libboost-locale1.74-dev:amd64 1.74.0+ds1-21
2025-03-23 21:32:51 configure libboost-locale-dev:amd64 1.74.0.3 <none>
2025-03-23 21:32:51 status installed libboost-locale-dev:amd64 1.74.0.3
2025-03-23 21:32:51 status installed libc-bin:amd64 2.36-9+deb12u9
[... cut ...]
2025-03-28 09:31:08 startup archives unpack
2025-03-28 09:31:09 install qdirstat:amd64 <none> 1.8.1-3
2025-03-28 09:31:09 startup packages configure
2025-03-28 09:31:09 configure qdirstat:amd64 1.8.1-3 <none>
2025-03-28 09:31:09 status installed qdirstat:amd64 1.8.1-3
2025-03-28 09:31:09 status installed hicolor-icon-theme:all 0.17-2
2025-03-28 09:31:10 status installed man-db:amd64 2.11.2-2
2025-03-28 09:31:10 status installed mailcap:all 3.70+nmu1
2025-03-28 09:31:10 status installed desktop-file-utils:amd64 0.26-1

</details>

CDRXavier, 2025-04-01:
So. A few things.
After reading this (funny thing with self-encrypting drives and firmware), I have decided to turn off the hard drive passwords.
https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/42

I have password on all my drives -- Windows drive, Linux drive, my data drive. It's simpler than Bitlocker, a lot more reliable.
This way I can decide which drive to "connect" to the selected OS to boot up. This is after the Linux-Windows incident. (where the Ubuntu 2404 installer wiped the Windows install disk).
And then my Linux install worked immediately. I have since moved to Windows, and for a very long time I straight up don't get any problem waking up.

WillNilges, 2025-04-16:
I have also not experienced this issue since at least 2025-03-19.

https://community.frame.work/t/spontaneous-and-ungraceful-restarts-upon-closing-the-lid/64806/21?u=will_nilges

I think this is what did it
```
[2025-03-19T18:45:23-0400] [ALPM] upgraded amd-ucode (20250311.b69d4b74-2 → 20250311.b69d4b74-3)
[2025-03-19T18:45:23-0400] [ALPM] upgraded linux-firmware-whence (20250311.b69d4b74-2 → 20250311.b69d4b74-3)
[2025-03-19T18:45:24-0400] [ALPM] upgraded linux-firmware (20250311.b69d4b74-2 → 20250311.b69d4b74-3)
```

michaelmehl, 2025-10-26:
I see this issue on my FW13:

6.16.12-200.fc42.x86_64
AMD Ryzen 5 7640U w/ Radeon 760M Graphics
bios-version: 03.16

laghoule, 2026-03-14:
USB C on port 1, 3
USB A on port 2, 4

Wake on sleep work ~95% of the time.
~5% result in a reboot instead of wake

```
journalctl -b 0 | grep "reset reason"
Mar 14 09:38:07 xxx kernel: x86/amd: Previous system reset reason [0x08000800]: an uncorrected error caused a data fabric sync flood event
```

Kernel: 6.19.6+deb14-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.19.6-1 (2026-03-06) x86_64 GNU/Linux

jcdutton, 2026-03-22:
> [edit] Maybe this is different - I get 0x08000800 not 0x00800800 [/edit]

Hi.

For people who see "0x08000800 not 0x00800800", please look at this other issue:
https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/41

leo60228, 2026-09-08:
I'm also having this issue. I've noticed it occurs while playing games, even lightweight ones, but I have not seen it occur during productivity tasks. It also does not occur in power saving mode.

I'm not sure if the conclusion that this is a triple fault is accurate. Bit 11 in S5_RESET_STATUS is listed as reserved in AMD documentation.

Tanguille, 2026-09-15:
I have the same issue on a Framework Laptop 16 Ryzen AI 300.

Setup: 
- Ryzen AI 9 HX 370, no dGPU 
- BIOS 04.01 / EC 4.0.1
- 64 GB (2×32)
- WD PC SN8000S 2 TB (fw 63100000), HDMI + DP expansion cards, docked on a 100 W PD source. 
- Fedora 44: kernels 7.2.4-200 (stock) and 7.2.5-cachyos both are affected
- linux-firmware 20260910,
- amdgpu.dcdebugmask=0x40000

Tanguille, 2026-09-15:
Framework Laptop 16 Ryzen AI 300 (HX 370), mainboard replaced 5 Sep 2026: two `0x00800800` events on 15 Sep, awake and in use, no kernel distress, pstore empty. The original board produced `0x08000800` for the same symptom, so like @reedlaw I'm treating the two codes as one fault. Full details, timeline since first occurrence (6 May 2026) and version history posted in #41: https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/41#issuecomment-5683812987

jcdutton, 2026-10-02:
Note, it used to be called "triple fault"
Now it is called: "internal CPU shutdown event occurred"

duude21, 2026-10-23:
I have got the exact same problem, but it only occured randomly. (reset code 0x08000800 sync flood) Now I changed my HDMI Expansion Card to Port 4 and hope it does not happen. 
Wiredly enough I got (almost?) the same SSD as OP. Hope it is not connected to that.

Laptop 13 (AMD Ryzen 7040Series)
Bios: 03.20
Kernel: 7.2.11-200.fc44.x86_64
SSD: MZVLC1T0HFLU-00BLL

daniel-metalntwx, 2026-11-07:
Same reset code on a Framework Laptop 13 Pro (AMD Ryzen AI 300). I haven't seen this model reported here yet.

- CPU: Ryzen AI 9 HX 370
- BIOS: 04.02
- RAM: 2 x 32 GB Crucial CT32G56C46S5.M16D1, supplied by Framework
- SSD: WD_BLACK SN850X 1 TB
- Expansion cards: USB-A, Ethernet, 2 x USB-C. No HDMI or DP cards.
- OS: Ubuntu 26.04.1, kernels 7.0.0-38 and 7.0.0-42

It has reset four times since Oct 21, always while awake and in use, never during suspend or resume:

| Date (AEDT) | Reset reason | Kernel | Power |
|---|---|---|---|
| Oct 21 | `0x00800800` internal CPU shutdown | 7.0.0-38 | likely charger |
| Oct 30 | `0x00080800` CF9 reset, with no shutdown logged | 7.0.0-38 | charger, 100% |
| Nov 2 | `0x00800800` internal CPU shutdown | 7.0.0-42 | battery, 88% |
| Nov 6 | `0x00800800` internal CPU shutdown | 7.0.0-42 | charger, 100% |

The screen goes black and the laptop reboots, with no freeze first. The journal just stops. kdump was loaded and saved nothing, pstore is empty, and rasdaemon has logged no MCE, memory or PCIe errors. Load was light each time.

So far I've loaded BIOS defaults and run memtest86+ for 10 passes with no errors. I have a support ticket open with Framework.

Happy to collect anything that would help.