FrameworkComputer/SoftwareFirmwareIssueTracker #51: Chassis getting uncomfortably warm due to changed fan curve after update to BIOS 3.09 beta (FW13 12th Gen)

taosxx, 2025-03-10:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [x] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

Processor: 1240P (FRANDA0074)

### BIOS VERSION
03.09

### DIY Edition information

Memory: 32GB Corsair Vengeance DDR4-3200 SO-DIMM CL22 Dual Kit (CMSX32GX4M2A3200C22)
Storage: 1TB Samsung SSD 980 M.2 PCIe 3.0 x4 3D-NAND TLC (MZ-V8V1T0BW)

### Port/Peripheral information
1. USB-C expansion card
2. MicroSD expansion card
3. USB-C expansion card connected to Framework 60W power adapter
4. USB-A expansion card

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug
After updating the BIOS from 3.08 to 3.09 (via EFI method) the fan curve changed significantly. The fan now takes much longer (in my opinion too long) to accelerate if the temperature rises. While I appreciate the very low noise level (who doesn't?), I prefer my fingers not getting toasted while typing a lot more.
Even with the PPD power profile set to "Power save" under Linux, the laptop chassis can get quite warm when "stressing" the CPU (or iGPU) a little because the fan doesn't seem to accelerate at all (at first, I mistakenly thought it wouldn't spin at all but you can hear a very faint noise with the ear next to the air intake). Since the heat from the CPU can no longer leave "by air" through the back exhaust near the bottom of the display, the heat has to dissipate slowly through the chassis. This keeps the input cover uncomfortably warm for a prolonged time and the bottom even hotter.

## Steps To Reproduce
Steps to reproduce the behavior:
1. Stress the CPU a little by, e.g., watching a video.
2. Wait a couple of minutes and notice the input cover getting warmer and warmer while the fan seems idle. The bottom might even get hot.
3. Stop watching the video. The chassis still gets warmer for a few more minutes until it starts to cool down again. The bottom will stay warm even longer.

## Expected behavior
I expect the fan to start to accelerate in order to cool down the laptop so it doesn't heat up the chassis too much. With BIOS 3.08, I wouldn't notice the noise of the fan spinning up in Linux (with power-save mode) but the fan would still spin faster to get the heat out before the chassis could get noticeably warmer.

## Screenshots
--

## Operating Systems:
 - OS/Distribution: Archlinux
   - Version: Rolling release (updated today)
   - Linux Kernel Version: `Linux FrameWorkArch 6.13.5-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 27 Feb 2025 18:09:44 +0000 x86_64 GNU/Linux`

 - OS/Distribution: Windows 11
   - Version: 24H2

 - OS/Distribution: Haiku 64bit Nightly
   - Version: hrev58715
 
 - OS/Distribution: OpenBSD 7.7
   - Version: yesterday's snapshot

## Additional context
I see this problem in all aforementioned operating systems. In Windows, however, the fan at least spins up audibly (where it would hit jet-engine levels of noise with the previous BIOS 3.08).

kiram9, 2025-03-15:
We have been able to reproduce this issue, and are working on an urgent fix.

taosxx, 2025-05-17:
After one week with BIOS 3.17 Beta, I'd say input cover temperatures are again back to BIOS 3.08 levels.

kiram9, 2025-05-18:
Fixed in 3.17. Currently in beta here https://community.frame.work/t/framework-laptop-13-12th-gen-intel-core-bios-3-17-release-beta/68919

closed 2025-05-18 by kiram9