FrameworkComputer/SoftwareFirmwareIssueTracker #52: Battery Charge Limit unreliable after installing Laptop 13 Ryzen 7040 BIOS 3.07 Release

timwenger, 2025-03-10:
## Device Information

### System Model or SKU

- [X] Framework Laptop 13 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
Upgraded from 3.06 to 3.07 on Windows 11 Pro

### DIY Edition information
Memory: 2x Kingston 32GB 5600 MT
Storage: WD Black SN850X, 1TB

### Port/Peripheral information
Connected to a USB C port on [port 1](https://knowledgebase.frame.work/expansion-card-functionality-on-framework-laptop-13-amd-ryzen-7040-series-SkrVx7gAh) which is connected to a Dell U3219Q monitor with Power Delivery (this is how the laptop is charged).
All other ports are disconnected.

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [X] No

## Describe the bug
BIOS Battery Charge limit settings below 100% are not adhered to. Instead it charges to 100%.

## Steps To Reproduce
Steps to reproduce the behavior:
1. Installed bios 3.07 from 3.06, with charge limit set at 65%
2. After restarting, the charge started rising above. I noticed it at 80%
3. I found on the [3.07 release page](https://community.frame.work/t/framework-laptop-13-ryzen-7040-bios-3-07-release-beta/65582?utm_campaign=2025-03-05-fl13-amd-beta-software-update) that others had success to get the charge limit back by setting the limit to 100%, rebooting, then setting back down to the desired %. I did this, and afterwards I found it stopped charging and dropped back to my limit.
4. The next day I found that the battery was charged to 100% again (perhaps I restarted it, or it hibernated). So I repeated the above step (reboot twice to set the limit to 100% then back down to my desired %  - this time I chose 80% arbitrarily, and it's now staying limited to 80%, for the time being.
5. At first I had the battery extender enabled, but I believe I have experienced the issue regardless of whether battery extender is enabled or not.

## Operating System (please complete the following information):
 - OS/Distribution: Windows 11 Pro
 - Version: 24H2, 26100.3194

## Additional context
Many others are having this issue, and have expressed their situation on the  [3.07 release page](https://community.frame.work/t/framework-laptop-13-ryzen-7040-bios-3-07-release-beta/65582?utm_campaign=2025-03-05-fl13-amd-beta-software-update)

quinchou77, 2025-03-12:
When you set the charge limit, do you disable the battery extender at the same time?

timwenger, 2025-03-12:
> When you set the charge limit, do you disable the battery extender at the same time?

Here's what I did this morning - hopefully this answers your question:
1. My charge limit had been correctly adhered to yesterday, plugged in with charge limit enabled to 80% and the battery extender disabled. I then closed the lid to sleep it and then unplugged it.
2. My laptop had been sleeping, unplugged for 18h+
3. I first plugged it in, then pressed the power button. Upon waking, the framework logo showed. So it was in a deep sleep / hibernation.
4. after 10 min or so plugged in, it passed my charge limit. This was now the 2nd time since upgrading to 3.07 where I had had the charge limit working, but then it stopped working. 
5. I rebooted and swapped both charge settings: I limited the charge to 100%, and enabled battery extender.
6. I rebooted again, and only changed the charge limit to 75%. After booting though, it was still charging past 75%.
7. I rebooted again, and limited the charge to 100% and disabled the battery extender.
8. I rebooted again, and only changed the charge limit to 75%. After booting, it stopped charging and began to discharge to 75%.

quinchou77, 2025-03-13:
We found a known issue that "If the battery extender mode is disabled, and charge limit is set below 100%, the charge limit will not be enforced causing the battery to charge to 100%."

You could try to enable battery extender first, then see whether the charge limit works normally or not.

Stebalien, 2025-03-13:
> We found a known issue that "If the battery extender mode is disabled, and charge limit is set below 100%, the charge limit will not be enforced causing the battery to charge to 100%."

The issue is reproducible both with and without the battery extender mode enabled.

conradsnicta, 2025-03-14:
@quinchou77 I can confirm the same problem when using Linux (Fedora 40).  Reproducible as per the comment from @Stebalien 

> We found a known issue that "If the battery extender mode is disabled, and charge limit is set below 100%, the charge limit will not be enforced causing the battery to charge to 100%."

sgilder1, 2025-03-14:
And Framework have shipped 3.07 as stable, even with this known regression....

jcdutton, 2025-03-14:
Hi,
I don’t have a FW13 AMD, but I do have a FW16.
I have not observed the battery problem. I limit mine to 70% and it stays at 70%. I disable the battery safe feature because I don’t need it. If you power off and unplug the laptop every night, it never will trigger the battery save feature.
But, my understanding is that battery charge/control is done by the EC (embedded controller) which is 100% open source. So, if people are seeing battery charge problems, they could look at the source code and suggest fixes. I have made changes to the EC code that fix some bugs in the EC, but none of those are charge related.

Algorithmically, the battery needs some input parameters:

Fixed set by user:

    global charge limit
    battery save enable/disabled
    battery save trigger setting.

Measured values:
4) battery current charge level
5) EC uptime.

State that must survive a power off of the EC / laptop
6) state of hyterisis cycle.
1-charging to the limit
2-discharging to limit -5.

The rules:
Say you have set the global limit to 80%.
That is fine until the uptime triggers the battery save. It then is looking to use the 98%, then less over time.
That set of rules is odd and suspect in my opinion, but is consistent with what problems people are reporting. I would have chosen a different set of rules.

My suggested work around is therefore to never trigger the battery save function or set the global limit to 100%. Then reset the EC so the uptime is reset to zero and not triggering the battery save function.

To reset the EC:

    power off the laptop, not suspend.
    unplug the PSU for 2 minutes.
    plug power back in.
    power on laptop.
    use ectool console to check the uptime has been reset.

Summary: unplugging the power for a few seconds does NOT reset the EC.

kiram9, 2025-03-15:
Hi Everyone, 
We have reproduced this issue and have a fix that we will be releasing to stable before the end of march.

jcdutton, 2025-03-15:
@kiram9 
If the fix is a change to the EC. Please point me to the git commit so I can have a look.

mercertom, 2025-03-27:
> by the end of march

Is this still happening? March is almost up.

quinchou77, 2025-03-31:
@mercertom 3.08 beta has been posted to forum https://community.frame.work/t/framework-laptop-13-ryzen-7040-bios-3-08-release-beta/66858

mercertom, 2025-03-31:
@quinchou77 doesn't exist on lvfs-testing, as the notes claim.

I just get:

Devices with no available firmware updates:

HastD, 2025-03-31:
@mercertom As noted on the forum post:

> The LVFS was down today. Consequently, this version cannot be upgraded via fwupd through the LVFS. Once the LVFS is restored, I will move this version to testing for BIOS updates.

quinchou77, 2025-03-31:
@mercertom LVFS is restored, you can try it now.

mercertom, 2025-04-01:
@quinchou77 

I still get

Devices with no available firmware updates:

quinchou77, 2025-04-01:
The LVFS seems remain unstable. 
'fwupdmgr refresh --force' is intermittently failing.

amstan, 2025-04-01:
> Devices with no available firmware updates:

There's also a fwupd bug that we've been hitting recently, that I think causes that error message.

fwupd 2.0.6 is affected, 2.0.4 and 2.0.7 is not.

JohnAZoidberg, 2025-04-01:
See https://github.com/fwupd/fwupd/issues/8510

timwenger, 2025-04-01:
Thank you, Framework, for acknowledging the situation and following through with firmware v3.08 to fix this; I Appreciate it.

quinchou77, 2025-04-09:
@timwenger Thanks for the feedback. Assuming you've disabled the battery extender in the BIOS, the battery limit should work correctly on your system with version 3.08. Just sharing below information. I'll close this issue tracker and follow up on the other open one regarding the related issue.

**We have identified an issue in version 3.08: when the battery charge limit is set below 100% and the battery extender is enabled, the device incorrectly charges to 100% after AC power is disconnected for more than 30 minutes. We are prioritizing a resolution for this issue.
To use the battery limit function, disable the battery extender and then set your desired charge limit.**

kenneth-ge, 2025-04-09:
Just wanted to say that I'm having the same bug on BIOS 3.07! Frustrating to say the least, especially since I was instructed by FW support to upgrade BIOS version after receiving a faulty replacement part. When will 3.08 reach stable?

balpha, 2025-04-12:
Since the posts here are all about Windows, I want to note that I'm seeing this with Ubuntu as well. Since the 3.07 upgrade, the laptop charges to 100% despite the limit set to 80%.

Will try if 3.08 fixes that once it's available.

Thanks for being transparent!

Olybo, 2025-04-23:
Hello,

I have updated from 3.07 to 3.08.
With battery extender disabled, the battery limit is working correctly.

However, [here](https://community.frame.work/t/framework-laptop-13-ryzen-7040-bios-3-08-release-beta-held/66858) it is indicated :

> This release modifies the battery charge limit functionality to add a 5% float range. [..]  the battery will not start to charge until the battery has dropped 5% below the charge limit.

Yet, once the battery charge has reached the limit I have set in the bios, the charge alternates between charge and not charge every 30s - 1min.
I was able to notice this since my Fedora system plays a small sound notification each time the system charge or stop to charge. For now, I have disabled system sound as it is very annoying.

I post my issue here as the problem is directly related to this first issue, but if you prefer, I can open a new issue.

Don't hesitate to ping me if you need more information.

Thanks in advance,

jcdutton, 2025-04-23:
I have seen the mini-cycle behavior on my FW16.
There are a number of edge cases, here are a few:
1) It can do the cycling at almost any point. At the charge/discharge point and at the Idle/charge and idle/discharge points. 
2) If the battery re-calibrates itself at a particular moment, it can go on to charge to 100%, ignoring limits.
These are all fixable by fixing bugs in the EC source code.

quinchou77, 2025-05-07:
BIOS 3.09 beta for the Laptop13 with Ryzen 7040 processors is now available. This update resolves a bug affecting the battery charge limit, ensuring it works correctly whether a battery extender is enabled or disabled. #

sdlarsen, 2025-05-15:
With BIOS upgraded to 3.09 I don't have charge_control_{start,end}_threshold for /sys/class/power_supply/BAT1 anymore. And hence tlp is not able to function. This worked just fine before upgrading.

```
sudo tlp-stat -s -b
Password:
--- TLP 1.8.0 --------------------------------------------

+++ System Info
System         = Framework A7 Laptop 13 (AMD Ryzen 7040Series)
BIOS           = 03.09
OS Release     = Gentoo Linux
Kernel         = 6.14.6-gentoo #1 SMP PREEMPT_DYNAMIC Mon May 12 08:18:15 CEST 2025 x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-6.14.6-gentoo root=/dev/mapper/vg0-root ro rd.luks.allow-discards quiet splash
Init system    = openrc
Boot mode      = UEFI
Suspend mode   = [s2idle]

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 11:16:21, 21 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: generic
Supported features: none available

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = NVT
/sys/class/power_supply/BAT1/model_name                     = FRANGWA
/sys/class/power_supply/BAT1/cycle_count                    =     18
/sys/class/power_supply/BAT1/charge_full_design             =   3915 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   4121 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   4119 [mAh]
/sys/class/power_supply/BAT1/current_now                    =      0 [mA]
/sys/class/power_supply/BAT1/status                         = Not charging

/sys/class/power_supply/BAT1/charge_control_start_threshold = (not available)
/sys/class/power_supply/BAT1/charge_control_end_threshold   = (not available)
/sys/class/power_supply/BAT1/charge_behaviour               = (not available)

Charge                                                      =  100.0 [%]
Capacity                                                    =  105.3 [%]
```

t-8ch, 2025-05-15:
> With BIOS upgraded to 3.09 I don't have charge_control_{start,end}_threshold for /sys/class/power_supply/BAT1 anymore.

I guess it is https://github.com/FrameworkComputer/EmbeddedController/commit/1d7bcf1d50137c8c01969eb65880bc83e424597e interacting badly with https://github.com/torvalds/linux/blob/v6.14/drivers/mfd/cros_ec_dev.c#L263

balpha, 2025-05-17:
> Since the posts here are all about Windows, I want to note that I'm seeing this with Ubuntu as well. Since the 3.07 upgrade, the laptop charges to 100% despite the limit set to 80%. [[me on April 12](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/52#issuecomment-2799013818)]

I never got 3.08, but I upgraded to 3.09 yesterday, and now it's respecting the charge limit set in the bios again.

Thanks for the fix!

johansmitsnl, 2025-05-17:
How do you get the 3.09 because fwupdate gives me still 3.08 when i updated to the beta.

balpha, 2025-05-17:
I got the notification that it's ready to install a day or two ago. Maybe it's a staggered rollout?

sdlarsen, 2025-05-17:
> How do you get the 3.09 because fwupdate gives me still 3.08 when i updated to the beta.

@johansmitsnl You may need to do a ```fwupdmgr refresh``` to get the latest version

closed 2025-07-11 by quinchou77