FrameworkComputer/SoftwareFirmwareIssueTracker #53: UEFI Boot Entries Deleted

deepcube, 2025-03-17:
## Device Information

### System Model or SKU

Framework Laptop 13 (AMD Ryzen™ 7040 Series)

### BIOS VERSION

3.05

### DIY Edition information

Memory: Crucial 2x16GB DDR5 5600MHz CT2K16G56C46S5
Storage: Samsung 990 PRO 2TB MZ-V9P2T0B/AM

## Describe the bug

When a given UEFI boot entry does not exist on boot, the entry is deleted. In my case this happens because I use a storage password, and if I hit escape instead of entering the password then the disk is not decrypted and the custome UEFI entries added using `efibootmgr` are deleted and are no longer available on next boot when I do decrypt the drive. I have also shown this behavior with a USB drive, registering a boot entry with `efibootmgr` which is then deleted when I boot without the drive plugged in.

Note that these are entries for kernels with EFI stub support, not for a bootloader. I have confirmed that an EFI executable in a default path (e.g. `\EFI\BOOT\BOOTX64.EFI`) will be found by the BIOS even if other boot entries for the drive have been deleted.

## Steps To Reproduce

Steps to reproduce the behavior:
1. Register a boot entry with `efibootmgr` for a kernel (or other EFI executable) not in the default EFI path.
2. Remove the drive with the given boot entry, or in the case of a SED press escape instead of entering the storage password.
3. Boot and receive boot device not found message.
4. Navigate to boot menu and note that the boot entry is not shown.
5. Boot into Linux (easiest with a live USB at this point).
6. Run `efibootmgr` to list boot entries and note that the previously added entry is gone.

## Expected behavior

UEFI boot entries remain until manually deleted.

## Operating System (please complete the following information):

- OS/Distribution: Void Linux
- Linux Kernel Version: 6.13.7

## Additional context

Forum thread: https://community.frame.work/t/uefi-boot-entries-deleted/66169/3

quinchou77, 2025-04-01:
efibootmgr writes boot entries to BIOS variables, but this behavior depends on the BIOS's boot entry handling. We will examine the code design to determine if this is a bug or expected behavior.

closed 2025-05-07 by quinchou77

JohnAZoidberg, 2025-08-24:
> [@quinchou77](https://github.com/quinchou77) Oh, come on. If this is not a bug in the software, then it's a bug in the design.

It's a trade-off of keeping all entries vs keeping only entries that are actually still bootable. Thanks for your feedback, we may revisit this trade-off. 

But one thing that is always guaranteed to work is `EFI\BOOT\BOOTX64.EFI`. If you place a efistub or bootloader there it'll always show up.

Ofenhed, 2025-08-24:
@JohnAZoidberg Thank you, that's good to know.

I currently have 4 boot entries all booting the same unified kernel image, but with different arguments. I found this issue when I considered booting those UKI commands from an external disk. Am I correct in assuming that boot entries with command line arguments are treated the same way?