FrameworkComputer/SoftwareFirmwareIssueTracker #54: Laptop Issue

deepcube, 2025-03-17:
## Device Information

### System Model or SKU

Framework Laptop 13 (AMD Ryzen™ 7040 Series)

### BIOS VERSION

3.05

### DIY Edition information

Memory: Crucial 2x16GB DDR5 5600MHz CT2K16G56C46S5
Storage: Samsung 990 PRO 2 TB MZ-V9P2T0B/AM

## Describe the bug

When using storage password in the BIOS the system fails to resume from suspend to RAM (s2idle) correctly. Adding `iommu=pt` works around the issue.

## Steps To Reproduce

Steps to reproduce the behavior:
1. Enable storage password for NVMe drive in BIOS.
2. Boot into Linux (confirm `CONFIG_BLK_SED_OPAL` is enabled in kernel).
3. Suspend to RAM `echo mem > /sys/power/state`.
4. Wake laptop with keypress.
5. Note laptop does not wake correctly (hangs or resumes but disk is unavailable).
6. Add `iommu=pt` to kernel cmdline, repeat, note resume works.

## Expected behavior

Can resume from suspend to RAM successfully without addition of `iommu=pt` to kernel cmdline.

## Operating System (please complete the following information):
- OS/Distribution: Void Linux
- Linux Kernel Version: 6.13.7

## Additional context

Forum thread: https://community.frame.work/t/responded-amd-7040-setting-storage-password-breaks-resume-from-suspend/42554/24

Eloston, 2025-03-17:
For those reproducing, I would like to note that the Storage password might be disabled in the BIOS settings unless a Master password is set first

I am able to reproduce this issue with the following setup:

- AMD Ryzen 5 7640U
- Storage: WD_BLACK SN850X 1TB
- OS: Debian 12 (bookworm)
- Linux Kernel Version: 6.12.12 (from Debian backports)
- Secure Boot enabled (I believe OP's system has it disabled)

deepcube, 2025-03-17:
> Secure Boot enabled (I believe OP's system has it disabled)

I have reproduced both with secure boot enabled and disabled.

superm1, 2025-04-22:
Can you guys please try [this patch](https://lore.kernel.org/linux-pci/20250422133944.1940679-1-superm1@kernel.org/T/#u)?

Eloston, 2025-05-13:
@superm1 I tried your v2 patch ([link](https://lore.kernel.org/linux-pci/20250424043232.1848107-1-superm1@kernel.org/raw)) on Linux 6.12.22 (technically on top of Debian's `6.12.22-1~bpo12+1`). I tested suspend and resume using amd-debug-tools' `amd_s2idle.py` script

* When I remove `iommu=pt` from the kernel cmdline, resume from suspend crashes the system (screen is black, unresponsive to inputs, and I have no system logs nor s2idle results persisted when I hard reset the system).
* When I keep `iommu=pt` in the cmdline, resume from suspend works normally.

tl;dr there seems to be no change with your patch applied.

superm1, 2025-05-13:
> @superm1 I tried your v2 patch ([link](https://lore.kernel.org/linux-pci/20250424043232.1848107-1-superm1@kernel.org/raw)) on Linux 6.12.22 (technically on top of Debian's `6.12.22-1~bpo12+1`). I [...]

Thanks for checking. It was a bit of a long shot.

I feel that likely the problem here is a missing unity mapping in the ACPI IVRS table for the region that the BIOS uses for authentication.