FrameworkComputer/SoftwareFirmwareIssueTracker #58: FW16 Freeze then Hang (FTH)

jcdutton, 2025-03-24:
## Device Information

### System Model or SKU
 [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)
No dGPU.

### BIOS VERSION
3.0.5

Windows:
N/A

Linux: 
Open a terminal and run the following command
```sudo dmidecode --string bios-version```
03.05

### DIY Edition information
Memory: Manufacture and SKU
Kingston Fury Impact: Part Number: KF556S40-32
2x making 64GB total.
Storage: Manufacture and SKU
Model Number:                       WD_BLACK SN850X 1000GB
Firmware Version:                   620361WD

### Port/Peripheral information
1. USB-C card, nothing plugged in.
2. Empty
3. Empty
4. Empty
5. USB-C card, FW16 PSU plugged in.
6. USB-A card, nothing plugged in.

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] No

## Describe the bug
This has only happened to me once so far.
The symptoms are:
Power plugged in.
Playing a video.
Laptop hangs, screen freezes, laptop plays audio in a short loop.
Wait 20 seconds, no automatic reboot.
Wait 60 seconds, still no reboot.
Force reboot by pressing the power button for 10 seconds.
No logs are stored, so no crash log is available.
No pstore crash log output.
No useful S5_RESET_STATUS, because I had to manually long press the power button to reset it.
EC port80 output runs over the 4096 log limit I had so no useful output captured their either.
I think that while it was in the "hang" state, it was still outputting port 80 output (keeps repeating the pattern):
Log index, Port 80 Value, Port 80 Value in ASCII or a decoded value. 
"00005324","e825f022",".%.."
"00005325","e825f028",".%.("
"00005326","e825f90e",".%.."
"00005327","e825f90d",".%.."
"00005328","e825f90e",".%.."
"00005329","e825f90d",".%.."
"0000532a","e825f022",".%.."
...
Those are all values it also outputs when the laptop is running fine.
Long press of power button to power off gives:
"000054af","00001001","(S3->S0)"

## Steps To Reproduce
Steps to reproduce the behavior:
1. Start from a powered off laptop.
2. Power on laptop
4. Wait a random amount of time. Play videos, netflix, youtube etc.
5. System freezes and does not self-reboot.

Note: I generally have the power plugged in most of the time. For all the FTH I have seen, the power was plugged in at the time.

## Expected behavior
It should not randomly freeze then hang.  (FTH)

## Screenshots
N/A

## Operating System (please complete the following information):
 - OS/Distribution: Linux/Ubuntu
 - Version: 24.04
- Linux Kernel Version: `uname -a`  6.13.7  <- Mainline compiled kernel.

## Additional context
Add any other context about the problem here.

CDRXavier, 2025-04-01:
Related? (Sync Flood)
https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/41#top

jcdutton, 2025-04-06:
Just saw another of these.
No extra info to report.
The (Sync Flood) one results in a automatic reboot after 20 seconds, so this is different, because there is no automatic reboot when this one happens.

sinatosk, 2025-04-30:
This is an issue I too am experiencing ( Linux ) and I'm not using luks/dm-crypt or SED/OPAL

Now using Linux 6.15-rc4 but this issue has been going on back starting with 6.12.0/6.12.1 and upgraded BIOS to 3.05 on 21st November 2024.

The common thing I've noticed with FTH and FTR ( #41 ) issues is none of these issues happen until I start changing between AC ( Framework 180W power adapter ) and DC ( battery ). One other saying the same back on the Framework community forum last time I checked

As long as I stay in AC after system is booted ( including suspend/resume ), all is good. based on what @jcdutton and some others have said, I think it's something to do with how the BIOS/EC is managing power state.

There was one bug in Linux sometime in 6.13 that caused a freeze but that was fixed and there were warnings/errors about it

The FTH/FTR, no warnings and/or errors and this has been going on since BIOS 3.05

I've disabled the battery extender since BIOS 3.04 and charge limit has been set to 80% since day one.

jcdutton, 2025-04-30:
I am running my own firmware on the EC that fixes all the charge / discharge cycling bugs with the FW EC firmware. I wrote the bug fixes myself.
I observed the problem about once a month. It has not been a month yet, but since using my own EC firmware, I have not observed either the FTH or FTR problem.
For those interested, my EC Firmware changes and install instructions are here:
https://github.com/jcdutton/EmbeddedController/wiki

jcdutton, 2025-05-26:
I have recently observed this problem again with my FW16 AMD 7840HS.
I cannot prove it 100% yet, but it appears to happen more often when the PSU is connected to port 4,5.
I used to use port 4,5 for PSU when I reported this problem initially, For whatever reason, I have been using port 1,2 for PSU, and did not see the problem. I recently switched back to port 4,5 and am seeing the problem again.

Has anyone else see any correlation between PSU port and the issue "FTH" appearing ?

For me it has often be with me playing a youtube video and then system hangs, and plays the audio in a loop from the speakers.

Another new observation:
While in "FTH" state, using EC CCD.
I see many "SM-RMI Error 4". About one a second, repeating.
"apreset" does nothing.
"apshutdown" does shut down the laptop, like switching it off.

Although the Capslock light toggles when pressed, no input from the keyboard reaches the CPU.
I know this because I have rigged "sysrq-h" to output a port80 code, and I don't see that code appear on the EC CCD.

So, the new ask is, does changing the port the PSU is plugged into change how often people see FTH ?

jcdutton, 2025-06-04:
I got a FTH today, while on battery. No PSU plugged in at all.
The same "SM-RMI Error 4" message on the EC console.

sinatosk, 2025-06-07:
spoke too soon, it just FTH twice in port 2.

One of the things I've noticed after FTH, I reboot and then put my session back into the same state as my previous, it FTH in less than 10 minutes.

I've noticed this numerous times where I don't FTH for days or weeks ( this is on AC ) and when it does FTH, reboot and try to continue as before and it FTH in less than 10 minutes, it's like there something lingering for some time. So I put things in a less demanding state ( as mentioned previously )

sinatosk, 2025-07-30:
I've bought [Crucial 128GB Kit (64GBx2) DDR5-5600 SODIMM](https://uk.crucial.com/memory/ddr5/CT2K64G56C46S5) and installed today.

After installing the memory, I reset the BIOS and later did my usual stuff to try and trigger FTR/FTH and it FTH in less than an hour, not surprising but thought I add that just incase

sinatosk, 2025-08-06:
ok, maybe spoke to soon

After the (what I thought FTH but a crash) I looked at the kernel logs and noticed that there was some errors related to tasks being soft locked and noticed my battery was quite warm.

After rebooting, battery health went from 99% (before BIOS 3.06 update) this morning down to 62% after doing BIOS update and later crash.

No idea what's going on with the battery, I'll buy a new one...

how did my battery lose approximately 32Wh in less than 6 hours and now it's only charging to 53Wh ( 85-86Wh earlier, BIOS update )

sinatosk, 2025-09-20:
Framework 16 AMD Ryzen 7 7840HS using Radeon 780M
RAM/Memory: [128GB ( 2x64GB ) - Crucial 128GB Kit (64GBx2) DDR5-5600](https://uk.crucial.com/memory/ddr5/ct2k64g56c46s5)
NVME 2280: [Western Digital SN850X 2TB - Firmware 620361WD](https://shop.sandisk.com/en-gb/products/ssd/internal-ssd/wd-black-sn850x-nvme-ssd?sku=WDS200T2X0E-00BCA0)
NVME 2230: [Western Digital SN770M 2TB - Firmware 731120WD](https://shop.sandisk.com/en-gb/products/ssd/internal-ssd/wd-black-sn770m-nvme-ssd1?sku=WDBDNH0020BBK-WRSN)
BIOS: 3.07
Gentoo Linux 2.17 ( Linux 6.17-rc6 mainline realtime, compiled by clang 21.1.1 march and mtune set to znver4)
KDE Plasma 6.4.5 Wayland

Well, with BIOS 3.05 and 3.06, I'm still getting FTH issues

The last two FTH's I've had seem to be happening during usage regardless of the processor work load ( CPU or iGPU or CPU+iGPU ) and performance profile and if I'm using battery or power adapter

I've upgraded to BIOS 3.07 and not done my usual triggers yet so I don't know what the FTH situation is like

While not related, I'm still experiencing [FTR's](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/41) with BIOS 3.05, 3.06 and now 3.07 which so far seems to be happening after resuming

simonsays-techtalk, 2026-02-05:
I have basically the same issue freezes and lockups, randomly (to me) . but maybe my cause is different? This list goes on and on and on, hundreds om messages in dmesg. I am in contact with framework support. I see many USB related issues as well. It has become worse since 1: I reset Bios, 2: I swapped the expansioncards around (no dGPU).0000:00:08.1 points to USB controller.  

[27383.488939] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27385.121102] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27386.553529] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27388.103239] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27391.330557] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27392.833641] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27394.267683] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27395.681075] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27397.153114] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27398.592683] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27401.530631] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27403.097665] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27404.515993] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27407.480228] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27408.932705] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27412.001919] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27415.072817] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27416.488661] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27418.018774] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27419.651823] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27421.089865] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27424.224845] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27425.844861] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27427.330070] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27428.887823] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27430.348146] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27432.004927] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27433.509084] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27435.113424] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27436.753791] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27438.419995] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27440.001061] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27444.593060] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27446.245159] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27447.810136] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27449.441972] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27451.080110] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27452.531120] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27454.060156] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27455.521400] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27457.042147] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27458.481175] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27460.003211] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27461.593249] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27463.034211] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27464.452236] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27465.937507] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27467.474290] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27470.746451] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27474.026241] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27474.058527] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27475.478320] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27477.024265] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27478.684361] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27481.957374] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27483.441419] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27485.082456] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27486.578112] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27488.201916] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27489.633516] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27491.049997] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27492.670456] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27494.243494] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27495.906579] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27497.377739] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27500.459188] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27501.939577] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27503.529620] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[... 277 lines cut ...]
[27981.617086] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27983.034111] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27986.169055] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[27987.598220] pcieport 0000:00:08.1: PME: Spurious native interrupt!
[... cut ...]
[  569.984912] usb 1-1: new full-speed USB device number 8 using xhci_hcd
[  570.147093] usb 1-1: New USB device found, idVendor=32ac, idProduct=0002, bcdDevice= 0.00
[  570.147110] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  570.147118] usb 1-1: Product: HDMI Expansion Card
[  570.147123] usb 1-1: Manufacturer: Framework
[  570.147129] usb 1-1: SerialNumber: 11AD1D00A49C40152B2B0B00
[  570.228294] hid-generic 0003:32AC:0002.0007: hiddev2,hidraw6: USB HID v1.11 Device [Framework HDMI Expansion Card] on usb-0000:c1:00.3-1/input1