FrameworkComputer/SoftwareFirmwareIssueTracker #60: Framework 13 11th Gen S3 also not working properly in Windows

ohne-ende, 2025-03-30:
## Device Information

### System Model or SKU

Please select one of the following

- [x] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
3.20

### DIY Edition information

Memory: 2x Crucial CT32G4SFD832A.C16FF 32GB DDR4-3200
Storage: Samsung SSD 980 1TB

### Port/Peripheral information
1. USB-C Card -> Sonnet eGPU Breakaway Box 750ex (contains Radeon RX7800 XT)
2. USB-A Card 
3. USB-A Card -> Dell WD15 (K17A) [with Adapter so it doesn't output to the displays attached on it]
4. directly -> UGREEN CM578 SSD enclosure

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug
I've read under the Known issues that S3 might not work for Linux users. I"ve just installed the new version to hopefully get WiFi 6e to work (still won't work, Thanks Intel), not thinking that the issue also appears on Windows, bur here we are. I mostly set my Laptop into sleep (suspension) so I don't need to close and reopen most things the next day. The S3 mode seems to work just fine for around 2 minutes, after which it just turns on again.

## Steps To Reproduce
Steps to reproduce the behavior:
1. Open start menu
2. Click on Power options
3. Click on suspend (Energie sparen in the German version)
4. Wait 2 minutes
5. Laptop wakes up without any interaction

## Expected behavior
The laptop should stay in S3 until I press the power button or any key on the keyboard

## Operating System
 - OS/Distribution: Windows 11 Pro German
 - Version: 23H2

## Additional context
I've added the registry entry in Windows so I don't have to suffer under Microsoft's shitty S0 implementation

JohnAZoidberg, 2025-03-30:
> I've added the registry entry in Windows so I don't have to suffer under Microsoft's shitty S0 implementation

Could you provide some more information about this please?

S0 means the system is active, it's not a sleep mode. What are you trying to achieve?

JohnAZoidberg, 2025-03-30:
> I"ve just installed the new version to hopefully get WiFi 6e to work (still won't work, Thanks Intel),

New version of what? Bios? Drivers?
What wifi card are you using?
When you say 6e doesn't work, you mean wifi is working but the 6e specific frequency bands aren't?

quinchou77, 2025-04-08:
Do you want to use  S3 not Modern Standby(S0iX)?
I assume the system now is in S0iX when you click the sleep button.

ohne-ende, 2025-04-09:
So, I wanna use the S3 mode again, where the system is in actual sleep mode again. I didn't remove the registry entry that disables S0, so S3 is used, which is under:
`Computer\HKLM\SYSTEM\CurrentControlSet\Control\Power\`
Name: `PlatformAoAcOverride`
Value: `0x00000000 (0)`
I have the Intel AX210 wifi card installed in my laptop (driver version: 23.120.0.3), which somehow won't connect to a WiFi 6e network although USB WiFi 6e cards (w/ Mediatek chipset) do connect to those networks, so I know the region settings in Windows aren't blocking the 6GHz range. 2.4 and 5GHz work just fine on both options.
In the BIOS changelog was mentioned that WiFi 6e support was added to BIOS version 3.20 so I assumed that it would now work but it still seems to just be driver-related.
I've read the warning that I couldn't downgrade after the update and Linux users are experiencing the same issues.
At this time I just shut down my laptop because I don't want to use S0 sleep as an alternative that'll roast my CPU and drain my battery in no time

quinchou77, 2025-05-07:
You can use "powercfg /a" command to check what the current sleep type supported on you system. However, we don't recommend to use legacy S3 since it is not the Intel 11th Gen and Windows 11 POR. You may get some unexpected issue when you using Legacy S3 mode.