FrameworkComputer/SoftwareFirmwareIssueTracker #62: Upower issues with Framework laptops

sanderkoenders, 2025-04-08:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [x] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [x] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSIONS
- 03.05 (my bios version)
- 03.08 (Jan Alexander Steffens, as mentioned [here](https://gitlab.freedesktop.org/upower/upower/-/issues/306#note_2855316))

### DIY Edition information
If you are experiencing an issue on a DIY system, Please also fill out the memory and storage devices you are using. 

Memory: DDR5-5600 - 32GB (1 x 32GB)
Storage: WD_BLACK™ SN850X NVMe™ M.2 2280 - 2TB

### Port/Peripheral information
Not applicable

### Standalone Operation
Not applicable

## Describe the bug
I am experiencing an issue related to battery status reporting on my Framework 16 laptop running Arch Linux. It appears that the laptop is not delivering the correct summarized battery status, which is causing problems with how upower (and potentially other tools) interact with the system. This issue was discussed in [this GitLab issue on upower](https://gitlab.freedesktop.org/upower/upower/-/issues/306), where it was mentioned that the Framework should provide a summarized battery status when performing certain actions on the battery. Without this, user-facing applications struggle to process battery information correctly.

Concretely this means that users are getting repetitive (and annoying) “connect” and “disconnect” sounds every few seconds in their desktop environment because the status is constantly cycling between “start charging” and “stop charging”.

Unfortunately I am not an expert on this particular subject but I would like to further debug this issue where possible because the recent release of upower breaks a lot of Framework systems that run Linux. I can imagine that it is in both our best interests to solve this issue for the people that run Linux on their Framework laptops. Please do let me know what I can do to help debug this issue. Please also review all the information that is already given in the Gitlab discussion.

Furthermore, I acknowledge that Archlinux is not officially supported on the framework laptops but upower is also present in Fedora Linux and I suspect users on Fedora will face the same issues when the newest version of upower is released for them.

## Steps To Reproduce
Steps to reproduce the behavior:
1. Install upower version 1.90.8
2. Connect a charger (in my case Lenovo Docking Station Thunderbord 4)
3. Observe the cycle of "connected" and "disconnected"

## Expected behavior
As far as my understanding goes Framework should report the correct battery summary to upower as described [here](https://gitlab.freedesktop.org/upower/upower/-/issues/306#note_2855780):

> They have to give us a summarized battery status when they are doing some features on your battery, rather than only charging/discharging when the line power is attached. Otherwise, it confuses the user, and the upper layer applications are difficult to process such kind of battery status information.

## Screenshots
Not applicable

## Operating System (please complete the following information):
 - OS/Distribution: Archlinux
 - Version: rolling release
- Linux Kernel Version: 6.13.8-arch1-1

## Additional context
- [Issue description on upower](https://gitlab.freedesktop.org/upower/upower/-/issues/306)
- [Forum issue](https://community.frame.work/t/upower-issues-with-framework-laptops/67217)

sanderkoenders, 2025-04-11:
As discussed in [comment 2856003](https://gitlab.freedesktop.org/upower/upower/-/issues/306#note_2856003) and [comment 2857205](https://gitlab.freedesktop.org/upower/upower/-/issues/306#note_2857205) this is not an issue with Framework. The issue has been resolved in version 1.90.9

closed 2025-04-11 by sanderkoenders

JohnAZoidberg, 2025-04-11:
Awesome, thanks for the update :)