FrameworkComputer/SoftwareFirmwareIssueTracker #63: Security Updates for INSYDE-SA-2024015 and INSYDE-SA-2024021

Klaas-, 2025-04-16:
## Device Information
Framework 13 12th Gen

### System Model or SKU
FRANDACP08

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [x] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
03.09

## Describe the bug
Your bios vendor has released security updates on 2025-04-08, I would like to know when this is being incorporated into frameworks Bioses. Dell for example has classified this as a high severity issue and has released updates for all their affected laptops in the last couple of days: https://www.dell.com/support/kbdoc/en-us/000285110/dsa-2025-091

## Expected behavior
A clear and concise description of what you expected to happen.

## Screenshots
If applicable, add screenshots to help explain your problem.

## Operating System (please complete the following information):
 - OS/Distribution: [e.g. Windows 11]
 - Version: [Version]
- Linux Kernel Version: `uname -a`

## Additional context
Add any other context about the problem here.

quinchou77, 2025-04-17:
Thanks for information. We will start to plan this fixed, then update the schedule.

quinchou77, 2025-04-22:
We plan to fix INSYDE-SA-2024015 for 12th gen in the next release.
Target date will be around mid-May.

JohnAZoidberg, 2025-04-23:
INSYDE-SA-2024021 is CVE-2024-7344
This is not a vulnerability in Insyde BIOS, it's a third party application that's signed by the Microsoft UEFI keys.
Because we include the Microsoft public keys to be able to boot Windows, we are vulnerable to that.

The mitigation is simple, in future updates we will include dbx entries to blacklist this third party application.

But before that, it's also really easy to mitigate:
- For Windows users the [2025 January 14 Update](https://support.microsoft.com/en-us/topic/january-14-2025-kb5050009-os-build-26100-2894-bdbfb097-ea20-487d-9171-718d15e26f1b) includes a DBX Update
- For Linux users DBX can be updated using LVFS and an updated one is already available: https://fwupd.org/lvfs/devices/com.microsoft.dbx.x64.firmware

quinchou77, 2025-05-28:
We have released Framework Laptop 13 (12th Gen Intel® Core™) BIOS 3.17 beta 2 weeks ago in community. We will move it to stable and release on website this week.

closed 2025-07-04 by quinchou77