FrameworkComputer/SoftwareFirmwareIssueTracker #64: [BIOS Firmware Laptop16] Impossible to update 3.03 => 3.05

borrougagnou, 2025-04-20:
### Device Information
Framework Laptop 16
Extension Card: HDMI/ETH/AUDIO/USB/USB-C/GPU AMD 7700s

BIOS: IFGP6.03.03
BIOS Release date: 2024-03-27 (yyyy-mm-dd)

Ubuntu 22.04.5
Windows 11 Pro

---
Hello guys !
I have a problem when I update bios from 3.03==3.05 in Framework Laptop 16

when I try with Linux: 

[image]

when I try with Windows:

[image]
(And when I force reboot I have EFI ERROR 2: decompression failed)

Thank you in advance.

borrougagnou, 2025-04-21:
I think I fixed my problem
because now I have the "Framework gear"

First thing is: check the EFI partition size !
by default I have 100MB for this partition, after installing Windows/Ubuntu/other the size will be reduced

including the EFI capsule, the size at the end is: ~30MB...

And IDK why, the system need to use this "space" to upgrade the system an 30MB is clearly not enough.
so I backuped/removed all EFI files.

Just keep my Ubuntu EFI file,
I also `sudo apt purge fwupd fwupd-signed -y && sudo apt install fwupd fwupd-signed`
then I did again:
`fwupdmgr refresh --force`
`fwupd get-updates`
`fwupd update`

And now it work !

Conclusion, the EFI partition is needed when you update your system, and it's not limited to the "size of the EFI capsule file"

JohnAZoidberg, 2025-04-22:
> by default I have 100MB for this partition

That's pretty small, especially if you dualboot from that partition. Because then there'll be almost no space left.
Yes, you need AT LEAST ~35MB.

For FWUPD, see https://github.com/fwupd/fwupd/discussions/8341
It requires at least 90MB, but we'll enable update with 35MB free space for updates in the future.

borrougagnou, 2025-04-23:
Hello !
Thank you for your answer, yes it would be cool !
The MR is cool ! and it's released in UFW 2.0.8
Debian already put this version in the repo sid and trixie (debian 13) !
Fedora too !

it mean it will be available for the next release