FrameworkComputer/SoftwareFirmwareIssueTracker #67: F13 AMD AI 300 - Battery charge limit ignored on subsequent boots

YLee42, 2025-04-30:
## Device Information

### System Model or SKU

- [x] Framework Laptop 13 (AMD Ryzen™ AI 300 Series)

### BIOS VERSION
3.03 (out-of-the-box and current)

### DIY Edition information

Memory: Crucial DDR5-5600 2x16GB, CT2KG56C4635
Storage: Crucial P310 1TB

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug
"Battery charge limit" setting is ignored after reboot, after the PC is turned on. This was observed while Battery Life Extender is disabled to isolate the issue.

## Steps To Reproduce
First, a check to verify Battery charge limit works:
1. Boot up F13 while charger is plugged in.
2. Set "Battery charge limit" to a new number lower than the current battery percentage.
3. Save & exit. Boot continues and port LED should turn white.

Now, set up the conditions for reboot:
1. Power off computer.
2. Unplug charger.
3. Wait 2 minutes.

Now, steps that produce the error:
1. Plug in the charger. LED should turn white as the battery is at or above the charge limit.
2. Turn on the PC. The LED eventually turns orange and stays there.
3. Verify in the OS that the battery is charging above the set limit.
4. Unplug and plug the charger again. Verify that the LED is orange and that the battery continues to charge above the set limit.

Powering off, unplugging, and waiting re-sets the issue, but the issue re-occurs on boot again.

## Expected behavior
Battery charge limit is honored regardless of the combination of events.
1. Charger plugged while powered off (works)
2. Powered on while plugged in (issue)
3. Charger plugged while powered on (issue)

## Operating System (please complete the following information):
 - OS/Distribution: Windows 11

GreyXor, 2025-04-30:
I have the same issue. charge limit feature is broken now. It's not correctly respected.

jcdutton, 2025-04-30:
Are you able to run "ectool version". It is a Linux command, but there might be a Windows equivalent.  
Maybe the FW tool can be made to work on Windows. It is a rust program that you might be able to compile on windows.
Alternatively, use a Linux Live USB stick.

framework_tool --versions

Post the output here.

YLee42, 2025-04-30:
I found a Windows build of ectool [here](https://gitlab.howett.net/DHowett/ectool/-/artifacts) and ran it locally. Output here:

```RO version:    lilac-3.0.3-413f018
RW version:    lilac-3.0.3-413f018
Firmware copy: RO
Build info:    lilac-3.0.3-413f018 2025-03-06 05:45:28 marigold2@ip-172-26-3-226
Tool version:  0.0.1-isolate Nov 13 2024 none```

jcdutton, 2025-04-30:
FW have not published any of the source code for the "lilac" EC Firmware.

jcdutton, 2025-04-30:
Would you be able to clarify.
Say you set the charge limit to 80%  and the current charge is 80%.
You then set the charge limit to 60%. What are you expecting?
FYI, FW expect it to immediately discharge to 60%, even with the AC plugged in.
I did not like that, so part of the reason I wrote my own bug fixes to the EC Firmware for my FW16 AMD 7840HS.

YLee42, 2025-04-30:
Whether the laptop discharges or relies on AC power is not high-priority on my list of concerns; the issue here is that objectively incorrect behavior happens, which is charging to 100%. It defeats the whole point of a battery charging limit if the limit is only enforced during that session for the EC.

As for how it looks at the BIOS setting level:
* The BIOS does remember what the entered charging limit was; re-entering the BIOS shows the limit as entered.
* The BIOS does _not_ re-inforce the setting even after you re-enter the BIOS settings. I need to enter a different limit in order for the EC to enforce a charging limit, and of course a power-down makes the limit ineffective again.

Parsifal-2045, 2025-05-03:
> Whether the laptop discharges or relies on AC power is not high-priority on my list of concerns; the issue here is that objectively incorrect behavior happens, which is charging to 100%. It defeats [...]

I've observed the same issue since I got the laptop, running Fedora 42. To slightly expand, the setting is correctly maintained if rebooting or if the period of time the laptop stays off is "short enough" but any longer shutdown will not enforce the limit.

jcdutton, 2025-05-03:
The messaging from the BIOS to the EC is via HC (Host Commands).
At boot up the BIOS sends an HC to the EC telling it what the charge limit is.
The EC then just has some bugs in its code that cause it to accidentally not take it into account in particular edge cases.
If you wish you can download and install my version of the "ectool". Instructions here:
https://github.com/jcdutton/EmbeddedController/wiki/ectool

To stop it charging or discharging:
`sudo ectool chargecontrol idle`

To cause it to start discharging down to the BIOS set Battery Limit level.
`sudo ectool chargecontrol discharge`

To cause it to start charging, if it is below the BIOS set Battery Limit level:
`sudo ectool chargecontrol discharge`

They are not fixes for the bugs, but at least let you bring the charge back to where you want it.
Due to EC bugs, the above commands have limited success, but might help.

I have written a version of the EC firmware that fixes all the bugs, but it currently has only been tested on FW16 AMD 7840HS and FW13 AMD 7840U/7640U.
I don't have a version that would work on the FW13 AMD AI 300.

tlvince, 2025-05-05:
Presumably, this is the same issue as found on the 7040 series, BIOS v3.07:

> We found a known issue that "If the battery extender mode is disabled, and charge limit is set below 100%, the charge limit will not be enforced causing the battery to charge to 100%."

https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/52#issuecomment-2720106686

It was fixed in [BIOS v3.09](https://community.frame.work/t/framework-laptop-13-ryzen-7040-bios-3-09-release-beta/68601). @quinchou77, could you confirm? Will the same fix be brought over to the AI 300 series?

quinchou77, 2025-05-05:
> Presumably, this is the same issue as found on the 7040 series, BIOS v3.07:
> 
> > We found a known issue that "If the battery extender mode is disabled, and charge limit is set below 100%, the charge limit will not be enforced causing the battery to charge to 100%."
> 
> [#52 (comment)](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/52#issuecomment-2720106686)
> 
> It was fixed in [BIOS v3.09](https://community.frame.work/t/framework-laptop-13-ryzen-7040-bios-3-09-release-beta/68601). [@quinchou77](https://github.com/quinchou77), could you confirm? Will the same fix be brought over to the AI 300 series?

Yes, it will also be fixed in the next AI 300 Series BIOS. We are planning for each product update. I will update the schedule for AI 300 Series.

JohnAZoidberg, 2025-05-06:
> Are you able to run "ectool version". It is a Linux command, but there might be a Windows equivalent. Maybe the FW tool can be made to work on Windows. It is a rust program that you might be able to [...]

FYI, prebuilt binaries of framework_tool are available for windows: https://github.com/FrameworkComputer/framework-system/releases

wturgeonw, 2025-05-15:
I have a similar issue, my GaN charger stays at the 80% charging limit, but my dock doesn't care and charges over 80%. Hoping the upcoming BIOS update fixes this too
batch 4 FW13 AMD AI 340.

YLee42, 2025-05-16:
Could you note here which dock you are using to charge? Probably good to have that data point.

wturgeonw, 2025-05-16:
Right, good idea. It's a Plugable UD-CA1A. https://plugable.com/products/ud-ca1a/

YLee42, 2025-05-25:
> FYI, prebuilt binaries of framework_tool are available for windows: https://github.com/FrameworkComputer/framework-system/releases

Thank you for the heads-up—confirming that while booted in Windows the issue is circumvented by executing from terminal (assuming same directory as .exe):

`.\framework_tool --charge-limit [INTEGER OTHER THAN PREVIOUSLY SET LIMIT]`

This acts similar to re-entering a different battery limit in the BIOS as I previously documented.

There may be other options that work too, but didn't want to fuss around too much with hardware I don't understand.

quinchou77, 2025-07-04:
> Hi [@quinchou77](https://github.com/quinchou77), do you already have an ETA for the new AI 300 bios ?

Sorry for late reply. Since we are waiting for AMD PI update for a USB4 issue, so it is delay. The ETA for beta version is 7/10.

GreyXor, 2025-07-11:
> F13 AMD AI300 BIOS 3.04 beta is in community, here is the link of the forum. https://community.frame.work/t/framework-laptop-13-ryzen-ai-300-bios-3-04-release-beta/71934

Thanks!
Upgrade completed successfully. The charge limit issue appears to be resolved.

quinchou77, 2025-07-17:
The latest code of this version has not push to external. We will push it when this version gets stable next week.

YLee42, 2025-07-22:
As the issue opener, I wanted to update that upgrading to 3.04 stable appears to have fixed the issue. It's also recognized by Windows as "smart charging":

[image]

quinchou77, 2025-07-24:
> As the issue opener, I wanted to update that upgrading to 3.04 stable appears to have fixed the issue. It's also recognized by Windows as "smart charging":
> 
> <img alt="Image" width="549" height="336" src="https://private-user-images.githubusercontent.com/7243906/469489424-2539706e-f9a2-4fde-b09f-5cfe3a8a9130.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTM0MDExNTIsIm5iZiI6MTc1MzQwMDg1MiwicGF0aCI6Ii83MjQzOTA2LzQ2OTQ4OTQyNC0yNTM5NzA2ZS1mOWEyLTRmZGUtYjA5Zi01Y2ZlM2E4YTkxMzAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDcyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA3MjRUMjM0NzMyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTU2MTZkMWU5NGUzNGRjYjEzNGIzNThhYTZmMzgzZDIzOGM2ZDVlY2FiZDMyMDA5ZDRkMTY1YjZkODVmMThiYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.G8QbZ_QNrjkJ7eWhyMxI8beqWDql-xnMEmjA60yYVJc">

This version adds a smart charging icon feature, based on the ACPI spec definition.

arthursimonyantsaero, 2025-07-25:
I can also report that successful update over LVFS fixed the issue.

quinchou77, 2025-08-15:
Issue is fixed with the latest bios 3.04, closed.