FrameworkComputer/SoftwareFirmwareIssueTracker #68: FW16  Schematics need updating

jcdutton, 2025-05-16:
Hi,

I have looked at the FW16 schematics here:
https://github.com/FrameworkComputer/Framework-Laptop-16/blob/main/Mainboard/Mainboard_Interfaces_Schematic_Framework_Laptop_16_7040_Series.pdf

The power to the USB ports is provided by at SY8370C chip.
One for +5V ALW A, that feeds 3 USB ports, and another +5V ALW B, that feeds the other 3 USB ports.
The SY8370C can output 11A.
So that should be enough for 15W from ever USB port at the same time.
But the FW16 can only do 15W from one USB port at a time.

Please update the schematic to match what the FW16 can actually do.

JohnAZoidberg, 2025-09-25:
Sorry what exactly do you think is wrong on the schematic?

jcdutton, 2025-09-25:
What is wrong with the schematic?
"The power to the USB ports is provided by at SY8370C chip.
One for +5V ALW A, that feeds 3 USB ports, and another +5V ALW B, that feeds the other 3 USB ports.
The SY8370C can output 11A.
So that should be enough for 15W from every USB port at the same time."
So, the existing schematic implies that 15W should be available from every USB port at the same time.
But, the FW documentation here, says it cannot.
https://knowledgebase.frame.work/expansion-card-slot-functionality-on-framework-laptop-16-amd-ryzen-7040-series-rkUjGm7cn
"Ports 1,2,4,5 are shared on a first come first serve basis. So they are 1.5A by default, but one port from any of these can negotiate 3A over PD. Once that is taken, the other ports are limited to 1.5A. Ports 3,6 are 900mA."

So, my assumption that either the schematic is wrong, or the FW documentation is wrong.