FrameworkComputer/SoftwareFirmwareIssueTracker #69: Laptop Issue

dimitris-personal, 2025-05-18:
## Device Information

### System Model or SKU Framework laptop 13

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [x] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
03.09

### DIY Edition information

Memory: Bought as part of the laptop order from Framework
Storage: Samsung SSD 990 PRO 2TB

### Port/Peripheral information
N/A

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug
Fan RPM ramp-up is now very very delayed while the processor is virtually at 100 deg C

## Steps To Reproduce
- While running `stress-ng -c 16`:
- Watch temperature by periodically running `sensors`

## Expected behavior
Previously the fan would quickly spin up.

## Regression behavior
Fan does not spin up for a long while (seems over a minute) while the processor is at 100 deg C according to `sensors`.

## Operating System (please complete the following information):
 - OS/Distribution: Fedora Workstation
 - Version: 42
- Linux Kernel Version: 6.14.6-300.fc42.x86_64

## Additional context
Laptop is on a stand/arm which leaves some space beneath so there is airflow into the fan intake.

amstan, 2025-05-19:
Hey Dimitris. Yeah, we noticed this behavior on another project, we have some ideas on how to fix it, it should be applicable to your device too.

In the meantime you can use something like `sudo ectool pwmsetfanrpm` or `sudo framework_tool --fansetrpm` to manually adjust the fan before you expect a sudden load (but i'm sure that will get tiring quickly). You can set it back to auto once the slower fan temperature sensor catches up with reality.

JohnAZoidberg, 2025-05-19:
> Fan does not spin up for a long while (seems over a minute) while the processor is at 100 deg C according to sensors.

What do the rest of the sensors measure at this time when the CPU is not but the fans aren't on yet?

You can nicely log them with `sudo s-tui --csv -r 1`

dimitris-personal, 2025-05-19:
@JohnAZoidberg here you go:

[s-tui_log_2025-05-18_18_50_00.csv](https://github.com/user-attachments/files/20276253/s-tui_log_2025-05-18_18_50_00.csv)

and because I'm ~~lazy today~~ not that great with spreadsheets, Claude's visualization of temp and fan readings (forgot to say "briefly" in the prompt so it went a bit above and beyond but oh well):

[image]

Overall this feels/looks better than last time I tried `stress-ng`.  It's on a fresh boot after some updates that don't look super relevant to me:

<details>
<summary>Today's update</summary>

```
$ dnf history info 346
Transaction ID : 346
Begin time     : 2025-05-18 18:00:19
Begin rpmdb    : 2c5aca09d524dfd575910bcb3614dfb92ac52131dfd0e7e0b9d9df1cf8143506
End time       : 2025-05-18 18:00:36
End rpmdb      : 19cc81b2e14047efc8aa5aeb2438ac73a985b057c9d805b9e64be873b1d31849
User           : 1000 D <d>
Status         : Ok
Releasever     : 42
Description    : dnf --refresh up
Comment        : 
Packages altered:
  Action   Package                                        Reason          Repository
  Install  simdutf-0:6.5.0-1.fc42.x86_64                  Dependency      updates
  Upgrade  chromium-0:136.0.7103.113-1.fc42.x86_64        User            updates
  Upgrade  chromium-common-0:136.0.7103.113-1.fc42.x86_64 Dependency      updates
  Upgrade  cups-1:2.4.12-3.fc42.x86_64                    Group           updates
  Upgrade  cups-client-1:2.4.12-3.fc42.x86_64             Dependency      updates
  Upgrade  cups-filesystem-1:2.4.12-3.fc42.noarch         Dependency      updates
  Upgrade  cups-libs-1:2.4.12-3.fc42.x86_64               Dependency      updates
  Upgrade  cups-ipptool-1:2.4.12-3.fc42.x86_64            Weak Dependency updates
  Upgrade  gdb-0:16.3-1.fc42.x86_64                       Dependency      updates
  Upgrade  gdb-headless-0:16.3-1.fc42.x86_64              Dependency      updates
  Upgrade  pam-0:1.7.0-5.fc42.x86_64                      Dependency      updates
  Upgrade  pam-libs-0:1.7.0-5.fc42.x86_64                 Dependency      updates
  Upgrade  simple-scan-0:48.1-1.fc42.x86_64               Group           updates
  Upgrade  xdg-desktop-portal-0:1.20.1-3.fc42.x86_64      Group           updates
  Replaced chromium-0:136.0.7103.92-1.fc42.x86_64         User            @System
  Replaced chromium-common-0:136.0.7103.92-1.fc42.x86_64  Dependency      @System
  Replaced cups-1:2.4.12-2.fc42.x86_64                    Group           @System
  Replaced cups-client-1:2.4.12-2.fc42.x86_64             Dependency      @System
  Replaced cups-filesystem-1:2.4.12-2.fc42.noarch         Dependency      @System
  Replaced cups-ipptool-1:2.4.12-2.fc42.x86_64            Weak Dependency @System
  Replaced cups-libs-1:2.4.12-2.fc42.x86_64               Dependency      @System
  Replaced gdb-0:16.2-3.fc42.x86_64                       Dependency      @System
  Replaced gdb-headless-0:16.2-3.fc42.x86_64              Dependency      @System
  Replaced pam-0:1.7.0-4.fc42.x86_64                      Dependency      @System
  Replaced pam-libs-0:1.7.0-4.fc42.x86_64                 Dependency      @System
  Replaced simple-scan-0:46.0-3.fc42.x86_64               Group           @System
  Replaced xdg-desktop-portal-0:1.20.0-1.fc42.x86_64      Group           @System
```

</details>