FrameworkComputer/SoftwareFirmwareIssueTracker #70: Unsupported userspace charge control

tlvince, 2025-05-21:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [x] Framework Laptop 13 (AMD Ryzen™ AI 300 Series)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION

3.03

### DIY Edition information

- Memory: [Crucial CT2K16G56C46S5](https://uk.crucial.com/memory/DDR5/CT2K16G56C46S5) 32GB DDR5-5600 SODIMM
- Storage: [SK hynix Platinum P41](https://ssd.skhynix.com/platinum_p41/) 1TB SSD

### Port/Peripheral information

1. USB-C expansion card: Framework 60W charger
2. USB-C expansion card (no peripheral)
3. USB-C expansion card (no peripheral)
4. USB-C expansion card (no peripheral)

### Standalone Operation

- [ ] Yes
- [x] No

## Describe the bug

Loading `cros_charge-control` with [probe_with_fwk_charge_control=1](https://github.com/torvalds/linux/blob/4a95bc121ccdaee04c4d72f84dbfa6b880a514b6/drivers/power/supply/cros_charge-control.c#L253-L256) _should_ provide `/sys/class/power_supply/BAT1/charge_control_{start,end}_threshold` for userspace charge control e.g. [GNOME 48's Preserve Battery Health](https://release.gnome.org/48/#preserve-battery-health) feature.

This doesn't work on the AI 300 series (BIOS 3.03). On the 7040's, BIOS 3.09 [caused a regression](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/52#issuecomment-2884949366).

## Operating System (please complete the following information):

- OS/Distribution: NixOS
- Version: 25
- Linux Kernel Version:  6.14.6

tlvince, 2025-05-28:
> The TLDR version is you need [this kernel patch](https://lore.kernel.org/lkml/20250521-cros-ec-mfd-chctl-probe-v1-1-6ebfe3a6efa7@weissschuh.net/) and probe_with_fwk_charge_control=1.

Thanks. Applied on top of v6.14.7. Works for me:

```
❯ upower -i /org/freedesktop/UPower/devices/battery_BAT1
    ...
    charge-start-threshold:        75%
    charge-end-threshold:          80%
    charge-threshold-enabled:          1
    charge-threshold-supported:          1
```

Although the start threshold seems to be ignored - it hovers around 80% /cc @t-8ch.

Serafean, 2025-05-29:
@tlvince since you're also using charge control, could you please confirm a separate issue?
When I set the start and end thresholds to the same number, it never starts charging. The discharge rate slows down to a minimum when it hits that percentage, but level still falls.

JohnAZoidberg, 2025-05-29:
> When I set the start and end thresholds to the same number, it never starts charging. The discharge rate slows down to a minimum when it hits that percentage, but level still falls.

What do you expect to happen? What's the purpose of setting them to the same number?

> Although the start threshold seems to be ignored

Yes, it's not implemented in the EC

Serafean, 2025-05-29:
Also the KDE UI sort of leads to setting them to the same number.

https://github.com/user-attachments/assets/0e87663a-878d-407c-805c-edf7da307e81
Manually set max to 83, min to 82, then used scroll wheel in "stop charging at" . Note how they're tied.

JohnAZoidberg, 2025-05-29:
> > What do you expect to happen? What's the purpose of setting them to the same number?
> 
> My expectation is for the charge level to stay at that percentage while on AC.
> 
> right now I have start 79%, end 80%.

But it's enough to set the maximum, no? Then the system charges until that number, stops and stays there

closed 2025-06-09 by JohnAZoidberg

Serafean, 2025-07-22:
Yes, I can, kernel 6.15.7, but I'm still on 3.03 
`DMI: Framework Laptop 13 (AMD Ryzen AI 300 Series)/FRANMGCP09, BIOS 03.03 03/10/2025`

I also note that you have
```
    charge-start-threshold:        75%
    charge-end-threshold:          80%
```
Which are the exact same numbers I show, but aren't the limits I have set.

tlvince, 2025-07-22:
Thanks. I can reproduce on 6.15.5 with 3.04, but couldn't with 6.15.5 with 3.03

Serafean, 2025-07-27:
`BIOS 03.04 06/10/2025` , kernel 6.15.8, charge limit sort of works. 
I think it is only the BIOS limit that is respected though, changing the min/max values from userspace doesn't seem to do anything.
And because I'm an idiot who deleted the old kernels almost immediately, checking that a previous version works remains on my TODO list :/ 
My gut tells me it's coming from the patches not being correct anymore.

tlvince, 2025-07-28:
Yeah, I'm using the EC charge limiter for now. It can be controlled via `ectool`, but its less convenient. I've pinged the driver maintainer in the meantime.