FrameworkComputer/SoftwareFirmwareIssueTracker #74: Error, and Solution, upgrading BIOS 3.04 to 3.17

rinchen, 2025-06-06:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [X] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
3.04

## Describe the bug
* The EFI update method fails to update 3.04 to 3.17.
* startup.nsh needed two adjustments to make this work. 

## Steps To Reproduce
Steps to reproduce the behavior:
1. Follow [EFI boot method](https://knowledgebase.frame.work/framework-laptop-bios-and-driver-releases-12th-gen-intel-core-Bkx2kosqq#Linux_BIOS) and fail to update firmware.

## Resolution
I documented the failures, system configuration, and adjustments needed in https://community.frame.work/t/unable-to-update-firmware-from-3-04-to-latest-on-website/70157/10?u=joey_stanford

laffer1, 2025-06-28:
It looks like 3.06 also has problems updating

quinchou77, 2025-07-04:
Thanks for feedback. We are starting to check for it.

quinchou77, 2025-07-11:
With -OD, it means that BIOS update goes through capsule on disk. 
Without -OD, it means that BIOS update goes through capsule on memory.
In previous version, 12th gen doesn't support capsule on disk. I think it is why you can update it after remove the -OD.
I will check what version start to support it, then do batch modification to prevent the problem.