FrameworkComputer/SoftwareFirmwareIssueTracker #75: FW13 Ryzen AI 300 - PD negotiation failure / power cycling with Dell U3225QE (TB4-certified docking with 140W advertised PD output)

ramonwirsch, 2025-06-14:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)
- [x] Framework Laptop 13 (AMD Ryzen™ AI 300 Series)

### BIOS VERSION
Please provide the bios version. 
3.03

### Port/Peripheral information
* Peripheral vendor and name: Dell U3225QE as TB4/PD dock
* In port 1 / left USB4
* Expansion cards: USB-C in all 4 ports, nothing else connected

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug
When connecting the monitor's TB4 input with the laptop, it will negotiate 90W PD for a second (laptop starts drawing power as expected and charging), then the connection will get interrupted / reset and it will repeat this indefinitely. It will never negotiate a data connection while doing this.

## Steps To Reproduce
Steps to reproduce the behavior:
1. Use official Dell TB4 240W cable included with the monitor to connect it to USB4 port
2. See power cycling / connection resetting on repeat

## Steps to workaround
I found the following to get a otherwise fully working USB4 40G connection to the monitor
1. Connect a stronger / preferred power supply to any other port of the Framework (I am using a true 100W PD supply), to prevent it from trying to draw power from the monitor
2. Framework never draws power from the monitor, USB4 connection now works as expected

## Expected behavior
The monitor should work as a power supply with at least 90W (ideally 100W, but this may be the fault of the monitor, possibly because of non-compliance with USB-PD). And we should get a working USB4 connection, while the monitor is being used as power supply for the notebook.

## Screenshots
I have traced the power cycling and the working USB4 connection without power supply from the monitor with a Chargerlab KM003C.
Here a screenshot of the voltage and current showing the cycling
[image]

I cannot attach the raw .sqlite files that the Power-Z Windows tool creates and stretches across the entire voltage graph above. Thus,
here the PD trace from first attaching until the first reset / shutoff in the graph as images (the raw trace includes all of the data and PD framing, but cannot be sensibly shown in a screenshot of the trace)

[image]
[image]
[image]
[image]

I also captured the EC console output (with framework-tool) in parallel:
[U3225QE broken - FW13 Strix EC Log .txt](https://github.com/user-attachments/files/20739932/U3225QE.broken.-.FW13.Strix.EC.Log.txt)

## Operating System (please complete the following information):
 - OS/Distribution: Win11 / Fedora 42
 - Version:  26100.4351
- Linux Kernel Version: (don't remember exact kernel version. Up to date with F42 as of at least start of June. Either way, exact same behavior as with Windows)

## Additional context
This entire monitor seems problematic on multiple fronts.
And while the monitor is officially advertised as supporting 140W PD output, the manual also says, that according to SPR protocol, only 90W are available. The PD traces and readouts also confirm that the monitor only advertises 90W (20V, 4.5A) in SPR descriptors, but advertises 140W (28V, 5A) in EPR descriptors. This may already make the monitor non-compliant with the USB-PD spec that requires downward compatibility for 100W SPR (20V, 5A) from all EPR / >100W suppliers. But I also don't see the FW13 doing anything with EPR, so I don't think it knows or cares.

The monitor may also have other, rare issues that may or may not be related. While it uses a Intel JHL8440 controller, with my FW13 12th gen it only makes TB3 connections (instead of USB4 connections), but otherwise works fully with 90W PD supply out of the box. With a TB4 hub in between (CalDigit Element Hub, also JHL8440 controller), the monitor also only gets a TB3 connection, while the hub itself uses a USB4 connection to USB4 hosts, like it should. This should absolutely be the fault of the monitor, as I don't think the USB4 host controls the connection type of downstream-chained devices. It could request / limit compatibility, which I have checked it does not under Linux).

For my work laptop (HP Elitebook G1a, also Strix Halo, even with TB4 certification), it works for 90W PD, but fails to make any USB4 or TB3 connection at all, in any TB4 port, unless I put a TB4 hub or TB3 dock in between (then it works with the same TB3 connection as the 12th gen Framework).

This monitor has a sibling model U2725QE that seems to be identical other than the size according to other reports in the internet. And if any of it is related to Dell's TB4 controller configuration or PD implementation it may very well affect other monitors from Dell's  Uxx25 lineup, as most have TB4 input with 140W PD (and also vaguely referring to only supplying 90W to SPR devices in the manual).

I also can all but exclude any production variances of the monitor, as I have 2 of them and both show the same problems / behavior.

quinchou77, 2025-07-04:
We have another customer report the issue with this monitor and 2725QE. We are looking about this.

ramonwirsch, 2025-07-14:
Well, that was fast. I have installed the Beta on my device (no issues so far).
Will report back once I have taken my Framework into the office to retest again.

quinchou77, 2025-07-15:
> Well, that was fast. I have installed the Beta on my device (no issues so far). Will report back once I have taken my Framework into the office to retest again.

Thanks for update. Looks like the result is also positive with your system. This version is target to stable next week.

closed 2025-07-16 by ramonwirsch