FrameworkComputer/SoftwareFirmwareIssueTracker #82: keyboard backlight randomly turning on/flashing  issue  AMD AI 300 BIOS 3.04 BETA

pmichenf, 2025-07-16:
## Device Information

### System Model or SKU
Framework 13 AMD AI 9 HX 370

Please select one of the following
- [X] Framework Laptop 13 (AMD Ryzen AI 300 Series)

### BIOS VERSION
Please provide the bios version. 
AI 300 BIOS 3.04 Beta

### DIY Edition information
If you are experiencing an issue on a DIY system, Please also fill out the memory and storage devices you are using. 

Memory: Manufacture and SKU Crucial 2X24GB RAM
Storage: Manufacture and SKU  Samsung 980 pro 1TB

### Port/Peripheral information
N/A

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [X] No

## Describe the bug
When working in low light, if i turn off the keyboard backlight, it will randomly re-enable at full brightness. This happens quite frequently. I turn off the backlight and it'll after a brief period turn back on again. 

If i set the keyboard to the dimmest backlight setting, they symptom remains, however now exhibits itself as a quick flash bright, and then it returns to the dim setting i put it on.

## Steps To Reproduce
Steps to reproduce the behavior:
in a dim room, turn the keyboard backlight off. 
Just use the computer for a little bit
after a brief period the backlight will randomly turn itself back on.

## Expected behavior
the keyboard backlight setting to remain as i set it.

## Screenshots
If applicable, add screenshots to help explain your problem.

## Operating System (please complete the following information):
Linux fedora 6.15.5-200.fc42.x86_64 #1 SMP PREEMPT_DYNAMIC Sun Jul  6 09:16:17 UTC 2025 x86_64 GNU/Linux

## Additional context
Add any other context about the problem here.

pmichenf, 2025-07-16:
Another observation, i was controlling the keyboard brightness by using the function key and space bar. Today i noticed that this control doesn't seem connected to fedora 42 workstations built in control to toggle or set the keyboard backlighting. If I have the keyboard backlight set to on, in fedora's control center, and toggle the keyboard backlight settings using the function+spacebar keys, the settings in the fedora's control center does not change. 

I will play around more with this later to see if i turn off the keyboard backlight from fedora's settings, if the issue persists. My suspicion now is the symptom i'm seeing is the bios and fedora battling for control of the keyboard backlight.

ColtonIdle, 2025-07-26:
side question... how do you get beta builds of bios? google searching for framework bios beta doesn't turn up any opt in threads or anything like that

PureKrome, 2025-08-28:
Not sure if this helps, but I'm on an FW16 and have had intermittent backlight issues for about 6+ months since the most recent bios updates.

ppaslan, 2025-09-25:
I have this exact same issue and its actually quite annoying.