FrameworkComputer/SoftwareFirmwareIssueTracker #91: CPU frequency stuck at low values after suspend/resume on firmware 3.06

lsiudut, 2025-08-16:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [x] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
3.06

### DIY Edition information
If you are experiencing an issue on a DIY system, Please also fill out the memory and storage devices you are using. 

Memory: 2 x Kingston KF556S40-32
Storage: WD Black SN850X

## Describe the bug
After upgrading to firmware version 3.06, the system exhibits CPU throttling following a suspend/resume cycle.
- The issue only occurs if the laptop remains suspended for a longer period. A quick suspend/resume does not trigger it.
- So far, it reliably reproduces when the laptop is connected to the PSU. I have not yet tested battery-only scenarios but will provide feedback in a follow-up comment.
- Once resumed, CPU frequencies remain limited between ~540 MHz and 1.1 GHz, resulting in a noticeable performance drop.

**Additional observations:**

- According to cpufreq sysfs, the configured min/max CPU speeds are 1.1 GHz and ~5.1 GHz respectively.
- Switching power profiles or adjusting the energy performance preference does not restore normal operation.
- Disconnecting and reconnecting the PSU immediately resolves the throttling.

Next steps:
I will collect and attach EC logs from ectool console in the comments for further analysis.
## Steps To Reproduce
Steps to reproduce the behavior (Linux):
1. Connect PSU.
2. Suspend the system.
3. Keep it suspended for a while (~1h)
4. Resume the system and monitor CPU frequency on idle and load. None of it exceeds 1100Mhz, bottom remains around 540Mhz.

## Expected behavior
Observing CPU frequencies above 4GHz under load.

## Operating System (please complete the following information):
 - OS/Distribution: Arch Linux
- Linux Kernel Version: 6.16.1

## Additional context
Discussed on the Framework Community Forum: [here](https://community.frame.work/t/framework-laptop-16-ryzen-7040-bios-3-06-release-beta/73326/75)

lsiudut, 2025-08-16:
EC logs after disconnecting and reconnecting the PSU:
```
[262039.947200 cypd_write_reg8_wait_ack pre 0x84 ]
[262039.950100 PORT_DISCONNECT]
[262039.951000 events = 2, pre_events = 0]
[262039.952000 set AP throttling type 1 to on (0x00000010)]
[262039.956100 board_set_active_charge_port port -1, prev:3]
[262039.961500 event set 0x0400000000000000]
[262039.961300 cypd_write_reg8_wait_ack pre 0x4 ]
[262039.976600 cypd_write_reg8_wait_ack C:1 0x2032 response 0x0]
[262039.982600 event set 0x0400000000000000]
[262039.981000 cypd_cfet_vbus_control:3 fail:5]
[262039.994900 cypd_write_reg8_wait_ack C:0 0x1032 response 0x0]
[262039.996800 cypd_cfet_vbus_control:0 fail:5]
[262040.009100 cypd_write_reg8_wait_ack pre 0x0 ]
[262040.022900 cypd_write_reg8_wait_ack C:1 0x2032 response 0x0]
[262040.024500 cypd_cfet_vbus_control:3 fail:5]
[262040.025800 event set 0x0100000000000000]
[262040.081700 PMF: SPL 40000mW, sPPT 48000mW, fPPT 58000mW, p3T 118000mW, ao_sppt 0mW]
[262040.083500 events = 0, pre_events = 2]
[262040.084600 set AP throttling type 1 to off (0x00000000)]
[262040.085800 Updating charger with EPR correction: ma 440]
[262040.089600 CL: p-1 s-1 i500 v0]
[262040.091100 update charger!!]
[262040.099500 AC off]
[262040.101000 event set 0x0000000000000010]
[262040.109900 TODO Implement pd_set_new_power_request port 3]
PORT80: 3F40
[262040.115000 set AP throttling type 1 to off (0x00000000)]
[262040.121000 DC BALANCED]
[262040.122100 event set 0x0100000000000000]
[262040.152700 Battery 98% (Display 99.9 %) / 6h:40 to empty, not accepting current]
[262040.181100 PMF: SPL 30000mW, sPPT 36000mW, fPPT 44000mW, p3T 118000mW, ao_sppt 0mW]
PORT80: AA8F
[262040.223500 cypd_update_power_status:0=0x8]
[262040.227000 cypd_update_power_status:1=0x8]
PORT80: AA8E
[262040.705500 Battery 98% (Display 97.2 %) / 6h:40 to empty, not accepting current]
PORT80: 0008
[262041.215900 DC BATTERY SAVER]
[262041.216900 event set 0x0100000000000000]
[262041.266200 PMF: SPL 20000mW, sPPT 20000mW, fPPT 20000mW, p3T 118000mW, ao_sppt 0mW]
PORT80: AA8F

[262043.025400 CYPD_RESPONSE_PORT_CONNECT 3]
[262043.030100 board_set_active_charge_port port 3, prev:-1]
[262043.034600 event set 0x0400000000000000]
[262043.051600 cypd_write_reg8_wait_ack pre 0x80 ]
[262043.055700 event set 0x0400000000000000]
[262043.073200 Updating charger with EPR correction: ma 2640]
[262043.075500 event set 0x0400000000000000]
[262043.094000 CL: p3 s1 i3000 v5000]
[262043.095600 update charger!!]
[262043.101400 AC BEST EFFICIENCY]
[262043.102300 event set 0x0100000000000000]
[262043.112700 AC on]
[262043.114200 event set 0x0000000000000008]
[262043.123400 TODO Implement pd_set_new_power_request port 3]
PORT80: 0004
PORT80: AA8E
PORT80: AA8F
[262043.163000 CCG_RESPONSE_ACCEPT_MSG_RX 3]
[262043.164100 Updating charger with EPR correction: ma 440]
[262043.169700 sustain_battery_soc: Switched control mode to DISCHARGE]
[262043.173800 Battery 98% (Display 97.2 %) / 6h:45 to empty, not accepting current]
[262043.183200 cypd_update_power_status:0=0xe]
[262043.186800 cypd_update_power_status:1=0xe]
[262043.238300 Battery 98% (Display 99.9 %) / 6h:45 to empty, not accepting current]
[262043.295500 event set 0x0400000000000000]
[262043.304700 CYPD_RESPONSE_PD_CONTRACT_NEGOTIATION_COMPLETE 3]
[262043.308900 board_set_active_charge_port port 3, prev:3]
[262043.315500 cypd_write_reg8_wait_ack pre 0x84 ]
[262043.318100 event set 0x0400000000000000]
[262043.329100 AC BALANCED]
[262043.331400 event set 0x0100000000000000]
[262043.342000 event set 0x0400000000000000]
[262043.369000 event set 0x0400000000000000]
[262043.387000 event set 0x0400000000000000]
[262043.405900 event set 0x0400000000000000]
[262043.420400 PMF: SPL 30000mW, sPPT 30000mW, fPPT 30000mW, p3T 59752mW, ao_sppt 0mW]
[262043.422700 Updating charger with EPR correction: ma 2860]
[262043.434200 CL: p3 s0 i3250 v20000]
[262043.435400 TODO Implement pd_set_new_power_request port 3]
PORT80: 003B
PORT80: 0020
```

quinchou77, 2025-08-26:
We found the issue was caused by a change that adjusts the GPU power limit based on its power state. The original power limit code didn't control this well. After recently refactoring the power limit code, we can no longer reproduce the issue, though it is still under validation. Since the new code has a different structure, we don't think it's worth creating another solution for the current EC code. We plan to revert the change and have 3.07 BIOS for now and will soon release a new BIOS that includes the refactored power limit code.

PureKrome, 2025-09-04:
@quinchou77 - thank you kindly for the prompt updates. really appreciate them!

Hope 4.00 isn't too far away, then.

quinchou77, 2025-09-26:
We plan to conduct the internal 4.00 alpha test next week, followed by the beta release two weeks later. Version 3.07 will still be released as stable, even with the known issue that 240W power causes PROHOT.

xamindar, 2025-10-01:
> We plan to conduct the internal 4.00 alpha test next week, followed by the beta release two weeks later. Version 3.07 will still be released as stable, even with the known issue that 240W power causes PROHOT.

Correction, PROCHOT is caused by ANY power supply, not only 240W.

I hope you guys resolve it.

quinchou77, 2025-12-30:
We have implemented several fixes regarding the power table settings and adapter compatibility issues. Please test these updates using the latest BIOS v4.03 Beta. We are aiming for a stable release next week, provided no critical issues are reported.

https://community.frame.work/t/framework-laptop-16-ryzen-7040-bios-4-03-release-beta/79274

jo-oe, 2026-01-06:
I'm unfortunately experiencing this or a very similar issue with 4.03.
It apparently only occurs when in "performance" power profile. After a few minutes of high system load + dGPU activity, frequency drops to 544Mhz (with a few cores sometimes jumping to 1Ghz and back).

scolton99, 2026-08-17:
@quinchou77  Are you still looking for people to try out the test BIOS version that purports to fix this? I would gladly also be a guinea pig at this point.

xamindar, 2026-08-21:
> [@xamindar](https://github.com/xamindar) While I haven't been able to reproduce the issue on my end, I spotted a design pattern in the EC code that might explain what you're seeing. To test my theory, could you run framework_tool to grab the current sensor temperatures when the issue occurs? Checking these readings will help me confirm whether there's a flaw in the PROCHOT de-assert logic under those specific thermal conditions.
> 
> `framework_tool --thermal`

Alright, since applying the test bios you shared with me, I have caught it getting stuck in the 544MHz throttle three times in the past 2 weeks (much better than daily), so you are correct that it is not completely solved, but it is much better than before. 

The Framework16 is currently sitting on a desk with only the 240w Framework psu connected to it and nothing else.
It is currently in this throttled state so here are the results of the framework_tool command you requested:
```
[root@framework16 xamindar]# framework_tool  --thermal
  F75303_Local: 50 C
  F75303_CPU:   53 C
  F75303_DDR:   47 C
  APU:          44 C
  dGPU VR:      0 C
  dGPU VRAM:    0 C
  dGPU AMB:     0 C
  dGPU temp:    NotPowered
  Fan Speed:   995 RPM
  Fan Speed:   981 RPM
[root@framework16 xamindar]#
```

More possibly related or unrelated info that may help:
```
[root@framework16 xamindar]# cat /proc/cpuinfo | grep "cpu MHz";upower -b;echo "-----------------";cat /sys/class/drm/card1/device/power_state
cpu MHz         : 544.329
cpu MHz         : 1100.947
cpu MHz         : 544.257
cpu MHz         : 1100.947
cpu MHz         : 544.160
cpu MHz         : 1100.947
cpu MHz         : 544.414
cpu MHz         : 543.997
cpu MHz         : 544.334
cpu MHz         : 1100.947
cpu MHz         : 1100.947
cpu MHz         : 1100.947
cpu MHz         : 544.445
cpu MHz         : 1100.947
cpu MHz         : 544.267
cpu MHz         : 544.234
Device: /org/freedesktop/UPower/devices/battery_BAT1
  native-path:          BAT1
  vendor:               NVT
  model:                FRANDBA
  serial:               0298
  power supply:         yes
  updated:              Fri 21 Aug 2026 08:52:53 AM CDT (27 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               pending-charge
    warning-level:       none
    energy:              52.3688 Wh
    energy-empty:        0 Wh
    energy-full:         73.7467 Wh
    energy-full-design:  85.0007 Wh
    voltage-min-design:  15.48 V
    capacity-level:      Normal
    energy-rate:         0 W
    voltage:             16.361 V
    charge-cycles:       123
    percentage:          71%
    capacity:            86.7602%
    technology:          lithium-ion
    icon-name:          'battery-full-charging-symbolic'
  History (voltage):
    1787320373  16.361  pending-charge
    1787320343  16.363  pending-charge

-----------------
D3cold
[root@framework16 xamindar]#
```
Current throttle statuses reported by amdgpu_top:
```
Throttle Status: [SPPT_APU, PROCHOT_CPU, PROCHOT_GPU]  
```

vadi2, 2026-08-28:
Adding a measured data set from BIOS 04.05.

Environment: Framework Laptop 16 (AMD Ryzen 7040 Series), BIOS 04.05, Ryzen 9 7940HS, RX 7700S,
Ubuntu 24.04.4 LTS, kernel 6.17.0-35-generic, `amd-pstate-epp` status `active`.

Third-party software present and running throughout: `power-profiles-daemon`, and `fw-fanctrl`,
which takes manual fan control via `ectool fanduty` and invokes `ectool` roughly once a second.
I have not re-tested with it stopped.

## Method

Delivered clock comes from the ACPI CPPC feedback counters rather than `/proc/cpuinfo`:

```
freq = nominal_freq * (delta_delivered / delta_reference)
```

read from `/sys/devices/system/cpu/cpu*/acpi_cppc/feedback_ctrs`. Every figure below was taken
with a 16-thread busy loop running, and 16/16 cores returned advancing counters in all quoted
numbers.

CPPC statics: `highest_perf 196` (232 on prefcore cpu2/3), `lowest_perf 15`, `nominal_perf 149`,
`nominal_freq 4001 MHz`, `scaling_max_freq 5263061 kHz`.

## Clamped state

```
delivered           964 - 1067 MHz (16/16 cores)
governor / EPP      performance / performance
platform_profile    performance          boost 1
scaling_max_freq    5263061 kHz          CPPC highest_perf 196 (232 prefcore)
Tctl                43.2 - 56.0 C  (across episodes)
EC temps            ambient 45.8  charger 48.8  apu 49.8  cpu 46.8
fans                ~1050 / ~1030 RPM
dGPU                power_state D3cold
```

Not clamped, same load: 4025 - 4577 MHz.

## EC data captured while clamped

`ectool console`, emitted on each `powerprofilesctl set`:

```
[PMF: SPL 43000mW, sPPT 52000mW, fPPT 65000mW, p3T 227000mW, ao_sppt 0mW]   (balanced)
[PMF: SPL 45000mW, sPPT 54000mW, fPPT 65000mW, p3T 227000mW, ao_sppt 0mW]   (performance)
```

`ectool gpioget | grep prochot`, three captures at 11:57 and twice at 12:59 either side of a
profile toggle. All three were taken while clamped; I have no reading from an unclamped window:

```
 1 h_prochot_l                      0xA0006
```

`ectool chargestate show` while clamped:

```
ac = 1   chg_voltage = 17600mV   chg_current = 3840mA
chg_input_current = 7680mA   batt_state_of_charge = 91%
```

`ectool usbpdpower` and `ectool powerinfo` both return `EC result 1 (INVALID_COMMAND)` on this
build. The EC console ring fills with `HC 0x0051` / `HC 0x0070` at roughly 1/s on this system,
so the `PMF:` lines above were captured by draining the buffer, toggling, and dumping within ~1s.

## Recovery attempts

Each verified under a 16-thread load, measured as above.

- `powerprofilesctl` balanced then performance: cleared it once, at 11:29 (1036 to 4025 MHz)
- the same toggle on every subsequent attempt: no effect
- power-saver then performance: no effect
- charger unplug 5s and replug: no effect
- s2idle suspend, about 55 minutes: no effect
- toggle with the machine at 44 C Tctl: no effect
- `modprobe -r amd_pmf` followed by `modprobe amd_pmf`: no effect, 969 MHz before and after
- warm reboot: cleared, 4059 MHz at 1 minute uptime

The `amd_pmf` reload completed cleanly (`No Smart PC policy present`, `registered PMF device
successfully`), and `platform_profile` disappeared and returned across the unload and reload.

## Trigger

Two episodes in one session. The first was observed at 11:29; `journalctl -k` shows
`PM: suspend entry (s2idle)` at 10:47:35 and `ucsi_acpi USBC000:00: unknown error 256` in the
same second. The second began roughly 20 minutes after the first was cleared, during a sustained
16-thread C++ compile, with no suspend or resume between the two.

The dGPU reported `power_state D3cold` throughout both episodes.

## Thermal behaviour when not clamped

Measured after the reboot, 60s 16-thread load, `fw-fanctrl` on its `lazy` curve:

```
  t     Tctl   cpu@4c   fan1 RPM    PPT
   0s   62.1    60.9      1435      4 W
   5s   99.5    98.8      1570     46 W
  15s  100.1    99.8      3202     54 W
  25s  100.0    99.8      5371     48 W
  55s  100.0    99.8      5461     57 W
```

The die reaches 99.5 C within 5s of load onset and holds 100.0 C for the remainder, including
from t=25s onward with the fans at their maximum 5461 RPM.

## Logged data

I have an 84-sample CSV logged at roughly 15s intervals from 11:52 to 13:03, covering the second
episode, the 55-minute suspend and the recovery attempts: timestamp, delivered MHz, load,
running procs, four EC temps, Tctl, both fan RPMs, profile, AC, battery status/pct/watts and
dGPU power state. Happy to attach it if useful.

xamindar, 2026-09-15:
I notice there are a lot of separate reports of this, or related, bugs on here. [255](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/255), [202](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/202), [182](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/182), [133](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/133), [118](https://github.com/FrameworkComputer/SoftwareFirmwareIssueTracker/issues/118).

As a follow-up to my post above, my machine still does go into the 544MHz throttle on occasion. But not every day as it did before. It also does seem to occur more often when connected to the 240w psu vs the 180w psu. In all my instances, simply sleeping the laptop and waking it again clears the throttle without rebooting.

Has there been any further movement on this issue? I am really looking forward to a permanent fix.

pjbuterbaugh, 2026-09-16:
+1 Affected, my personal work-around has been using hibernate instead of sleep.

dmx0987654321, 2026-09-28:
+1 affected, BIOS 4.05. Does not seem connected to sleep for me, as it can occur following hibernation or restart.

xamindar, 2026-10-23:
It really feels like Framework has abandoned us on this. Or is it some hardware fault that can't be fixed?
This throttling is an almost daily occurrence for me when using the 240W Framework PSU. Yet we go months at a time without anything from Framework on a fix for it.

spockfish, 2026-10-23:
> It really feels like Framework has abandoned us on this.

Yep. I've contacted them officially, and they weren't even aware of this issue. They really don't care about customers, only new ones.

quinchou77, 2026-11-11:
We have identified a solution to address the CPU locks issue on both the AMD Ryzen 7040 and AI 300 series platforms. We have been validating the fix with select customers, and initial test results have been positive. We are targeting a BIOS release around mid-December and will provide further updates as progress continues. This build will also include several PD fixes.

LapysDev, 2026-11-11:
+1 affected, BIOS 4.05. The only “fix” so far has been to unplug the laptop, fully shut it down and wait a minute before rebooting.