FrameworkComputer/SoftwareFirmwareIssueTracker #94: Laptop Issue: FW13: fw3.09 > 3.16

BSDf-dasHaus, 2025-08-19:
## Device Information

### System Model or SKU
- [X] Framework Laptop 13 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
03.09

### DIY Edition information
Memory: 64gb supplied by FW with device
Storage: WD BLACK SN850X 2000GB

## Describe the bug
failed to set /sys/firmware/efi/efivars/fwupd-b5f7dcc1-568c-50f8-a4dd-e39d1f93fda1-0-0abba7dc-e516-4167-bbf5-4d9d1c739416 as mutable: failed to open: No such file or directory

ls -al /sys/firmware/efi/efivars/fwupd* shows no fwupd entries at all

## Steps To Reproduce
sudo bash 
fwupdmgr refresh --force
fwupdmgr get-updates
fwupdmgr update

confirmed behaviour consistency using the pop!OS GUI fw update button

## Expected behavior
installation of bios

## Operating System (please complete the following information):
Linux weasel 6.15.9-x64v3-xanmod1 #0~20250801.g053a7c7 SMP PREEMPT_DYNAMIC Sat Aug  2 22:07:41 UTC x86_64 x86_64 x86_64 GNU/Linux

fully patched Pop!OS 22.04

##extra context  -output of failed cli
root@weasel:/home/user# fwupdmgr update
Devices with no available firmware updates: 
 • WD BLACK SN850X 2000GB
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.0.3.9 to 0.0.3.16?                            ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Update AMD PI1.2.0.0c; Added Framework's dbx key and update Windows Secure   ║
║ Boot CA; Fixed hardware encryption on OPAL drives causes a missing boot      ║
║ drive issue on the next boot; Implement Battery Charge Limiting battery      ║
║ status; Implement the Framework EC device in BIOS;                           ║
║                                                                              ║
║ Laptop 13 (AMD Ryzen 7040Series) must remain plugged into a power source     ║
║ for the duration of the update to avoid damage.                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Decompressing…           [                                       ] Less than one minute remaining…
failed to set /sys/firmware/efi/efivars/fwupd-b5f7dcc1-568c-50f8-a4dd-e39d1f93fda1-0-0abba7dc-e516-4167-bbf5-4d9d1c739416 as mutable: failed to open: No such file or directory

root@weasel:/home/user# whoami
root

JohnAZoidberg, 2025-08-20:
PopOS has very old packages, so I think your fwupd package is too old.
This looks like the same issue as https://github.com/fwupd/fwupd/issues/8456
This issue is fixed in [2.0.7](https://github.com/fwupd/fwupd/releases/tag/2.0.7) and [1.9.29](https://github.com/fwupd/fwupd/releases/tag/2.0.7)

BSDf-dasHaus, 2025-08-22:
thank you for the response...

off to do a source build of fupdmgr

closed 2025-08-22 by BSDf-dasHaus