FrameworkComputer/SoftwareFirmwareIssueTracker #95: Poor Wi-Fi speeds when connected to a router with 6 GHz radio

SBCV-apegram, 2025-08-20:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [x] Framework Laptop 13 (AMD Ryzen™ AI 300 Series)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
3.04 (Driver bundle 1.04)

### DIY Edition information
If you are experiencing an issue on a DIY system, Please also fill out the memory and storage devices you are using. 

Memory: Crucial CT2K8G56C46S5
Storage: WD_BLACK WDS500G4X0E 

### Port/Peripheral information
If you are experiencing an issue with a peripheral or an expansion card/port please fill out the following information: 
* Peripheral vendor and name: AMD RZ717 Wi-Fi 7 card
* Port the Peripheral was connected to: Wi-Fi M.2 slot on mainboard

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug
When the laptop is connected to my coworker's home network (Verizon CR1000A Wi-Fi 6E router with mesh nodes), he gets ~30 - 40 Mbps up and down on a 1 Gbps connection. Other devices get ~300 - 400 Mbps up and down. This same laptop on the office Wi-Fi (UniFi Wi-Fi 6 AP's), my home Wi-Fi (Omada Wi-Fi 6 AP's) and various other Wi-Fi networks gets the normal speed for that network. Disabling the 6GHz radio on his home router nets positive results for the laptop, but devices connected to the mesh nodes have degraded performance. I don't have any other AP's with 6 GHz to confirm. We are testing setting the Wi-Fi card to 802.11ax mode instead of 802.11be mode.

## Steps To Reproduce
Steps to reproduce the behavior:
1. Connect to Verizon router with 6GHz radio enabled
2. Speeds are horrible
3. Connect to any other network
4. Speeds are normal

## Expected behavior
I expect speeds consistent with the stated practical speeds for a given Wi-Fi network

## Screenshots
If applicable, add screenshots to help explain your problem.

## Operating System (please complete the following information):
 - OS/Distribution: Windows 11
 - Version: 10.0.26100.4946

## Additional context
Add any other context about the problem here.

quinchou77, 2025-09-04:
We were unable to test with the Verizon CR1000A Wi-Fi 6E router, but our testing with the UniFi U7 Pro shows normal speeds. What is the driver version on your system? There is a speed issue on previous version, not sure if you are using the latest version or the old version.

[image]

[

[image]

](url)

quinchou77, 2025-09-04:
If you are using the most recent driver bundle, the version should be the 3366. The speed issue should be fixed with v.3366 or above.

SBCV-apegram, 2025-09-04:
[image]

Confirmed, driver version is 3366 and only works well in AC mode on that router. All other locations have normal speeds in BE mode.

I recommended that he upgrade his router system (I can't stand the crap that ISP's give out anyways), but I can't force him to, and this is the only device on his network that has problems. He has iPhones, an iPad, a couple of Apple TVs, an old Intel MacBook, and his old Dell laptop. None of them have speed issues, and except for his iPad and iPhones, I don't think any of them are Wi-Fi 6/6E or 7 devices. If you need any logs to try and solve this, I'll provide what I can.