FrameworkComputer/SoftwareFirmwareIssueTracker #96: BIOS 3.16 broke external monitor (USB-C DisplayPort Alt Mode) and caused severe boot regression on FW13 AMD 7040

hirefrank, 2025-08-20:
## Device Information
### System Model or SKU
- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [ ] Framework Laptop 13 (13th Gen Intel® Core™)
- [x] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
03.16

### DIY Edition information
Memory: Framework 64GB (2x32GB) DDR5-5600  
Storage: WD_BLACK SN850X NVMe SSD 1TB

### Port/Peripheral information
* Peripheral vendor and name: Samsung M8 Monitor via USB-C cable
* Port the Peripheral was connected to: Tested in both USB-C ports (1,3) - issue persists across all ports
* Device or expansion card attached to the **Adjacent ports**:
1. Framework USB-C Expansion Card 
2. Framework HDMI Expansion Card
3. Framework USB-C Expansion Card
4. Framework USB-A Expansion Card

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [x] No

## Describe the bug
After updating to BIOS 3.16, external monitors connected via USB-C completely stopped working. USB-C DisplayPort Alt Mode appears to be completely broken. Additionally, system boot time increased dramatically from ~30 seconds to over 3 minutes. The system shows critical UCSI (USB-C Connector System Software Interface) errors and AMD GPU timeouts in logs.

## Steps To Reproduce
1. Update Framework Laptop 13 AMD 7040 to BIOS 3.16
2. Connect external monitor (Samsung M8) via USB-C cable to any USB-C port
3. Monitor is not detected by system - no DisplayPort Alt Mode negotiation occurs
4. Check system logs for UCSI errors
5. Observe severely degraded boot performance

## Expected behavior
External monitor should be detected via USB-C DisplayPort Alt Mode as it was before BIOS 3.16. Boot time should remain fast (~30 seconds).

## Screenshots
System only detects internal display (eDP-1):
```
Monitor eDP-1 (ID 0):
        2256x1504@59.99900 at 0x0
        description: BOE 0x0BCA
        [... internal display working normally ...]
```

All external DisplayPort outputs show "disconnected":
```
/sys/class/drm/card1-DP-1/status: disconnected
/sys/class/drm/card1-DP-2/status: disconnected
[... all DP ports disconnected ...]
```

## Operating System (please complete the following information):
- OS/Distribution: Arch Linux
- Version: Rolling release (current)
- Linux Kernel Version: `Linux framework 6.16.1-arch1-1` (also tested with `6.12.42-1-lts` - same issue)

## Additional context

**Timeline:** Both BIOS 3.16 and kernel updates were performed while NOT connected to external monitor. Issue discovered when attempting to connect Samsung M8 monitor via USB-C afterward.

**Boot Performance Regression:**
```
Startup finished in 15.554s (firmware) + 1min 4.676s (loader) + 1min 51.875s (kernel) + 3.301s (userspace) = 3min 15.409s
```

**Critical System Errors:**
```
ucsi_acpi USBC000:00: unknown error 0
ucsi_acpi USBC000:00: GET_CABLE_PROPERTY failed (-5)  
ucsi_acpi USBC000:00: unknown error 256
amdgpu 0000:c1:00.0: [drm] REG_WAIT timeout 1us * 1000 tries - dcn314_dsc_pg_control
usb 5-1: device descriptor read/64, error -110/-71/-62
```

**Troubleshooting Completed:**
- ✅ BIOS reset to defaults (F9) - no improvement
- ✅ Tested Samsung M8 via USB-C on both USB-C ports - issue persists
- ✅ Tested different USB-C expansion cards - issue persists across both cards
- ✅ Verified Samsung M8 + USB-C cable work with other devices
- ✅ Tested both current and LTS kernels - confirms BIOS regression, not kernel issue
- ✅ Various kernel parameters attempted - no improvement

**Technical Analysis:** UCSI errors indicate BIOS 3.16 broke USB-C DisplayPort Alt Mode initialization at the firmware level. This affects core USB-C DisplayPort functionality and suggests a significant regression in BIOS 3.16.

**Request:** Is this a known USB-C DisplayPort Alt Mode regression? Any timeline for BIOS 3.17 to address these critical USB-C issues?

JohnAZoidberg, 2025-08-20:
> Additionally, system boot time increased dramatically from ~30 seconds to over 3 minutes.

Does that happen every time? Or just the first boot after the bios update?

hirefrank, 2025-08-20:
> <FrameworkComputer/SoftwareFirmwareIssueTracker/issues/96/3208476331@
> github.com>

Every time.

Frank Harris
hirefrank.com / @hirefrank <http://twitter.com/hirefrank>

JohnAZoidberg, 2025-08-21:
> Startup finished in 15.554s (firmware) + 1min 4.676s (loader) + 1min 51.875s (kernel) + 3.301s (userspace) = 3min 15.409s

Oh it's actually the kernel taking a long time, not the BIOS.
Can you upload your full dmesg log please?

hirefrank, 2025-08-22:
Thanks for the response! Just to clarify after additional testing:

✅ **Boot time issue is RESOLVED** - after re-applying kernel 6.16.1 update, boot time returned to normal (38s)  
❌ **USB-C DisplayPort still completely broken** - external monitor not detected on any USB-C port

**Additional testing performed:**
- ✅ Tested both kernel 6.16.1 and 6.12.42-LTS - external monitor broken on both kernels
- ✅ This confirms it's a **BIOS 3.16 regression**, not a kernel issue
- ✅ Boot time was fixed by re-applying kernel update, but USB-C DisplayPort remains broken

**Current UCSI errors showing USB-C firmware failure:**
```
ucsi_acpi USBC000:00: error -ETIMEDOUT: PPM init failed  
amdgpu REG_WAIT timeout dcn314_dsc_pg_control
```

The core issue is **UCSI (USB-C firmware) fails to initialize**, preventing any DisplayPort Alt Mode functionality. This is a BIOS 3.16 firmware regression."

This makes the story clearer and shows you did proper isolation testing to prove it's a BIOS issue.

[current_dmesg.log](https://github.com/user-attachments/files/21937855/current_dmesg.log)

hirefrank, 2025-08-22:
Should I downgrade the bios? If so, how do I do that?

quinchou77, 2025-08-26:
Could you try to downgrade the BIOS to check the issue again with the same Linux kernel? If this relates to the BIOS/EC, the issue should happens on every OS.
We tried to use 3.16 with different OSs each ports work well with DP Alt mode.

hirefrank, 2025-08-31:
Re-flashing 3.16 via EFI seemed to work. Closing this out now.

closed 2025-08-31 by hirefrank

pwnage101, 2025-11-17:
Just reporting that I've encountered the same issue and came to the same resolution:

1. Starting Condition:
   * Framework 13 AMD Ryzen 4xxx
   * Ubuntu 24.04
   * Kernel version == ???
   * System Firmware version == ???
2. Upgrade to system firmware 3.16 (using "Firmware Updater" app on Ubuntu).
3. **External monitor no longer detected** and `dmesg` contains lines that suggest amdgpu was experiencing timeouts when plugging in the monitor. Confirmed monitor not detectable with many kernel versions:
   * 6.8.x
   * 6.14.x (stock ubuntu 24.04 at time of writing)
   * 6.17.x
5. Downgraded to system firmware 3.09, external monitor **started working again.**

Subsequently, I re-upgraded to system firmware 3.16 using the USB/EFI method (extract firmware zip file to FAT32 USB drive), and the **external monitor continues to function.**

Call To Action
===
Please edit your [announcement thread](https://community.frame.work/t/framework-laptop-13-ryzen-7040-bios-3-16-release-stable/73325) to remove `fwupdmgr` as an approved upgrade path, instead directing people to use the EFI method.

medicalwei, 2025-11-19:
I am not a firmware developer, I downloaded firmware from LVFS here:

https://fwupd.org/lvfs/devices/work.frame.Laptop.Ryzen7040.BIOS.firmware

and compared against the UEFI shell firmware package:

https://downloads.frame.work/bios/Framework_Laptop_13_Ryzen7040_BIOS_3.16.zip

They are containing the same firmware payload

```
ced39dbf8cd5e0ce16328dc01cf735f0d25d31d5fc8db7590273824c73a620cd  Framework_Laptop_13_Ryzen7040_capsule_signed_allsku_3.16.cap
```

On UEFI shell firmware package, it seems to supply a capsule loader from TianoCore project:
https://github.com/tianocore/tianocore-wiki.github.io/blob/main/Capsule-Based-System-Firmware-Update-DSC-FDF.md?plain=1

On Debian, it is using fwupd-efi as a capsule loader:
https://github.com/fwupd/fwupd-efi/tree/main/efi

I am not sure whether it is the difference that causes firmware not updated as expected.