FrameworkComputer/SoftwareFirmwareIssueTracker #97: Framework 13 / 13th Gen BIOS 3.08: can only use single external monitor via USB-C DP

rh-sp, 2025-08-25:
## Device Information

### System Model or SKU

Please select one of the following

- [ ] Framework Laptop 13 (11th Gen Intel® Core™)
- [ ] Framework Laptop 13 (12th Gen Intel® Core™)
- [x] Framework Laptop 13 (13th Gen Intel® Core™)
- [ ] Framework Laptop 13 (AMD Ryzen™ 7040 Series)
- [ ] Framework Laptop 13 (Intel® Core™ Ultra Series 1)
- [ ] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
```
> sudo dmidecode --string bios-version
03.08
```

### Port/Peripheral information
1. USB-C expansion card (connected to [CableMatters USB4 Mini Dock with Dual DisplayPort](https://www.cablematters.com/pc-1577-126-usb4-mini-dock-with-dual-displayport.aspx))
    a. Connected to 2 1080p monitors, one of them 144Hz the other 60Hz
    b. Connected to wired Ethernet
    c. Connected to USB mouse and keyboard, USB headset
2. USB-C expansion card (disconnected)
3. USB-C expansion card (disconnected)
4. USB-C expansion card (disconnected)

### Standalone Operation
Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?
- [ ] Yes
- [ ] No
- [x] Unsure (can check if relevant)

## Describe the bug
Either screen by itself works fine. As soon as I plug in the second screen, both screens go black and disappear from the system's display configuration.

This only started happening after applying BIOS 3.08 EFI Shell update. Previously, I had applied BIOS 3.08 upgrade through `fwupdmgr` and I think didn't have the problem. I had never applied an EFI Shell update before, only updated BIOS using `fwupdmgr`.

## Steps To Reproduce
Steps to reproduce the behavior:
1. Plug CableMatters USB4 Mini Dock into expansion slot 1
2. Plug first monitor into CableMatters USB4 Mini Dock (it works)
3. Plug in second monitor (both monitors stop working and disappear from the system's display settings)

## Expected behavior
Both monitors work as expected.

## Screenshots
One monitor plugged in: 
[image]
Both monitors plugged in:
[image]

## Operating System
 - OS/Distribution: Linux Mint Cinnamon
 - Version: 22.1
- Linux Kernel Version: `Linux work-laptop 6.8.0-71-generic #71-Ubuntu SMP PREEMPT_DYNAMIC Tue Jul 22 16:52:38 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux`

## Additional context
Previously. both monitors worked at the same time with no issue. Originally, they even worked when they were plugged in at boot time. This broke some time ago (maybe due to an earlier BIOS update, but not sure which), external monitors have only been working if I plugged them in _after_ boot.

Since the 3.08 release notes mentioned "Added Dual Display (Panel + External Monitor) support for pre-boot (BIOS/POST)," I thought that maybe it would fix the issue with monitors not working if they were plugged in at boot time. Since applying the update through `fwmgr update` didn't help, I thought to apply the EFI shellk update.

Unfortunately, that only made things worse, and now only one external monitor ever works. It still has to be plugged in after boot. I feel like every BIOS update makes the system less usable and consider downgrading as far as I can to regain functionality.

quinchou77, 2025-09-26:
Added Dual Display (Panel + External Monitor) support for pre-boot (BIOS/POST) > This only affects the monitor display logic before entering the OS.

rh-sp, 2026-01-14:
Same problem still exists with 3.09.

`fwupdmgr downgrade` only allows me to downgrade to 3.08.

Is there a way I can downgrade to an older BIOS so that I can use my second external monitor again?