FrameworkComputer/SoftwareFirmwareIssueTracker #98: No upgrade actually occurs for System Firmware 0.0.3.4 to 0.0.3.8

Harmelodic, 2025-08-31:
## Device Information

### System Model or SKU

- [x] Framework Laptop 13 (13th Gen Intel® Core™)

### BIOS VERSION

`03.04`

### DIY Edition information

Memory: `DDR4-3200 - 32 GB (2 x 16 GB) (Unknown Manufacturer and SKU)`
Storage: `WD_BLACK™ SN770 NVMe™- M.2 2280 - 500GB`

### Port/Peripheral information

n/a

### Standalone Operation

Are you running your mainboard as a standalone device. Is standalone mode enabled in the BIOS?

- [ ] Yes
- [x] No

## Describe the bug

I can see in the Ubuntu Software app, and when using `fwupdmgr` that I am running System Firmware `0.0.3.4` and need to upgrade to `0.0.3.8`.

I've triggered the upgrade multiple times (both from the Ubuntu Software app and from `fwupdmgr`) and the same thing always happens:

- I am asked if I want to update to `0.0.3.8` (Yes)
- It downloads an update.
- It asks to reboot for this update (Yes)
- It reboots.
- I log in and find I'm still running `0.0.3.4`.

## Steps To Reproduce

Steps to reproduce the behavior:
1. `fwupdmgr refresh --force`
2. `fwupdmgr update`
3. Let reboot.
4. Log in
5. Open `Terminal`
6. `fwupdmgr get-devices`
7. See that "System Firmware" is still `Current version: 0.0.3.4`

## Expected behavior

The upgrade from `0.0.3.4` to `0.0.3.8` to actually work.

## Screenshots

TODO: Will attach.

## Operating System (please complete the following information):

- OS/Distribution: Ubuntu
- Version: 25.04
- Linux Kernel Version: `6.14.0-29-generic`

## Additional context

After attempting the update, rebooting and logging in, I also often get a notification from Ubuntu Software saying:

> Software Update Installed
> An important OS update has been installed.

Clicking "Review" on this notification opens Ubuntu Software and shows a pop-up window titled "Installed Updates" which shows:

> Fingerprint Sensor Upgrade Device Update
> ✅ Installed
> Device cannot be used during update.

Closing this window shows I still have a Device Firmware update available from `0.0.3.4` to `0.0.3.8`.

quinchou77, 2025-09-04:
Could you check the ESP partition size if it is enough?

Harmelodic, 2025-09-04:
> Could you check the ESP partition size if it is enough?

Partition mounted at `/boot/efi` has size 1.05 GiB and has 8.32 MiB used. Plenty of space! 👍

JohnAZoidberg, 2025-09-26:
> I'd really rather not re-install my distro, or try a different on

I understand

Would you mind to use the EFI Shell updater?
https://knowledgebase.frame.work/framework-laptop-13-bios-and-driver-releases-amd-ryzen-7040-series-r1rXGVL16

It uses the same EFI capsule as LVFS under the hood but avoids any issues related to the OS or bootloader configuration.

Harmelodic, 2025-09-26:
> Would you mind to use the EFI Shell updater? https://knowledgebase.frame.work/framework-laptop-13-bios-and-driver-releases-amd-ryzen-7040-series-r1rXGVL16
> 
> It uses the same EFI capsule as LVFS under the hood but avoids any issues related to the OS or bootloader configuration.

Not sure why you linked the Ryzen guide, when I'm running an Intel 13th Gen, but anyway, I'll assume that was just a misread/mislink 😅 

I followed the Intel guide: https://knowledgebase.frame.work/en_us/framework-laptop-bios-and-driver-releases-13th-gen-intel-core-BkQBvKWr3

First time it seemed like some updates failed and was searching for an adapter. I took this to mean connecting the USB-C power cable, so I plugged it in and some other updates happened succesfully. At the end of the first time, it still seemed to be running `0.0.3.4`.

Tried a second time with the USB-C power cable connected the whole time it was doing all the updates (and it was a bunch of updates), and now:

```
~ $ sudo dmidecode 
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.
55 structures occupying 4158 bytes.
...
BIOS Information
	Vendor: INSYDE Corp.
	Version: 03.09
```

and

```
~ $ fwupdmgr update
Devices with the latest available firmware version:
 • Fingerprint Sensor
 • System Firmware
Devices with no available firmware updates: 
...
```

So, it seemed like that worked! It looks like I'm running `03.09` / `0.0.3.9` now.

It's a shame LVFS didn't work...

JohnAZoidberg, 2025-10-06:
@universalhandle you've got a different issue.

If `fwupdmgr get-updates` says no update available then that's what it means. It cannot find a matching update and won't try to run an update.

`Device ID` and `GUID` is nothing specific to your unit. It's the same across all insurances of the same hardware.

> The "fedora" in the distributions is curious

That's the distributions we have tested it on. You see it says `Tested`

I just checked version 3.0.9. it's not in stable yet, it's in the testing repository. You don't have that enabled. Your refresh command doesn't fetch it.
So please run: 

```
fwupdmgr enable-remote lvfs-testing
fwupdmgr refresh --force
```

@quinchou77 
Our forum post says stable already. Did we forget to move it in lvfs?
https://community.frame.work/t/framework-laptop-13-13th-gen-intel-core-bios-3-09-release-stable/74869