FrameworkComputer/SoftwareFirmwareIssueTracker #99: Intel BE200 not getting past POST in AMD laptops

bzfbd, 2025-09-05:
## Device Information

### System Model or SKU

Please select one of the following

- [x] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### BIOS VERSION
3.05

FreeBSD:

### Port/Peripheral information
Intel BE200 in m.2 slot.

## Describe the bug
System does not boot if Intel BE200 wireless card is inserted into m.2 slot.

## Steps To Reproduce
Replace AMB/Mediatek wireless card with an Intel *BE200*

## Expected behavior
OS boots and wireless card is detected on PCI bus.

## Operating System (please complete the following information):
FreeBSD 16-CURRENT but N/A as never get to a boot screen.

## Additional context
Various issues in the community, e.g.,
https://community.frame.work/t/intel-wi-fi-7-be200-and-be202-compatible-with-framework-16/45786/10
https://community.frame.work/t/fw16-wifi-7-upgrade-w-intel-be200-wifi-7-card-wont-post/74884
https://community.frame.work/t/intel-wi-fi-7-be200-and-be202-compatible-with-framework-16/45786/44
....

JohnAZoidberg, 2025-09-06:
Unfortunately that is a limitation of the BE200 card.
It is PCIe but not compatible with AMD.
We could possibly make it POST but still the card wouldn't show up.
AX200 or AX210 work fine on AMD.

cc @quinchou77

bzfbd, 2025-09-10:
I think making it POST w/o the card showing up (or even a warning message during POST) would be preferable to a black screen.

bzfbd, 2025-09-10:
Thanks for the clarifications.

Sad state.  Let's hope the next gen cards will have better interop again.